RIOT-OS/RIOT #5557: STM8 port

spth, 2016-06-18:
How about porting RIOT to STM8? It already has an STM32 port, and AFAIK, the peripherals are quite similar.

Philipp

gebart, 2016-06-21:
I have never used any STM8 but from what I gather from the web it has similar specs as the ATmegas (avr) which puts it in the lower end of the spectrum of devices capable of running RIOT. The biggest issue may be that there is no GCC port, and we have not tested building RIOT on any compilers other than GCC and Clang AFAIK. I have never used SDCC or IAR so I wouldn't know how big the required effort is, but I'm guessing it is likely quite a bit more than just changing `CC=` in the Makefile.
Feel free to start your own port and open a PR against the RIOT repository when you have something you want feedback on. You can open a PR even when it is work in progress, it doesn't have to be a completely finished port.

spth, 2016-06-21:
There are four important C implementations for the STM8: http://colecovision.eu/stm8/compilers.shtml
Their standard-support is at different levels, but in general, I'd say IAR and SDCC are the most advanced. The SDCC regression test suite includes most of the tests that GCC uses (excluding in particular those that rely on GCC extensions). AFAIK neither IAR nor SDCC aim for GCC-compability the way Clang does.

Philipp

marrakchino, 2018-09-12:
Any update on this issue?

ST seems to produce 8 bit MCUs (e.g.: https://www.st.com/en/evaluation-tools/nucleo-8l152r8.html)

spth, 2018-09-17:
I do not know of any progress, and currently do not have time to initiate work on this myself. But if someone wants to work on it, I will help (I do not have much experience with RIOT, but know SDCC, and the STM8 reasonably well).

Philipp

closed 2019-09-10 by stale[bot]