RIOT-OS/RIOT #5561: C++11 extensions in header files

jia200x, 2016-06-20:
Hi.

I'm having some warnings when trying to compile C++ RIOT with some header files.
For example, I added "net/netdev2.h" in riot_and_cpp and example and got errors like this:

```
/home/jialamos/RIOT/sys/include/net/netopt.h:225:17: error: comma at end of enumerator list [-Werror=pedantic]
     NETOPT_NUMOF,
```

Commas at the end of enumerator list are valid in C99, but seems not to be valid when linking with G++ and pedantic flag.
I'm using Gcc 4.8.4. 

Anyone else with a problem like this?

Kijewski, 2016-06-20:
Why did you close the issue? You are right!

C++03 does not allow trailing commas, this feature was only added in C++11. C99 does allow trailing commas, but legends tell that C++03 was intended be compatible with ANSI C, which did not allow trailing commas.

jia200x, 2016-06-20:
Hi, I closed this issue because due to browser fault it was posted twice.

jia200x, 2016-06-20:
So, we would need to remove the trailing commas of all files.
I know some net header files that have this, I will try to identify the rest.

gebart, 2016-06-20:
There's another issue somewhere in the tracker, but I don't remember the
issue number. I think that it's only the msp430 which still uses a gcc 4.6
toolchain which would have trouble with c++11. All other platforms have
well supported toolchains based on gcc 4.8 or later.

kaspar030, 2016-06-20:
This [1] page from TI suggests there's a GCC 5.3.0 for msp430!

[1]
http://software-dl.ti.com/msp430/msp430_public_sw/mcu/msp430/MSPGCC/latest/index_FDS.html

gebart, 2016-06-20:
Yes, see my pr wip (search for newlib in the pr list) for the msp430 newlib
support. It needs some love though.
I'm replying via email so I'm sorry I couldn't quote the pr number directly.

gebart, 2016-06-23:
#4766 is the PR for supporting the msp430-elf-gcc-5.3.0 compiler with newlib as libc.

miri64, 2016-10-18:
I mark this as a known issue though it seems rather specific. If you don't agree, shout! ;-)

kYc0o, 2017-04-26:
Someone have news on this one?

smlng, 2018-01-29:
@jia200x please check if this issue persist, I do not have that problem in macOS. If resolved, please close!

Kijewski, 2018-01-29:
The trailing commas are still there [in master](https://github.com/RIOT-OS/RIOT/blob/5cf71e3096e72d7ceb891acd3cd2b70ea54f73e8/sys/include/net/netopt.h), lines 465, 473, 495, 507.

RIOT defaults to -std=c99, but AFAICT there is no global switch for C++, only for some modules.

kYc0o, 2018-07-17:
@jia200x can you confirm if one of your PRs solved this?

kYc0o, 2018-08-15:
@jia200x did you? Maybe this can be tackled by your efforts on coding conventions compliance?

jia200x, 2018-08-15:
@kYc0o sorry, I completely forgot. I will try to find some time today.

> Maybe this can be tackled by your efforts on coding conventions compliance?

Makes sense

jia200x, 2018-08-20:
This issue still persist for me:

```
Building application "riot_and_cpp" for "native" with MCU "native".

In file included from /Users/jia200x/Development/RIOT/examples/riot_and_cpp/main.cpp:31:
/Users/jia200x/Development/RIOT/sys/include/net/netopt.h:564:17: error: commas
      at the end of enumerator lists are a C++11 extension
      [-Werror,-Wc++11-extensions]
    NETOPT_NUMOF,
                ^
/Users/jia200x/Development/RIOT/sys/include/net/netopt.h:572:22: error: commas
      at the end of enumerator lists are a C++11 extension
      [-Werror,-Wc++11-extensions]
    NETOPT_ENABLE = 1,          /**< enable a given option */
                     ^~
/Users/jia200x/Development/RIOT/sys/include/net/netopt.h:594:25: error: commas
      at the end of enumerator lists are a C++11 extension
      [-Werror,-Wc++11-extensions]
    NETOPT_STATE_STANDBY,       /**< standby mode. The devices is awake but
                        ^~
/Users/jia200x/Development/RIOT/sys/include/net/netopt.h:606:33: error: commas
      at the end of enumerator lists are a C++11 extension
      [-Werror,-Wc++11-extensions]
    NETOPT_RF_TESTMODE_CTX_PRBS9,   /**< PRBS9 continuous tx mode */
                                ^~
4 errors generated.
make[1]: *** [/Users/jia200x/Development/RIOT/examples/riot_and_cpp/bin/native/application_riot_and_cpp/main.o] Error 1
make: *** [/Users/jia200x/Development/RIOT/examples/riot_and_cpp/bin/native/application_riot_and_cpp.a] Error 2
```

BP-de-Jose-4:riot_and_cpp jia200x$ clang --version
Apple LLVM version 7.3.0 (clang-703.0.31)
Target: x86_64-apple-darwin15.3.0
Thread model: posix

I will try to update my toolchain here.

maribu, 2022-09-19:
We since have bumped `-std` to `-std=c++14`. So trailing commas in initializer lists are no longer an issue :)

closed 2022-09-19 by maribu