RIOT-OS/RIOT #5566: RTC periph test

dondonprad, 2016-06-22:
i tried compile main.c in tests/periph_rtc 
i using arduino-due board  
i think rtc.h not compiled in main.c, i ask how fix it?? 

[image]

PeterKietzmann, 2016-06-23:
@dondonprad it seems there is currently no rtc implementation for the SAM3x controller in RIOT. One could fix it by writing an appropriate driver to the [peripheral rtc interface](https://github.com/RIOT-OS/RIOT/blob/master/drivers/include/periph/rtc.h.

iia, 2018-04-30:
Hi,

I'm interested to fix this issue. What is the current state of this issue ?

Cheers.

aabadie, 2019-09-10:
Still valid issue

closed 2020-04-13 by stale[bot]