RIOT-OS/RIOT #5575: boards/msb-430(h?): can not flash

PeterKietzmann, 2016-06-28:
1.What is the difference between *msb-430* and *msb-430h*?
2. I'm using the OLIMEX adapter for flashing. `ls /dev/` shows up a `ttyACM0` device but no `ttyUSB0` device. However, in the [Makefile.include](https://github.com/RIOT-OS/RIOT/blob/master/boards/msb-430-common/Makefile.include#L6) the port is defined as `ttyUSB0`. Did I confuse something?
3. Neither with `ttyUSB0` nor with `ttyACM0` I was able to flash my msb-430h board. Anyone had this before?

```
MSPDebug version 0.22 - debugging tool for MSP430 MCUs
Copyright (C) 2009-2013 Daniel Beer <dlbeer@gmail.com>
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Resetting Olimex command processor...
Initializing FET...
FET protocol version is 10004002
Set Vcc: 3000 mV
Configured for JTAG (2)
Sending reset...
fet: FET returned error code 4 (Could not find device or device not supported)
warning: fet: reset failed
fet: FET returned error code 4 (Could not find device or device not supported)
fet: command C_IDENT1 failed
Using Olimex identification procedure
fet: FET returned error code 4 (Could not find device or device not supported)
fet: missing info
fet: identify failed
Trying again...
```

PeterKietzmann, 2016-06-28:
@malosek might help?

A-Paul, 2016-07-01:
> 3. Neither with ttyUSB0 nor with ttyACM0 I was able to flash my msb-430h board. Anyone had this before?

I think this will come up when trying to use the console.
*mspdebug* connects through a native usb device *(/dev/bus/usb/XXX/NNN)*, at least it is called by make to do so.

For now I'm ending up with the same error output as you.
Is it assured that no batteries are needed? :-)

OlegHahm, 2016-07-04:
> What is the difference between msb-430 and msb-430h?

Short answer: they have different radios and a slightly different board design.
see http://www.mi.fu-berlin.de/inf/groups/ag-tech/projects/Z_Finished_Projects/ScatterWeb/modules/index.html

> I'm using the OLIMEX adapter for flashing. ls /dev/ shows up a ttyACM0 device but no ttyUSB0 device. However, in the Makefile.include the port is defined as ttyUSB0. Did I confuse something?

Yes, the msb-430 gets flashed using the JTAG, not the serial the define for `$PORT` in the Makefile defines only the serial port (you need an additional serial converter (FTDI)).

> Neither with ttyUSB0 nor with ttyACM0 I was able to flash my msb-430h board. Anyone had this before?

I could imagine two things:
1. There are different versions for the JTAG available. mspdebug supports `olimex-v1`, `olimex` (version 2) and `rf2500` (which is on the cc430). RIOT uses per default `olimex`, but it might be that I sent you a v1 one.
2. A power problem. IIRC the msb-430(h) needs to be externally powered in addition to the JTAG. So, either you use a FTDI cabel (or any other USB-to-serial convert) *with* the VCC connected or you insert some batteries.

From the error messages you get, I would assume that the latter is the cause of your problem.

PeterKietzmann, 2016-07-04:
Yepp. It was just about the power supply. Thanks @OlegHahm, @A-Paul for looking into this!

closed 2016-07-04 by PeterKietzmann