RIOT-OS/RIOT #5581: Issues about installing 6lowpan on a raspberry pi

zmingshi, 2016-06-29:
Hi,
The Wiki page [How-to install and use 6LoWPAN on a Raspberry Pi with Linux Kernel rpi-4.1.y](https://github.com/RIOT-OS/RIOT/wiki/How-to-install-6LoWPAN-Linux-Kernel-on-Raspberry-Pi) shows a detailed method to install 6LoWPAN on a raspberry pi. But it refers to linux kernel rpi-4.1.y, I want to know if the higher version like rpi-4.4.y is ok for the 6LoWPAN support?
Thank you!

smlng, 2016-06-29:
Hey @zmingshi, the wiki page is kind of outdated (should be updated or link to newer guides). Though 6Lo on RasPi is (very) important to RIOT, i.e., for testing and realistic deployments, (IMO) it does not fit into RIOT wiki. Which is why I created some (RIOT external) guides on that topic, take a look [here](https://github.com/RIOT-Makers/wpan-raspbian/wiki).

Hope this helps ...

zmingshi, 2016-06-29:
Hi, @smlng, thank you so much! It is helpful, and I'll take a careful look at the details in that page.

miri64, 2016-10-18:
@smlng could you please update the Wiki according what you wrote?

PeterKietzmann, 2016-10-19:
This issue was "solved" and isn't directly RIOT related -> can be closed. The "6LoWPAN on RasPi" page needs some polishing, agreed.

closed 2016-10-19 by PeterKietzmann