RIOT-OS/RIOT #5583: Clash between LITTLE_ENDIAN defines in RIOT and in arm-none-eabi-gcc 5.4

josephnoir, 2016-06-29:
The newest `gcc-arm-none-eabi-5_4-2016q2-20160622` from [Launchpad](https://launchpad.net/gcc-arm-embedded/+download) creates a clash between LITTLE_ENDIAN defines in RIOT and in gcc. Here is the output from building the hello world example for the samr21-xpro:

```bash
In file included from $HOME/Git/RIOT/cpu/samd21/include/cpu_conf.h:23:0,
                 from $HOME/Git/RIOT/cpu/cortexm_common/include/cpu.h:35,
                 from $HOME/Git/RIOT/cpu/samd21/cpu.c:21:
$HOME/Git/RIOT/cpu/samd21/include/samd21.h:230:0: error: "LITTLE_ENDIAN" redefined [-Werror]
 #define LITTLE_ENDIAN          1
 ^
In file included from $HOME/Tools/gcc/gcc-arm-none-eabi-5_4-2016q2/arm-none-eabi/include/sys/types.h:67:0,
                 from $HOME/Tools/gcc/gcc-arm-none-eabi-5_4-2016q2/arm-none-eabi/include/stdio.h:48,
                 from $HOME/Git/RIOT/cpu/cortexm_common/include/cpu.h:33,
                 from $HOME/Git/RIOT/cpu/samd21/cpu.c:21:
$HOME/Tools/gcc/gcc-arm-none-eabi-5_4-2016q2/arm-none-eabi/include/machine/endian.h:17:0: note: this is the location of the previous definition
 #define LITTLE_ENDIAN _LITTLE_ENDIAN
 ^
```

And a [gist](https://gist.github.com/josephnoir/f6b7bb0491818aedd05e44473202efd1) with the complete output. Just in case the OS builds differ in some way, I'm using OS X.

gebart, 2016-06-29:
I have a commit somewhere offline for fixing this, but can't make a PR
today. The fix is to delete the defines in the Sam headers, they are
unused. Feel free to open a PR if you have the time.

OlegHahm, 2016-07-05:
How about #5600?

PR #5600 (2016-07-05): samd21: do not redefine LITTLE_ENDIAN
fixes #5583
closed 2016-07-05 by gebart