RIOT-OS/RIOT #5594: boards/samr21-xpro: is SPI_1_DIPO correct?

keestux, 2016-07-03:
Hi,

Can someone explain why SPI_1_DIPO is the correct value? I'm talking about this [periph_conf.h](https://github.com/RIOT-OS/RIOT/blob/master/boards/samr21-xpro/include/periph_conf.h#L181)

    /*      SPI1             */
    #define SPI_1_DEV          SERCOM5->SPI
    #define SPI_IRQ_1          SERCOM5_IRQn
    #define SPI_1_DOPO         (1)
    #define SPI_1_DIPO         (2)

    #define SPI_1_SCLK_DEV     PORT->Group[1]
    #define SPI_1_SCLK_PIN     (23)

    #define SPI_1_MISO_DEV     PORT->Group[1]
    #define SPI_1_MISO_PIN     (02)

    #define SPI_1_MOSI_DEV     PORT->Group[1]
    #define SPI_1_MOSI_PIN     (22)

If I translate this a little I get
SCLK -> PB23, SERCOM5, PAD3, MUX D
MISO -> PB02, SERCOM5, PAD0, MUX D
MOSI -> PB22, SERCOM5, PAD2, MUX D

Notice MISO at PB02 on PAD0.

Now, if we look at the required value for the DIPO field. The datesheet Table 25-6 says: PAD0 => DIPO=0, but the define has (2). This is either a bug, or it needs an explanation.

bapclenet, 2016-07-05:
Yes it is a little bug: this should be by default:
```
#define SPI_1_DOPO         (1)
#define SPI_1_DIPO         (0)
```

You can open a PR and change it.

keestux, 2016-07-05:
It's not really up to me to create a PR. I don't have this board, so I can't test. I'm just trying to understand the current SPI of SAMD21.

PeterKietzmann, 2016-07-06:
#5609 was merged

closed 2016-07-06 by PeterKietzmann