RIOT-OS/RIOT #5595: MAC layer in complex L2 protocols (Request for comments)

jia200x, 2016-07-04:
Hi all.

We were discussing with @kYc0o about the proper way to implement complex L2 protocols in RIOT.
I have been working mainly with the L2 layer of RIOT while porting OpenThread, implementing PPP, etc. And sometimes I have had some problems trying to handle L2. 

For example, PPP is a stack of protocols in L2 (LCP, Auth, NCP, etc). These subprotocols have to communicate each other via IPC msgs (msg_send_to_self [msg_send_to_self]) and have some timeouts (handled by xtimer_set_msg [xtimer_set_msg] via IPC too). The driver and MAC layer are running in the same thread, but I couldn't find a clean way to extend gnrc_netdev2 to handle this kind of messages (they might go through NETDEV2_MSG_TYPE_EVENT, but in the end they go to the driver's isr which only receives a pointer to the driver itself [gnrc_netdev2 loop]).

So, I didn't use gnrc_netdev2 and made a small L2 layer that handles this (see [gnrc_ppp loop]).

Also, sometimes PPP requires to send specific command to the driver (dial up, reset, etc). I did the same structure as netdev2, but added some of these functions (see [pppdev_driver_t]). Also, I slightly modified the isr function (renamed to driver_ev that also has an extra argument for giving an event).

I think would be nice to reuse the existing structures for not reinventing the wheel, so:

- What do you think is the right way to handle these kind of L2 protocols in RIOT trying to maximize code reuse?
- Is there something like a generic MAC layer that can handle these kind of L2 protocols? If not, does it make sense to have one?
- How should actions from L2 to the driver be treated? (like netdev2's get and set, but for actions like hang out, dial up, etc).

Best regards
 

[xtimer_set_msg]: http://riot-os.org/api/group__sys__xtimer.html#ga1f882b73e60fa5854b4e5e843dcff5f6
[msg_send_to_self]: http://riot-os.org/api/group__core__msg.html#ga8711110548c90fe5ca82215e89bb83dd
[gnrc_netdev2 loop]: https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/link_layer/netdev2/gnrc_netdev2.c#L141
[gnrc_ppp loop]: https://github.com/RIOT-OS/RIOT/pull/5470/files#diff-ed6a8c8bc3c7ce6f932da3bf4ab7e173R338
[ppp_driver_t]: https://github.com/RIOT-OS/RIOT/pull/5470/files#diff-f2ab1d3d0bb4214f88e8a2ad9492d52dR108

miri64, 2016-07-04:
> What do you think is the right way to handle these kind of L2 protocols in RIOT trying to maximize code reuse?

I don't quite understand why you didn't implement PPP on top of netdev2, like other MAC protocols would also be implemented (see e.g. among others #3730) or 6LoWPAN already is? This might not make it available to other stacks other than GNRC, but e.g. lwIP comes with its own PPP implementation anyway (and they do it on top of their network device layer [not existant and expected to be provided by the OS ;-)], too). I don't quite understand what the problem with NETDEV2_MSG_TYPE_EVENT here is? That's for interaction with the actual device, not the MAC layer... Maybe I'm missing (and missed in the past) something, so please help me see.

> Is there something like a generic MAC layer that can handle these kind of L2 protocols? If not, does it make sense to have one?

At least up until this point I used MAC and L2 layer pretty much interchangeable (which is also why it is this way in GNRC), so maybe this is my problem of understanding your problem? At least design-wise there is nothing from keeping you stacking up as many L2 protocols below the IP-layer as you like, but e.g. PPPoE I would immagine like this:

```
+-------------+
|      IP     |
+-------------+
       ↕ netapi
+-------------+
|     PPP     |
+-------------+ ↕ netdev
|    Driver   |
+-------------+
```

This would require to "free" gnrc_netdev2_eth.c from `gnrc_netdev2` as its own submodule which should be done anyways, so I would advocate for this solution

Alternatively connect PPP over netapi to the MAC:

```
+-------------+
|      IP     |
+-------------+
       ↕ netapi
+-------------+
|     PPP     |
+-------------+
       ↕ netapi
+-------------+
| gnrc_netdev2|
+-------------+ ↕ netdev
|    Driver   |
+-------------+
```

I don't know about any timing requirements in PPP so if there are any this solution should be avoided in favor for the first.

> How should actions from L2 to the driver be treated? (like netdev2's get and set, but for actions like hang out, dial up, etc).

E.g. (just brainstorming)

For the first of the above solutions
```

gnrc_ppp_command_t hang_out = {
     GNRC_PPP_COMMAND_HANG_OUT,
     { params .... },
};
gnrc_ppp_command_t dial_up = {
     GNRC_PPP_COMMAND_HANG_OUT,
     { params .... },
}

netdev->driver->set(netdev, NETOPT_COMMAND, hang_out, sizeof(hang_out));
netdev->driver->set(netdev, NETOPT_COMMAND, dial_up, sizeof(dial_up));
```

Alternatively for the second (using netapi, and thus IPC => slower)
```C
gnrc_netapi_set(iface, NETOPT_COMMAND, hang_out, sizeof(hang_out));
gnrc_netapi_set(iface, NETOPT_COMMAND, dial_up, sizeof(dial_up));
```

jia200x, 2016-07-04:
Hi Martine.

> I don't quite understand why you didn't implement PPP on top of netdev2, like other MAC protocols would also be implemented

I did this for 2 reasons:

1. Handling of actions. 
2. Passing arguments from ISR to the isr handler (AT command finished, data received, AT timeout received, etc). 

As you said, would be possible to handle 1 as "set" options (just as brainstorming, does it make sense to have in a future release something like an "action" hook?). The second reason has some solutions too (e.g state variables).

Actually if these 2 things are solved, it's very easy to adapt the pppdev to netdev2 (almost straightforward, as they do almost the same).

>  I don't quite understand what the problem with NETDEV2_MSG_TYPE_EVENT here is? That's for interaction with the actual device, not the MAC layer... Maybe I'm missing (and missed in the past) something, so please help me see.

Please feel free to comment/discuss about this part:

Under the supposition of the same thread for MAC and driver (so, same msg loop), there are messages from device/driver and from the L2. 
From the device/driver part, I'm auto-sending messages like AT command finished, data received, etc (this suits ok with NETDEV2_MSG_TYPE_EVENT). 

From L2 part, I have defined some structs representing subprotocols that are stacked. There are some messages between them (e.g LCP telling NCP that the link is ready, xtimer msg triggered by NCP indicating a timeout of network protocol negotiation, etc). Basically these functions are abstracted as a MSG_TYPE_EVENT and an uint32_t that contains target and event. 

In the beginning I wanted to use gnrc_netdev2 for this, but couldn't find a way to pass these kind of events. How could I reuse the same gnrc_netdev2 loop to make this possible? If this is solved, then there's no need to have a different event loop.

If something was not clear, don't hesitate to ask again.

Cheers!

miri64, 2016-07-06:
Hi Jose,

Let me process the rest first, but quick answer to your last question:

> In the beginning I wanted to use gnrc_netdev2 for this, but couldn't find a way to pass these kind of events. How could I reuse the same gnrc_netdev2 loop to make this possible? If this is solved, then there's no need to have a different event loop.

The idea always was to have alternatives to gnrc_netdev2. So you wouldn't use gnrc_netdev2 at all, but implement your own PPP-specific glue-code layer between GNRC and netdev.

miri64, 2016-07-07:
> I did this for 2 reasons:
>
> 1. Handling of actions.
> 2. Passing arguments from ISR to the isr handler (AT command finished, data received, AT timeout received, etc).
>
> As you said, would be possible to [...]

1. you already said: `set` would be the way to go (for asynchronous actions use `set` + `isr`)
2. `isr()` is not supposed to have arguments, since it is a signifier for the device to handle the "aftermath" of a preceding interrupt. What kind of interrupt it is supposed to be determined by the device's internal state so any kind of value/parameters could and should be stored device-internally anyways (you need to store it anyways somewhere to use it as a IPC parameter to get out of ISR).

> Under the supposition of the same thread for MAC and driver (so, same msg loop), there are messages from device/driver and from the L2. 
From the device/driver part, I'm auto-sending messages like AT command finished, data received, etc (this suits ok with NETDEV2_MSG_TYPE_EVENT).

/me nods.

> From L2 part, I have defined some structs representing subprotocols that are stacked. There are some messages between them (e.g LCP telling NCP that the link is ready, xtimer msg triggered by NCP [...]

Why does it needs to be an IPC call? What prevent's you from using `set()`?

miri64, 2016-07-07:
Just an example for using set with a timeout:

```C
switch (msg.type) {
   NCP_TIMEOUT:
        // ... set ctx
        netdev->driver.set(netdev, NETOPT_TIMEOUT, ctx, sizeof(ctx));
        break;
}
```

Wouldn't that work?

miri64, 2016-10-18:
Is this issue resolved with my explanations? (#5470 seems to be in the process of being adapted for that anyway)

jia200x, 2016-10-18:
Yes, it's is solved. I'm working on the adoption.

Cheers!

closed 2016-10-18 by jia200x