RIOT-OS/RIOT #5597: XTIMER initialisation for non 1MHz capable boards

kYc0o, 2016-07-04:
Hi all!

I opened a PR (#5592) to solve an specific issue on the initialisation of XTIMER when the board clock is not compatible with the hardcoded 1MHz. This is the case on the waspmote-pro board, which integrates a 14.7456MHz clock.

Thus, I'd like to keep open this issue until we agree on a solution to this case.

gebart, 2016-07-04:
Closing as a duplicate of #5237, but I agree that the issue needs to be resolved.

kYc0o, 2016-07-04:
Why we don't look for a straightforward solution in the meantime? The PRs addressing this issue go very far, and they're quite old. I don't know if there will be a consensus in the short term and this issue is affecting the overall xtimer functionality at least on the waspmote. I don't know if this issue also avoid the use of the xtimer on other boards, but it seems to me that it's only avoiding to make work some features. Can you explain me more specifically why putting a platform defined ticks to xtimer_init would affect the overall functionality of the xtimer? 

I'm looking forward to help to solve this issue.

kYc0o, 2017-01-27:
Just reopen the issue because it persists, I also changed the name to make the difference between #5237.

closed 2019-09-10 by stale[bot]