RIOT-OS/RIOT #5599: UART1 support for msp430f1611 and SPI

kYc0o, 2016-07-05:
The SPI peripheral on msp430f1611 is configured to be used with the UART0 peripheral, so we can't have the shell output and the SPI working at the same time.

Therefore, we need to add support for UART1, in order to share it with the SPI and enable the use of UART0 and SPI together.

Anyone having the same issue or willing to add support to this?

thomaseichinger, 2016-07-05:
@kYc0o @OlegHahm I don't know the wsn430 boards, the question is if USART1 is actually connected? Or do you use a UART adapter with the board?

kYc0o, 2016-07-05:
Oh, I assigned you because it seems that you committed the current configuration of the SPI on this board.

USART1 is not used AFAIK, the issue here is that we need to add it because it shares buffers and some configurations with the SPI port. I don't have access to the actual boards but using IoT-Lab ones on the testbed.

kYc0o, 2016-07-05:
And what about configuring the USART1 as SPI? I think it should work since in the IoT-Lab repositories there is a port for contiki which uses UART1 for STDIO and SPI for radio...

thomaseichinger, 2016-07-05:
@kYc0o please see #5606.

kYc0o, 2016-07-26:
Postponed as the PR solving this was postponed too.

closed 2019-09-10 by stale[bot]