RIOT-OS/RIOT #5603: atmega boards second UART issue

kYc0o, 2016-07-05:
After a while trying to debug a second UART on atmega based boards I didn't succeed to make work both UART1 and UART2 (also UART3 and UART4 on CPUs including them) with other devices, although they work when bridging TX and RX on the board itself.

This affects, for now, arduino-mega2560 and waspmote-pro.

OlegHahm, 2017-01-15:
Does this still exist?

kYc0o, 2017-01-16:
Yes.

kYc0o, 2017-01-16:
Although the problem was in the test itself, I still cannot get working it with XBee... Maybe I just should change the name of the issue or open a new one with the specific problem.

PeterKietzmann, 2017-01-16:
Did you also try with a second UART/USB converter?

kYc0o, 2017-01-16:
Yes, with a second converter and the periph_uart test this works correctly, my issue is actually with XBee... I'll close this and open another with the specific XBee issue.

closed 2017-01-16 by kYc0o