RIOT-OS/RIOT #5611: tests/zep: Application attempts invalid registration of GNRC_NETTYPE_NETIF

aeneby, 2016-07-06:
Offending line [here](https://github.com/RIOT-OS/RIOT/blob/2cbc8fb8f723f1ad6720505484e2e2a00bbb894c/tests/zep/main.c#L46) - `GNRC_NETTYPE_NETIF` is an [invalid type](http://riot-os.org/api/group__net__gnrc__netreg.html#gafb6a9d0b840aae07c3c1617932835ef1) for registration via `gnrc_netreg_register()`.

miri64, 2016-07-07:
My current opinion to the ZEP module is "burn and toss it away", since last time I worked on it, every bug-fix opened a whole new can of worms, so if you like to fix it, please do, otherwise I would vote to delete the module altogether and maybe establish an altogether more stable solution.

aeneby, 2016-07-07:
I have zero familiarity with ZEP and no idea what a fix would/should look like; I just happened to stumble across this and thought I'd report it. So feel free to close this issue if you feel that removal is the preferred option.

miri64, 2016-10-18:
I guess fixing ZEP is another thing I need to do then :P

closed 2016-10-18 by miri64