RIOT-OS/RIOT #5647: dist/tools/uhcpd: allow running multiple instance

cladmi, 2016-07-15:
I want to run multiple instances of uhcpd on the same host at the same time.

My goal is to make a 'start_network.sh' equivalent script work on IoT-LAB frontend to multiple users.

For that it should be possible to run multiple instances of `start_network.sh` with different tap interface and notes. The current problem is `uhcpd`, as it is locking the udp socket port.

One other constraint, at best, only `ip tuntap and ip addr/route` commands should be run as root.
I have a working custom implementation on start_network.sh for this but which can run only be run once at a time.

I found this solution:
```
res = setsockopt(sock, SOL_SOCKET, SO_BINDTODEVICE, argv[1], IF_NAMESIZE);
if (res == -1) {
    perror("setsockopt failed\n");
    exit(1);
}
```
Which worked, but could unfortunately only be run as root.

I tried to find other "out of the box" stackoverflow solutions but failed and I do not really understand what I am doing or what I should do to have the multicast ipv6 socket binded to a link-local address or an interface.

So any help would be welcome.

miri64, 2016-10-18:
@kaspar030?

kaspar030, 2016-10-18:
Hm, maybe we should uhcpd to sock! ;)

OlegHahm, 2017-01-15:
Should we?

cladmi, 2018-01-05:
Issue was closed by #6011

closed 2018-01-05 by cladmi