RIOT-OS/RIOT #5657: 6LoWPAN: CC2538: kernel panic using 'radvd'

LucaZulberti, 2016-07-18:
Hi all,

Recently I've seen the RF support for the cc2538 has been added to the master branch, so I decided to add that types of hardware in my sensor network.
I've flashed the gnrc_networking example on two board (samr21-xpro, cc2538dk) and using radvd on my linux 6lowpan router they build their own global address with a specific prefix.

I turn on the cc2538 node, ifconfig output:
```
2016-07-18 15:03:27,962 - INFO # Iface  7   HWaddr: fa:dd  Channel: 26  Page: 0  NID: 0x23
2016-07-18 15:03:27,967 - INFO #            Long HWaddr: 00:12:4b:00:04:33:ee:f8 
2016-07-18 15:03:27,970 - INFO #            TX-Power: 3dBm  State: IDLE 
2016-07-18 15:03:27,975 - INFO #            AUTOACK  MTU:1280  HL:255  6LO  RTR  RTR_ADV  IPHC  
2016-07-18 15:03:27,981 - INFO #            Source address length: 8
2016-07-18 15:03:27,982 - INFO #            Link type: wireless
2016-07-18 15:03:27,987 - INFO #            inet6 addr: ff02::1/128  scope: local [multicast]
2016-07-18 15:03:27,993 - INFO #            inet6 addr: fe80::212:4b00:433:eef8/64  scope: local
2016-07-18 15:03:27,999 - INFO #            inet6 addr: ff02::1:ff33:eef8/128  scope: local [multicast]
2016-07-18 15:03:28,004 - INFO #            inet6 addr: ff02::1a/128  scope: local [multicast]
2016-07-18 15:03:28,010 - INFO #            inet6 addr: fdc5:122a:5e88:0:212:4b00:433:eef8/64  scope: global
2016-07-18 15:03:28,016 - INFO #            inet6 addr: ff02::2/128  scope: local [multicast]
2016-07-18 15:03:28,017 - INFO #            
2016-07-18 15:03:28,020 - INFO #            Statistics for Layer 2
2016-07-18 15:03:28,022 - INFO #             RX packets 0  bytes 0
2016-07-18 15:03:28,027 - INFO #             TX packets 5 (Multicast: 3)  bytes 0
2016-07-18 15:03:28,030 - INFO #             TX succeeded 0 errors 0
2016-07-18 15:03:28,033 - INFO #            Statistics for IPv6
2016-07-18 15:03:28,036 - INFO #             RX packets 1  bytes 128
2016-07-18 15:03:28,040 - INFO #             TX packets 5 (Multicast: 3)  bytes 370
2016-07-18 15:03:28,043 - INFO #             TX succeeded 5 errors 0
```
It has got the global ipv6 address with the correct prefix.
Then I turn on the samr21-xpro board and this is the output of the cc2538:
```
2016-07-18 15:06:59,544 - INFO #  0x2052c1
2016-07-18 15:06:59,546 - INFO # *** RIOT kernel panic:
2016-07-18 15:06:59,547 - INFO # FAILED ASSERTION.
2016-07-18 15:06:59,548 - INFO # 
2016-07-18 15:06:59,555 - INFO # 	pid | name                 | state    Q | pri | stack ( used) | base       | current    
2016-07-18 15:06:59,563 - INFO # 	  - | isr_stack            | -        - |   - |   512 (  180) | 0x20000000 | 0x200001c8
2016-07-18 15:06:59,570 - INFO # 	  1 | idle                 | pending  Q |  15 |   256 (  128) | 0x2000041c | 0x2000049c 
2016-07-18 15:06:59,578 - INFO # 	  2 | main                 | bl mutex _ |   7 |  1536 (  740) | 0x2000051c | 0x2000099c 
2016-07-18 15:06:59,586 - INFO # 	  3 | pktdump              | bl rx    _ |   6 |  1536 (  240) | 0x2000320c | 0x2000371c 
2016-07-18 15:06:59,594 - INFO # 	  4 | 6lo                  | running  Q |   3 |  1024 (  608) | 0x20003ddc | 0x20003f7c 
2016-07-18 15:06:59,602 - INFO # 	  5 | ipv6                 | bl rx    _ |   4 |  1024 (  636) | 0x20001130 | 0x20001434 
2016-07-18 15:06:59,615 - INFO # 	  6 | udp                  | bl rx    _ |   5 |  1024 (  264) | 0x200045e4 | 0x200048dc 
2016-07-18 15:06:59,621 - INFO # 	  7 | cc2538_rf            | bl mutex _ |   3 |  1024 (  332) | 0x20000ca0 | 0x20000f54 
2016-07-18 15:06:59,628 - INFO # 	  8 | RPL                  | bl rx    _ |   5 |  1024 (  216) | 0x200039c4 | 0x20003d04
2016-07-18 15:06:59,631 - INFO # 	    | SUM                  |            |     |  8960 ( 3344)
2016-07-18 15:06:59,634 - INFO # 
2016-07-18 15:06:59,634 - INFO # *** halted.
2016-07-18 15:06:59,635 - INFO # 
2016-07-18 15:06:59,637 - INFO # Stack pointer corrupted, reset to top of stack
2016-07-18 15:06:59,638 - INFO # Inside isr -13
```
The isr_stack is clearly corrupted, it has 180 used bytes but the pointer is at the 456 (0x1c8).
This bug happens only with radvd activated. This is the radvd.conf file:
```
interface lowpan0 
{
        IgnoreIfMissing on;
        AdvSendAdvert on;
        AdvReachableTime 1000;
        AdvSourceLLAddress on;
        UnicastOnly on;
        AdvCurHopLimit 255;

        prefix fdc5:122a:5e88::/64
        {
                AdvOnLink off;
                AdvAutonomous on;
                AdvRouterAddr on;
        };

        abro fe80::506:504:302:100 # linux ipv6 address
        {
                AdvVersionLow 10;
                AdvVersionHigh 2;
                AdvValidLifeTime 2;
        };
};
```
Do you have any suggestions? There could be a bug on the new RF driver?

Thanks a lot for your work,

Luca.

miri64, 2016-07-18:
Looks more like a failed assertion. The big "ASSERTION FAILED" could be a hint for that ;-). Can you check with `addr2line` or `objdump` where the assertion is failing, please?

LucaZulberti, 2016-07-18:
The kernel panic reports this address: 0x2052c1, right?
```
2016-07-18 15:06:59,544 - INFO #  0x2052c1
2016-07-18 15:06:59,546 - INFO # *** RIOT kernel panic:
2016-07-18 15:06:59,547 - INFO # FAILED ASSERTION.
```

I've give this command ('hce-iot' is a copy of the cc2538 folder with some personal defines):
> arm-none-eabi-objdump -DS bin/hce-iot/gnrc_networking.elf

And searching for '2052c1' I've found only this (match at label '188:')
``` 
 * @param x 32-bit integer
 * @return 32-bit integer
 */
static uint32_t ini_func1(uint32_t x)
{
    return (x ^ (x >> 27)) * UINT32_C(1664525);
     144:       0400001d        streq   r0, [r0], #-29  ; 0xffffffe3
     148:       569a2605        ldrpl   r2, [sl], r5, lsl #12
    }
    for (; j < count; j++) {
        r = ini_func1(st[i % size]
                      ^ st[(i + mid) % size]
                      ^ st[(i + size - 1) % size]);
        st[(i + mid) % size] += r;
     14c:       27030001        strcs   r0, [r3, -r1]
        r += i;
        st[(i + mid + lag) % size] += r;
     150:       1f9e0729        svcne   0x009e0729
     154:       05040000        streq   r0, [r4, #-0]
     158:       0031fe36        eorseq  pc, r1, r6, lsr lr      ; <UNPREDICTABLE>
     15c:       2a420300        bcs     1080d64 <cca+0xe00d90>
        st[i % size] = r;
        i = (i + 1) % size;
    }
    for (; j < count; j++) {
        r = ini_func1(st[i % size]
                      ^ st[(i + mid) % size]
     160:       001fae07        andseq  sl, pc, r7, lsl #28
                      ^ st[(i + size - 1) % size]);
        st[(i + mid) % size] += r;
     164:       c4070400        strgt   r0, [r7], #-1024        ; 0xfffffc00
        r += i;
        st[(i + mid + lag) % size] += r;
     168:       0400001f        streq   r0, [r0], #-31  ; 0xffffffe1
     16c:       052b1d03        streq   r1, [fp, #-3331]!       ; 0xfffff2fd
        r += init_key[j] + i;
        st[(i + mid + lag) % size] += r;
        st[i % size] = r;
        i = (i + 1) % size;
    }
    for (; j < count; j++) {
     170:       00cf0e08        sbceq   r0, pc, r8, lsl #28
        r = ini_func1(st[i % size]
                      ^ st[(i + mid) % size]
                      ^ st[(i + size - 1) % size]);
        st[(i + mid) % size] += r;
        r += i;
        st[(i + mid + lag) % size] += r;
     174:       240a0300        strcs   r0, [sl], #-768 ; 0xfffffd00
        st[i % size] = r;
     178:       211c0704        tstcs   ip, r4, lsl #14
        i = (i + 1) % size;
     17c:       10030000        andne   r0, r3, r0
    }
    for (j = 0; j < size; j++) {
        r = ini_func2(st[i % size]
     180:       1d020704        stcne   7, cr0, [r2, #-16]
     184:       03040000        movweq  r0, #16384      ; 0x4000
     188:       02052c14        andeq   r2, r5, #20, 24 ; 0x1400
     18c:       0000edbe                        ; <UNDEFINED> instruction: 0x0000edbe
     190:       2d160304        ldccs   3, cr0, [r6, #-16]
     194:       00213207        eoreq   r3, r1, r7, lsl #4
                      + st[(i + mid) % size]
     198:       48070400        stmdami r7, {sl}
                      + st[(i + size - 1) % size]);
     19c:       04000021        streq   r0, [r0], #-33  ; 0xffffffdf
        st[i % size] = r;
        i = (i + 1) % size;
    }
    for (j = 0; j < size; j++) {
        r = ini_func2(st[i % size]
                      + st[(i + mid) % size]
     1a0:       072e1e03        streq   r1, [lr, -r3, lsl #28]!
                      + st[(i + size - 1) % size]);
     1a4:       00002171        andeq   r2, r0, r1, ror r1
     1a8:       07041103        streq   r1, [r4, -r3, lsl #2]
        st[i % size] = r;
        i = (i + 1) % size;
    }
    for (j = 0; j < size; j++) {
        r = ini_func2(st[i % size]
                      + st[(i + mid) % size]
     1ac:       00001d02        andeq   r1, r0, r2, lsl #26
     1b0:       01b60304                        ; <UNDEFINED> instruction: 0x01b60304
                      + st[(i + size - 1) % size]);
     1b4:       0304042f        movweq  r0, #17455      ; 0x442f
     1b8:       02050b1f        andeq   r0, r5, #31744  ; 0x7c00
     1bc:       000119db        ldrdeq  r1, [r1], -fp
     1c0:       07300903        ldreq   r0, [r0, -r3, lsl #18]!
        st[i % size] = r;
        i = (i + 1) % size;
    }
    for (j = 0; j < size; j++) {
        r = ini_func2(st[i % size]
                      + st[(i + mid) % size]
     1c4:       00002187        andeq   r2, r0, r7, lsl #3
     1c8:       07040f03        streq   r0, [r4, -r3, lsl #30]
        st[(i + mid + lag) % size] += r;
        st[i % size] = r;
        i = (i + 1) % size;
    }
    for (j = 0; j < size; j++) {
        r = ini_func2(st[i % size]
     1cc:       00001d02        andeq   r1, r0, r2, lsl #26
 * @param x 32-bit integer
 * @return 32-bit integer
 */
```

miri64, 2016-07-18:
Yes, but we seem to have the same issue as on iotlab-m3. Please recompile without optimization (`CFLAGS_OPT=-O0`) and try again (at least that is what I remember how to resolve this, but I think @OlegHahm might be the more helping hand here).

LucaZulberti, 2016-07-18:
I've searched for '2052c', now is better:
```
static void _release_error_locked(gnrc_pktsnip_t *pkt, uint32_t err)
{
    while (pkt) {
        gnrc_pktsnip_t *tmp;
        assert(_pktbuf_contains(pkt));
  2052b2:       f5b3 5fc0       cmp.w   r3, #6144       ; 0x1800
  2052b6:       d303            bcc.n   2052c0 <_release_error_locked.isra.1+0x18>
  2052b8:       2003            movs    r0, #3
  2052ba:       490c            ldr     r1, [pc, #48]   ; (2052ec <_release_error_locked.isra.1+0x44>)
  2052bc:       f7fa ff64       bl      200188 <core_panic>
        tmp = pkt->next;
        if (pkt->users == 1) {
  2052c0:       e894 0028       ldmia.w r4, {r3, r5}
  2052c4:       2b01            cmp     r3, #1
  2052c6:       d10a            bne.n   2052de <_release_error_locked.isra.1+0x36>
            pkt->users = 0; /* not necessary but to be on the safe side */
  2052c8:       2300            movs    r3, #0
            _pktbuf_free(pkt->data, pkt->size);
  2052ca:       68a0            ldr     r0, [r4, #8]
  2052cc:       68e1            ldr     r1, [r4, #12]
    while (pkt) {
        gnrc_pktsnip_t *tmp;
        assert(_pktbuf_contains(pkt));
        tmp = pkt->next;
        if (pkt->users == 1) {
            pkt->users = 0; /* not necessary but to be on the safe side */
  2052ce:       6023            str     r3, [r4, #0]
            _pktbuf_free(pkt->data, pkt->size);
  2052d0:       f7ff ff80       bl      2051d4 <_pktbuf_free>
            _pktbuf_free(pkt, sizeof(gnrc_pktsnip_t));
  2052d4:       4620            mov     r0, r4
  2052d6:       2114            movs    r1, #20
  2052d8:       f7ff ff7c       bl      2051d4 <_pktbuf_free>
  2052dc:       e001            b.n     2052e2 <_release_error_locked.isra.1+0x3a>
        }
        else {
            pkt->users--;
  2052de:       3b01            subs    r3, #1
  2052e0:       6023            str     r3, [r4, #0]
pkt = pkt->next;
    }
    mutex_unlock(&_mutex);
}
```

Now i'll try to recompile without optimization.

LucaZulberti, 2016-07-18:
Compiling without optimization doesn't resolve the error:
```
2016-07-18 16:32:28,110 - INFO # > 0x2002b1
2016-07-18 16:32:28,112 - INFO # *** RIOT kernel panic:
2016-07-18 16:32:28,114 - INFO # FAILED ASSERTION.
```

At the 0x2002b1 address is located the core_panic function:
```
/* WARNING: this function NEVER returns! */
NORETURN void core_panic(core_panic_t crash_code, const char *message)
{
  20028c:       b082b580        addlt   fp, r2, r0, lsl #11
  200290:       af00            add     r7, sp, #0
  200292:       4603            mov     r3, r0
  200294:       6039            str     r1, [r7, #0]
  200296:       71fb            strb    r3, [r7, #7]
    (void) crash_code;

    if (crashed == 0) {
  200298:       4b0d            ldr     r3, [pc, #52]   ; (2002d0 <core_panic+0x44>)
  20029a:       681b            ldr     r3, [r3, #0]
  20029c:       d1132b00        tstle   r3, r0, lsl #22
        /* print panic message to console (if possible) */
        crashed = 1;
  2002a0:       4b0b            ldr     r3, [pc, #44]   ; (2002d0 <core_panic+0x44>)
  2002a2:       601a2201        andsvs  r2, sl, r1, lsl #4
#ifndef NDEBUG
        if (crash_code == PANIC_ASSERT_FAIL) {
  2002a6:       79fb            ldrb    r3, [r7, #7]
  2002a8:       2b03            cmp     r3, #3
  2002aa:       d101            bne.n   2002b0 <core_panic+0x24>
            cpu_print_last_instruction();
  2002ac:       f7ff ffe0       bl      200270 <cpu_print_last_instruction>
        }
#endif
        LOG_ERROR("*** RIOT kernel panic:\n%s\n\n", message);
  2002b0:       4808            ldr     r0, [pc, #32]   ; (2002d4 <core_panic+0x48>)
  2002b2:       6839            ldr     r1, [r7, #0]
  2002b4:       f01c f968       bl      21c588 <iprintf>
#ifdef DEVELHELP
#ifdef MODULE_PS
        ps();
  2002b8:       f000 fc3a       bl      200b30 <ps>
        LOG_ERROR("\n");
  2002bc:       4806            ldr     r0, [pc, #24]   ; (2002d8 <core_panic+0x4c>)
  2002be:       f963f01c                        ; <UNDEFINED> instruction: 0xf963f01c
#endif
        LOG_ERROR("*** halted.\n\n");
  2002c2:       4806            ldr     r0, [pc, #24]   ; (2002dc <core_panic+0x50>)
  2002c4:       f960f01c                        ; <UNDEFINED> instruction: 0xf960f01c
#else
        LOG_ERROR("*** rebooting...\n\n");
#endif
    }
    /* disable watchdog and all possible sources of interrupts */
    irq_disable();
  2002c8:       f000 fa50       bl      20076c <irq_arch_disable>
    panic_arch();
  2002cc:       f000 f952       bl      200574 <panic_arch>
  2002d0:       0300            lsls    r0, r0, #12
  2002d2:       2000            movs    r0, #0
  2002d4:       d6d4            bvs.n   200280 <cpu_print_last_instruction+0x10>
  2002d6:       0021            movs    r1, r4
  2002d8:       d6f0            bvs.n   2002bc <core_panic+0x30>
  2002da:       d6f40021        ldrbtle r0, [r4], r1, lsr #32
  2002de:       0021            movs    r1, r4
```
 How can I find where the core_panic is called?

miri64, 2016-07-18:
I didn't state that it resolves the error, just the "error" with not finding the proper line fo the assertion.

> How can I find where the core_panic is called?

It's called by the `assert()` macro in https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/pktbuf_static/gnrc_pktbuf_static.c#L224, most likely ;-). Seems like the new driver has a `gnrc_pktbuf_release()` to much

miri64, 2016-07-18:
Since this is device dependent I suspect it's more a problem with the device, so I rename the issue and reassign.

miri64, 2016-07-18:
@aeneby `git blame` revealed you wrote most of it. Can you have a look, too?

aeneby, 2016-07-18:
> It's called by the assert() macro in https://github.com/RIOT-OS/RIOT/blob/master/sys/net/gnrc/pktbuf_static/gnrc_pktbuf_static.c#L224, most likely ;-)

This failed assertion rings a bell - see #5535 .

aeneby, 2016-07-18:
@LucaZulberti you might try building with #5536 included to see if this prevents the crash - this would only be a temporary solution however, since I think that PR needs re-doing. It would be interesting to know what type of packet is causing the crash, though. Do you have a sniffer available?

LucaZulberti, 2016-07-19:
Yes I can sniff the packets, but I can't do it before this evening, I'll post the results with #5536 included.
Thank you for your support.

aeneby, 2016-07-21:
@LucaZulberti Did you manage to test with the PR included?

LucaZulberti, 2016-07-21:
I have some problems with the connection to the board, i think my USB-serial adapter is gone...
I'm sorry for the late in response, but i think i can't do nothing at the moment.. I will post the result as soon as i can.

LucaZulberti, 2016-07-21:
I've made this:
```
git fetch origin pull/5536/head:fix-0-payload
git checkout fix-0-payload
git pull
```

Then i recompiled the gnrc_networking example for both board, but the error is still present.
I sniffed the packets with tcpdump on the linux host, this is the 'fatal' sequence:
```
tcpdump -i lowpan0
21:12:35.434282 IP6 fe80::5846:267b:252a:8d4e > ff02::2: ICMP6, router solicitation, length 24
21:12:35.435913 IP6 fe80::506:504:302:100 > fe80::5846:267b:252a:8d4e: ICMP6, router advertisement, length 88
21:12:35.437592 IP6 fe80::5846:267b:252a:8d4e > ff02::1a: ICMP6, RPL, (CLR)DODAG Information Solicitation, length 10
21:12:35.451438 IP6 fdc5:122a:5e88:0:5846:267b:252a:8d4e > fe80::506:504:302:100: ICMP6, neighbor solicitation, who has fe80::506:504:302:100, length 56
21:12:35.451675 IP6 fe80::506:504:302:100 > fdc5:122a:5e88:0:5846:267b:252a:8d4e: ICMP6, neighbor advertisement, tgt is fe80::506:504:302:100, length 24
```

I try to sniff wpan0 too:
```
tcpdump -i wpan0
21:21:37.889868 IEEE 802.15.4 Data packet 
	0x0000:  7b3b 3a02 8500 17b1 0000 0000 0102 5a46  {;:...........ZF
	0x0010:  267b 252a 8d4e 0000 0000 0000            &{%*.N......
21:21:37.891419 IEEE 802.15.4 Data packet 
	0x0000:  7b33 3a86 00a8 f7ff 0007 0800 0003 e800  {3:.............
	0x0010:  0000 0003 0440 6000 0151 8000 0038 4000  .....@`..Q...8@.
	0x0020:  0000 00fd c512 2a5e 8800 0000 0000 0000  ......*^........
	0x0030:  0000 0001 0207 0605 0403 0201 0000 0000  ................
	0x0040:  0000 0023 030a 0002 0002 00fe 8000 0000  ...#............
	0x0050:  0000 0005 0605 0403 0201 00              ...........
21:21:37.893299 IEEE 802.15.4 Data packet 
	0x0000:  7a3b 3a1a 9b00 34e1 0000 0102 0000       z;:...4.......
21:21:37.907762 IEEE 802.15.4 Data packet 
	0x0000:  7b03 3afd c512 2a5e 8800 0058 4626 7b25  {.:...*^...XF&{%
	0x0010:  2a8d 4e87 0037 3900 0000 00fe 8000 0000  *.N..79.........
	0x0020:  0000 0005 0605 0403 0201 0021 0200 0000  ...........!....
	0x0030:  0000 0f5a 4626 7b25 2a8d 4e01 025a 4626  ...ZF&{%*.N..ZF&
	0x0040:  7b25 2a8d 4e00 0000 0000 00              {%*.N......
21:21:37.908064 IEEE 802.15.4 Data packet 
	0x0000:  7b30 3afd c512 2a5e 8800 0058 4626 7b25  {0:...*^...XF&{%
	0x0010:  2a8d 4e88 00fe dfc0 0000 00fe 8000 0000  *.N.............
	0x0020:  0000 0005 0605 0403 0201 00              ...........
21:21:37.912881 IEEE 802.15.4 Data packet 
	0x0000:  7b33 3a87 00a7 3000 0000 00fe 8000 0000  {3:...0.........
	0x0010:  0000 0005 0605 0403 0201 0021 0200 0000  ...........!....
	0x0020:  0000 0f5a 4626 7b25 2a8d 4e01 025a 4626  ...ZF&{%*.N..ZF&
	0x0030:  7b25 2a8d 4e00 0000 0000 00              {%*.N......
21:21:37.917223 IEEE 802.15.4 Data packet 
	0x0000:  7b33 3a88 006e d7c0 0000 00fe 8000 0000  {3:..n..........
	0x0010:  0000 0005 0605 0403 0201 00              ...........
```

aeneby, 2016-07-25:
I'm afraid I can't seem to reproduce this using two cc2538 devices and radvd. Your "fatal" packet sequence suggests that it is a response to a neighbor solicitation which provokes the crash, but this works fine for me even without #5536 included. Are you using the current git master?

I'm also curious as to which radio/driver you are using with radvd?

aeneby, 2016-07-25:
Hmm. Are you able to reproduce this reliably?

It's possible you won't see the offending packet via `tcpdump` on the Linux host, since it may be coming from your other node. You'd have to capture traffic using a separate radio in monitor mode, if possible.

Alternatively, try building from [this commit](https://github.com/aeneby/RIOT/commit/1bef81ea156265a5c2d44cedc00a4ab2d071ac33) and flashing to the cc2538 device - I've added some debug output which should trigger early enough to tell us what was received before it crashes.

LucaZulberti, 2016-07-25:
I haven't got an external sniffer now. This is the output with your commit:
```
2016-07-25 13:32:43,682 - INFO # > cc2538_rf: RX FIFO is 42:61:cc:ab:23:00:f8:ee:33:04:00:4b:12:00:00:01:02:03:04:05:06:07:7b:33:3a:87:00:dd:83:00:00:00:00:fe:80:00:00:00:00:00:00:02:12:4b:00:04:33:ee:f8:01:02:07:06:05:04:03:02:01:00:00:00:00:00:00:00:1e:ea
2016-07-25 13:32:43,811 - INFO # cc2538_rf: RX FIFO is 32:61:cc:ac:23:00:f8:ee:33:04:00:4b:12:00:00:01:02:03:04:05:06:07:7b:33:3a:88:00:5f:d3:c0:00:00:00:fe:80:00:00:00:00:00:00:05:06:05:04:03:02:01:00:1e:ea
2016-07-25 13:32:49,758 - INFO # cc2538_rf: RX FIFO is 42:61:cc:ad:23:00:f8:ee:33:04:00:4b:12:00:00:01:02:03:04:05:06:07:7b:33:3a:87:00:dd:83:00:00:00:00:fe:80:00:00:00:00:00:00:02:12:4b:00:04:33:ee:f8:01:02:07:06:05:04:03:02:01:00:00:00:00:00:00:00:1e:ea
2016-07-25 13:32:54,511 - INFO # cc2538_rf: RX FIFO is 2d:41:d8:00:23:00:ff:ff:4e:8d:2a:25:7b:26:46:5a:7b:3b:3a:02:85:00:17:b1:00:00:00:00:01:02:5a:46:26:7b:25:2a:8d:4e:00:00:00:00:00:00:0a:ea:1f:41:d8:01:23:00:ff:ff:4e:8d:2a:25:7b:26:46:5a:7a:3b:3a:1a:9b:00:34:e1:00:00:01:02:00:00:0a:ea
2016-07-25 13:32:54,554 - INFO # 05:02:00:ae:0a:ea:05:02:00:02:1e:ea:05:02:00:03:1f:ea
2016-07-25 13:32:54,646 - INFO # cc2538_rf: RX FIFO is RFCORE_ASSERT(RFCORE_XREG_RXFIFOCNT > 0) failed at line 81 in rfcore_read_byte()!
2016-07-25 13:32:54,671 - INFO #   RFCORE_SFR_RFERRF = 0x00
2016-07-25 13:32:54,751 - INFO # cc2538_rf: RX FIFO is RFCORE_ASSERT(RFCORE_XREG_RXFIFOCNT > 0) failed at line 81 in rfcore_read_byte()!
2016-07-25 13:32:54,776 - INFO #   RFCORE_SFR_RFERRF = 0x00
2016-07-25 13:32:54,840 - INFO # RFCORE_ASSERT(NOT(flags & RXUNDERF)) failed at line 42 in isr_rfcoreerr()!
2016-07-25 13:32:54,864 - INFO #   RFCORE_SFR_RFERRF = 0x00
2016-07-25 13:32:54,942 - INFO # cc2538_rf: RX FIFO is RFCORE_ASSERT(RFCORE_XREG_RXFIFOCNT > 0) failed at line 81 in rfcore_read_byte()!
2016-07-25 13:32:54,966 - INFO #   RFCORE_SFR_RFERRF = 0x00
2016-07-25 13:32:55,023 - INFO # RFCORE_ASSERT(NOT(flags & RXUNDERF)) failed at line 42 in isr_rfcoreerr()!
2016-07-25 13:32:55,047 - INFO #   RFCORE_SFR_RFERRF = 0x00
2016-07-25 13:32:55,124 - INFO # cc2538_rf: RX FIFO is RFCORE_ASSERT(RFCORE_XREG_RXFIFOCNT > 0) failed at line 81 in rfcore_read_byte()!
2016-07-25 13:32:55,148 - INFO #   RFCORE_SFR_RFERRF = 0x00
2016-07-25 13:32:55,205 - INFO # RFCORE_ASSERT(NOT(flags & RXUNDERF)) failed at line 42 in isr_rfcoreerr()!
2016-07-25 13:32:55,229 - INFO #   RFCORE_SFR_RFERRF = 0x00
2016-07-25 13:32:55,307 - INFO # cc2538_rf: RX FIFO is RFCORE_ASSERT(RFCORE_XREG_RXFIFOCNT > 0) failed at line 81 in rfcore_read_byte()!
2016-07-25 13:32:55,331 - INFO #   RFCORE_SFR_RFERRF = 0x00
2016-07-25 13:32:55,389 - INFO # RFCORE_ASSERT(NOT(flags & RXUNDERF)) failed at line 42 in isr_rfcoreerr()!
2016-07-25 13:32:55,415 - INFO #   RFCORE_SFR_RFERRF = 0x00
2016-07-25 13:32:55,492 - INFO # cc2538_rf: RX FIFO is RFCORE_ASSERT(RFCORE_XREG_RXFIFOCNT > 0) failed at line 81 in rfcore_read_byte()!
2016-07-25 13:32:55,516 - INFO #   RFCORE_SFR_RFERRF = 0x00
2016-07-25 13:32:55,574 - INFO # RFCORE_ASSERT(NOT(flags & RXUNDERF)) failed at line 42 in isr_rfcoreerr()!
2016-07-25 13:32:55,598 - INFO #   RFCORE_SFR_RFERRF = 0x00
2016-07-25 13:32:55,745 - INFO # cc2538_rf: RX FIFO is 32:61:cc:b1:23:00:f8:ee:33:04:00:4b:12:00:00:01:02:03:04:05:06:07:7b:33:3a:88:00:5f:d3:c0:00:00:00:fe:80:00:00:00:00:00:00:05:06:05:04:03:02:01:00:1e:ea
2016-07-25 13:32:55,910 - INFO # cc2538_rf: RX FIFO is 42:61:cc:b2:23:00:f8:ee:33:04:00:4b:12:00:00:01:02:03:04:05:06:07:7b:33:3a:87:00:dd:83:00:00:00:00:fe:80:00:00:00:00:00:00:02:12:4b:00:04:33:ee:f8:01:02:07:06:05:04:03:02:01:00:00:00:00:00:00:00:1e:ea
2016-07-25 13:32:59,344 - INFO # cc2538_rf: RX FIFO is 05:02:00:b3:19:ea
2016-07-25 13:32:59,355 - INFO # 0x205395
2016-07-25 13:32:59,403 - INFO # *** RIOT kernel paniccc2538_rf: RX FIFO is 05:02:00:04:1e:ea
2016-07-25 13:32:59,422 - INFO #  0FAILED ASSERTION.
2016-07-25 13:32:59,428 - INFO # 
2016-07-25 13:32:59,498 - INFO # 	pid | name                 | state    Q | pri | stack ( used) | base       | current    
2016-07-25 13:32:59,564 - INFO # 	  - | isr_stack            | -        - |   - |   512 (  360) | 0x20000000 | 0x200001c8
2016-07-25 13:32:59,632 - INFO # 	  1 | idle                 | pending  Q |  15 |   256 (  128) | 0x2000041c | 0x2000049c 
2016-07-25 13:32:59,700 - INFO # 	  2 | main                 | bl mutex _ |   7 |  1536 (  740) | 0x2000051c | 0x2000099c 
2016-07-25 13:32:59,767 - INFO # 	  3 | pktdump              | bl rx    _ |   6 |  1536 (  240) | 0x2000320c | 0x2000371c 
2016-07-25 13:32:59,835 - INFO # 	  4 | 6lo                  | running  Q |   3 |  1024 (  572) | 0x20003ddc | 0x20003fc4 
2016-07-25 13:32:59,908 - INFO # 	  5 | ipv6                 | bl rx    _ |   4 |  1024 (  636) | 0x20001130 | 0x20001434 
2016-07-25 13:32:59,976 - INFO # 	  6 | udp                  | bl rx    _ |   5 |  1024 (  264) | 0x200045e4 | 0x200048dc 
2016-07-25 13:33:00,043 - INFO # 	  7 | cc2538_rf            | bl mutex _ |   3 |  1024 (  560) | 0x20000ca0 | 0x20000f54 
2016-07-25 13:33:00,111 - INFO # 	  8 | RPL                  | bl rx    _ |   5 |  1024 (  216) | 0x200039c4 | 0x20003d04 
2016-07-25 13:33:00,160 - INFO # 	    | SUM                  |            |     |  8960 ( 3716)
2016-07-25 13:33:00,166 - INFO # 
2016-07-25 13:33:00,179 - INFO # *** halted.
2016-07-25 13:33:00,185 - INFO # 
2016-07-25 13:33:00,222 - INFO # Stack pointer corrupted, reset to top of stack
2016-07-25 13:33:00,233 - INFO # FSR/FAR:
2016-07-25 13:33:00,251 - INFO #  CFSR: 0x00000000
2016-07-25 13:33:00,269 - INFO #  HFSR: 0x80000000
2016-07-25 13:33:00,286 - INFO #  DFSR: 0x00000002
2016-07-25 13:33:00,304 - INFO #  AFSR: 0x00000000
2016-07-25 13:33:00,313 - INFO # Misc
2016-07-25 13:33:00,332 - INFO # EXC_RET: 0xfffffffd
```

aeneby, 2016-07-25:
@LucaZulberti thanks, this is great. There are some buffer underflows happening there which is a concern, but I think I already see why (a second packet being received while one is already being processed; a situation which isn't handled properly by the driver yet). But this is not the reason for your crash.

The failed assertion is being caused by that final packet `05 02 00 04 1e ea`, which is just garbage (likely from RF interference). I still say this is closely related to #5535 and should not be causing a crash that far up the stack, but the driver is still partly at fault for not checking the validity of the packet before passing it on up to the upper layers.

#5654 has just been merged (ironically enough, just 2 hours after you performed this test). Please re-pull master and see if this helps.

aeneby, 2016-07-25:
Actually, I just realised that this last frame is not garbage - it's a completely valid frame (ACK), including the CRC. So #5654 won't help you there. This is a duplicate of #5535. Sorry for the confusion.

LucaZulberti, 2016-07-25:
Ok, thank you for your support! :-) I have to wait for #5535 right? If so, this issue can be closed in favor of that.

aeneby, 2016-07-25:
Yes, I think @miri64 mentioned she was going to look at it after the 2016.07 release. @thomaseichinger please feel free to close this issue.

mojiuyu, 2016-07-26:
@LucaZulberti : sorry my english is not good.I am also interested in the radio of cc2538 module.Can you show me how to use this module? thank you !

LucaZulberti, 2016-07-26:
@mojiuyu This is out of the scope of this issue.
This radio can be used with the gnrc network stack of RIOT as the others one.
If you want you can send to me an email (zulberti.luca@gmail.com), anyway I'm not an expert of this OS.
I think you should start to read the Wiki and test some example projects.

PeterKietzmann, 2016-07-26:
@LucaZulberti do I get it correctly that #5535 fixes the above described issue?

LucaZulberti, 2016-07-26:
@PeterKietzmann yes, this can be closed in favor of #5535, I will wait and test with the new PR #5694.

PeterKietzmann, 2016-07-26:
Ok so I will close it. Thanks for your report!

closed 2016-07-26 by PeterKietzmann