RIOT-OS/RIOT #5678: [at86rf2xx] failed assertion in _isr

jia200x, 2016-07-22:
Hi.

Sometimes I get [this failed assertion] in the at86rf2xx when I'm working with more than 20 nodes.

I added an [option] for enabling/disabling the receiver of at86rf2xx (required for the OpenThread port, sorry for the indentation (: ). This option is working, but once in a very while I get the assert.

The pending_tx variable is modified [here]. When there's a TRX_END IRQ, the _isr function checks the state of the driver to see if it's the end of a reception or transmission. But for some reason I get an IRQ when there's no pending transmission. I got the same problem once before adding the RX_LISTENING option, so I'm not sure if it's related to that.

Also, I sometimes get a recv len of more than UINT16_MAX (very unusual though).

Anyone have had any of these problems before?

[this failed assertion]: https://github.com/jia200x/RIOT/blob/openthread/drivers/at86rf2xx/at86rf2xx_netdev.c#L637

[option]: https://github.com/jia200x/RIOT/blob/openthread/drivers/at86rf2xx/at86rf2xx_getset.c#L376

[here]: https://github.com/jia200x/RIOT/blob/openthread/drivers/at86rf2xx/at86rf2xx.c#L189

OlegHahm, 2016-07-22:
Could be related to https://github.com/RIOT-OS/RIOT/issues/5486.

jia200x, 2016-07-22:
Could be in fact. It's the same behavior

kYc0o, 2016-07-26:
Since the related issue is postponed, I'll postpone this one too.

jia200x, 2018-01-30:
Haven't seen this issue anymore, even working with a large number of nodes

miri64, 2018-01-30:
So close?

jia200x, 2018-01-31:
yup

closed 2018-01-31 by jia200x