RIOT-OS/RIOT #5683: Error when Booting VM with command: vagrant up

quangkhanh, 2016-07-24:
I'm following the step for preparation (Quick Setup - Using a Virtual Machine) in Tutorials. When I reached the step "Run the [Vagrant RIOT Setup"](https://github.com/RIOT-OS/RIOT/blob/master/dist/tools/vagrant/README.md), I got an error with boot VM.

>$ vagrant up
>Bringing machine 'default' up with 'virtualbox' provider...
>==> default: Checking if box 'boxcutter/ubuntu1510' is up to date...
>==> default: Clearing any previously set forwarded ports...
>==> default: Clearing any previously set network interfaces...
>==> default: Preparing network interfaces based on configuration...
>default: Adapter 1: nat
>==> default: Forwarding ports...
>default: 4242 (guest) => 4242 (host) (adapter 1)
>default: 22 (guest) => 2222 (host) (adapter 1)
>==> default: Running 'pre-boot' VM customizations...
>==> default: Booting VM...
>There was an error while executing `VBoxManage`, a CLI used by Vagrant for controlling VirtualBox. >The command and stderr is shown below.
>
>Command: ["startvm", "0c58c566-a164-4a23-82f9-a9f0c8a007b6", "--type", "headless"]
>
>Stderr: VBoxManage.exe: error: The virtual machine 'RIOT VM' has terminated unexpectedly during startup with exit code 1 (0x1). More details may be available in 'C:\Users\Dell\VirtualBox VMs\Logs\VBoxHardening.log'
>VBoxManage.exe: error: Details: code E_FAIL (0x80004005), component MachineWrap, interface IMachine

I'm using Windows 7 Ultimate 64-bit, VirtualBox 5.0.26 and VirtualBox Extension Pack, Vagrant 1.8.5.
I tried:
- Reinstall the VirtualBox
- Run the VirtualBox as administrator
- Disable the Avast Antivirus
- Uninstall the Avast Antivirus
However, the error is still there.

Could you please give a solution to this problem?
Thanks in advance.

basilfx, 2016-07-25:
I don't use Vagrant, but here are some things to check:

* Does a normal virtual machine start? You can create one and see if it boots. If not, than it is not a RIOT-OS issue.
* Do you have virtualization enabled in your EFI/BIOS?
* Have you also tried this order: remove VirtualBox, remove Avast, reboot, install VirtualBox?
* What does your log file say (C:\Users\Dell\VirtualBox VMs\Logs\VBoxHardening.log)?

quangkhanh, 2016-07-26:
I tried with some other normal virtual machines, and VirtualBox cannot boot them while VMware can. This must be the issue of VirtualBox.

Thank you.

closed 2016-07-26 by quangkhanh