RIOT-OS/RIOT #5684: ieee802154: Intra-pan packet headers are constructed incorrectly

aeneby, 2016-07-24:
When sending intra-pan packets (i.e. packets where the source and destination PANs are identical), the source PAN should be omitted from the header and the "intra-pan" bit set to 1. From the 802.15.4-2003 spec (page 154):

> If the PAN identifiers are identical, the intra-PAN subfield of the frame control field shall be set to 1, and the source PAN identifier shall be omitted from the transmitted frame.

Currently, drivers using RIOT's `ieee802154` implementation are sending out intra-pan packets with both source and destination PANs in the header, and the "intra-pan" bit set to 0, even though the PANs are identical. Linux will actually drop these incorrectly constructed packets.

aeneby, 2016-07-25:
#5685

mojiuyu, 2016-07-25:
dear aeneby : Sorry,my english is not very good.I want to use the radio frequency functions cc2538.Can you show me the example of it?

aeneby, 2016-07-25:
@mojiuyu your question is outside the scope of this issue, please try asking on the RIOT users' mailing list instead.

miri64, 2016-10-18:
Fixed in https://github.com/RIOT-OS/RIOT/pull/5897

closed 2016-10-18 by miri64