RIOT-OS/RIOT #5689: CI: Murdock start with other labels than "Ready for CI build"

kYc0o, 2016-07-25:
I noticed that Murdock starts to build when I set the label "new feature" in any PR. I didn't test with other labels but if you experience something similar please report it.

miri64, 2016-10-14:
Is this still an issue? Did anybody else see this? @kaspar030 e.g.?

miri64, 2016-10-14:
I take it out of the release however this this concerns https://github.com/kaspar030/murdock, not RIOT ;-)

kaspar030, 2016-10-14:
Never experienced this. And @miri64 is right, if this happens the bug should reported in the murdoc repo.

@kYc0o ok to close?

gebart, 2016-10-14:
This happened to me this morning when adding the hacktoberfest label to one
pr which was already green and ready for merging.

kaspar030, 2016-10-15:
> This happened to me this morning when adding the hacktoberfest label to one
> pr which was already green and ready for merging.

Ok, that helped, I found it.

When murdock stops, it currently doesn't save the state of the PR's it
knows. Every change to a PR then makes it reload that state. So after a
label change following a murdock restart, murdock thinks the head commit
has changed and thus rebuilds.

Will fix...

kYc0o, 2016-10-16:
Ok to close when the fix comes :P

OlegHahm, 2017-01-15:
Does this bug still exist?

kaspar030, 2017-01-15:
Nope, fixed.

closed 2017-01-15 by kaspar030