RIOT-OS/RIOT #5693: Building core module for native on raspberry pi fails

josephnoir, 2016-07-26:
Testing #5682 on raspberry pi, I noticed that building the `thread_basic` test for native (gcc version 4.9.2 (Raspbian 4.9.2-10)) fails with:

```
pi@raspberrypi:~/tmp/RIOT/tests/thread_basic $ make
Building application "thread_basic" for "native" with MCU "native".

"make" -C /home/pi/tmp/RIOT/boards/native
"make" -C /home/pi/tmp/RIOT/boards/native/drivers
"make" -C /home/pi/tmp/RIOT/core
/home/pi/tmp/RIOT/core/panic.c: In function ‘core_panic’:
/home/pi/tmp/RIOT/core/panic.c:83:5: error: expected declaration or statement at end of input
     UNREACHABLE();
     ^
/home/pi/tmp/RIOT/Makefile.base:60: recipe for target '/home/pi/tmp/RIOT/tests/thread_basic/bin/native/core/panic.o' failed
make[2]: *** [/home/pi/tmp/RIOT/tests/thread_basic/bin/native/core/panic.o] Error 1
/home/pi/tmp/RIOT/Makefile.base:20: recipe for target 'ALL--/home/pi/tmp/RIOT/core' failed
make[1]: *** [ALL--/home/pi/tmp/RIOT/core] Error 2
/home/pi/tmp/RIOT/tests/thread_basic/../../Makefile.include:260: recipe for target 'all' failed
make: *** [all] Error 2
```

(Requires changing the `#ifndef` in `RIOT/core/panic.c` line 49 to `#ifdef` as noted in the comments.)

gebart, 2016-07-26:
what's your compiler version?

gebart, 2016-07-26:
UNREACHABLE is defined in kernel_defines.h which is included by panic.c. Could you run it again with QUIET=0 and then use the same GCC command but remove the `-o [filename.o]` and add `-E -dD | less` and see what UNREACHABLE is defined to?

josephnoir, 2016-07-26:
Here is the output:

```
$ gcc \
> -DRIOT_FILE_RELATIVE=\"core/panic.c\" \
> -DRIOT_FILE_NOPATH=\"panic.c\" \
> -Werror -Wall -Wextra -pedantic -ffunction-sections -fdata-sections -std=gnu99 -fno-delete-null-pointer-checks -Wstrict-prototypes -Werror=strict-prototypes -Wold-style-definition -Werror=old-style-definition -fno-common -Wall -include '/home/pi/tmp/RIOT/tests/thread_basic/bin/native/riotbuild/riotbuild.h' -DRIOT_VERSION=\"2016.07-devel-500-g8594d5-raspberrypi-LudwigKnuepfer-pr/fix-5680\" -I/home/pi/tmp/RIOT/core/include -I/home/pi/tmp/RIOT/drivers/include -I/home/pi/tmp/RIOT/sys/include -I/home/pi/tmp/RIOT/cpu/native/include -I/home/pi/tmp/RIOT/boards/native/include -DNATIVE_INCLUDES -I/home/pi/tmp/RIOT/boards/native/include/ -I/home/pi/tmp/RIOT/core/include/ -I/home/pi/tmp/RIOT/drivers/include/ -I/home/pi/tmp/RIOT/cpu/native/include -I/home/pi/tmp/RIOT/sys/include -I/home/pi/tmp/RIOT/sys/libc/include  -MD -MP -c /home/pi/tmp/RIOT/core/panic.c -E -dD | less
[...]
#define UNREACHABLE() __builtin_unreachable()
[...]
```

Complete compiler version is: 
```
$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/4.9/lto-wrapper
Target: arm-linux-gnueabihf
Configured with: ../src/configure -v --with-pkgversion='Raspbian 4.9.2-10' --with-bugurl=file:///usr/share/doc/gcc-4.9/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.9 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.9 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-libitm --disable-libquadmath --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.9-armhf/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.9-armhf --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.9-armhf --with-arch-directory=arm --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-multiarch --disable-sjlj-exceptions --with-arch=armv6 --with-fpu=vfp --with-float=hard --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
Thread model: posix
gcc version 4.9.2 (Raspbian 4.9.2-10)
```

gebart, 2016-07-26:
okay, could you paste the whole preprocessed code somewhere? (e.g. https://gist.github.com)

josephnoir, 2016-07-26:
Sure. You mean [this output](https://gist.github.com/josephnoir/4ecaac0c195bdddc4e9e5efcc4c84ea5)?

gebart, 2016-07-26:
yeah, something is messed up on your side. It looks like panic.c ends after the `UNREACHABLE();` line.

gebart, 2016-07-26:
(It's missing the closing brace for the function `core_panic`)

josephnoir, 2016-07-26:
Damn, you are right! Sorry for the fuzz ...

closed 2016-07-26 by josephnoir