RIOT-OS/RIOT #5695: make: RIOT_VERSION not updated if core/kernel_init.c unchanged

aeneby, 2016-07-27:
So if you switch branches and rebuild, `RIOT_VERSION` will still have the old branch name in it unless `core/kernel_init.c` gets rebuilt.

This should be solvable with a `.FORCE` directive in a Makefile somewhere, but since I'm unsure exactly where that should be I'm opening an issue instead of a PR :)

LudwigKnuepfer, 2016-07-27:
I'm not convinced we want to force recompilation of `kernel_init` (=> `core` => everything) but the problem of untracked dependence exists for other environment variables as well. Maybe there is an open issue/past discussion on the mailing list about this already..

gebart, 2016-07-27:
This should be added to the riotbuild header instead, or if it is already there we need to investigate why it is not causing a rebuild.

LudwigKnuepfer, 2016-07-27:
```
$ cat bin/native/riotbuild/riotbuild.h
/* DO NOT edit this file, your changes will be overwritten and won't take any effect! */
/* Generated from CFLAGS: -DDEVELHELP -Werror -Wall -Wextra -pedantic -ffunction-sections -fdata-sections -m32 -fstack-protector-all -DBOARD_NATIVE="native" -DRIOT_BOARD=BOARD_NATIVE -DCPU_NATIVE="native" -DRIOT_CPU=CPU_NATIVE -DMCU_NATIVE="native" -DRIOT_MCU=MCU_NATIVE -std=gnu99 -fno-delete-null-pointer-checks -Wstrict-prototypes -Werror=strict-prototypes -Wold-style-definition -Werror=old-style-definition -fno-common -Wall -DMODULE_AUTO_INIT -DMODULE_BOARD -DMODULE_CORE -DMODULE_CORE_MSG -DMODULE_CPU -DMODULE_NATIVE_DRIVERS -DMODULE_PERIPH -DMODULE_SYS */
#define DEVELHELP 1
#define BOARD_NATIVE "native"
#define RIOT_BOARD BOARD_NATIVE
#define CPU_NATIVE "native"
#define RIOT_CPU CPU_NATIVE
#define MCU_NATIVE "native"
#define RIOT_MCU MCU_NATIVE
#define MODULE_AUTO_INIT 1
#define MODULE_BOARD 1
#define MODULE_CORE 1
#define MODULE_CORE_MSG 1
#define MODULE_CPU 1
#define MODULE_NATIVE_DRIVERS 1
#define MODULE_PERIPH 1
#define MODULE_SYS 1
```

LudwigKnuepfer, 2016-07-27:
Just found out about this new addition ;)

PR #5696 (2016-09-07): make: include RIOT_VERSION in CFLAGS for riotbuild
fixes #5695
commit dc183736 (2016-09-07): make: include RIOT_VERSION in CFLAGS for riotbuild

fixes #5695
closed 2016-09-07 by LudwigKnuepfer