RIOT-OS/RIOT #5697: CPP tests fail on ARM devices.

kYc0o, 2016-07-27:
While running the Task 01 for the release, all the tests for CPP fail, with this output:

```
Building application "cpp11_condition_variable" for "airfy-beacon" with MCU "nrf51".

In file included from /usr/arm-none-eabi/include/c++/4.9.3/random:38:0,
                 from /usr/arm-none-eabi/include/c++/4.9.3/bits/stl_algo.h:66,
                 from /usr/arm-none-eabi/include/c++/4.9.3/algorithm:62,
                 from /home/facosta/git/RIOT/sys/cpp11-compat/include/riot/chrono.hpp:29,
                 from /home/facosta/git/RIOT/sys/cpp11-compat/include/riot/condition_variable.hpp:32,
                 from /home/facosta/git/RIOT/sys/cpp11-compat/condition_variable.cpp:31:
/usr/arm-none-eabi/include/c++/4.9.3/cmath:1120:11: error: '::log2l' has not been declared
   using ::log2l;
           ^
/usr/arm-none-eabi/include/c++/4.9.3/cmath:1124:11: error: '::logbl' has not been declared
   using ::logbl;
           ^
/usr/arm-none-eabi/include/c++/4.9.3/cmath:1146:11: error: '::nexttoward' has not been declared
   using ::nexttoward;
           ^
/usr/arm-none-eabi/include/c++/4.9.3/cmath:1147:11: error: '::nexttowardf' has not been declared
   using ::nexttowardf;
           ^
/usr/arm-none-eabi/include/c++/4.9.3/cmath:1148:11: error: '::nexttowardl' has not been declared
   using ::nexttowardl;
           ^
make[3]: *** [/home/facosta/git/RIOT/tests/cpp11_condition_variable/bin/airfy-beacon/cpp11-compat/condition_variable.o] Error 1
make[2]: *** [ALL--/home/facosta/git/RIOT/sys/cpp11-compat] Error 2
make[1]: *** [ALL--/home/facosta/git/RIOT/sys] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
```

Any ideas? Maybe it's just a needed package that's not installed... 

This is the ubuntu version:

```
Distributor ID:	Ubuntu
Description:	Ubuntu 14.04.1 LTS
Release:	14.04
Codename:	trusty
```

and the compiler:

```
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977]
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

BytesGalore, 2016-07-27:
I cannot reproduce the error.

compiler:
`gcc version 5.3.1 20160307 (release) [ARM/embedded-5-branch revision 234589] (GNU Tools for ARM Embedded Processors)`

kYc0o, 2016-07-27:
OK I confirm that works with gcc 5.2.1, but it does not work on versions 4.8.4 and 4.9.3. Is this behaviour expected?

kYc0o, 2016-07-27:
On the wiki there's no recommended version for the arm-none-eabi-gcc, maybe we should put something about that, since 4.8.4 is carried with Debian Jessie and 4.9.3 with Ubuntu 14.04.

kaspar030, 2016-07-27:
> On the wiki there's no recommended version for the arm-none-eabi-gcc,
> maybe we should put something about that, since 4.8.4 is carried with
> Debian Jessie and 4.9.3 with Ubuntu 14.04.

+1

Maybe find some baseline we try to keep supported, e.g., the latest
Ubuntu LTS?

gebart, 2016-07-27:
See also https://lists.riot-os.org/pipermail/devel/2015-June/002777.html

The problem should go away if you use the toolchain from ARM on launchpad.

kYc0o, 2016-07-27:
OK so we need at least to discourage the use of the embedded toolchain.

kaspar030, 2016-07-27:
Seems to compile fine with xenial's gcc-arm-none-eabi.

gebart, 2016-07-27:
From some googling it seems the problem was with newlib. I don't have time to mess with old toolchains though, just leaving this piece of information here in case anyone stumbles upon this thread later.

I agree with Kaspar, let's use the latest Ubuntu LTS as a baseline. That is 16.04 (xenial) today. Next LTS is 18.04 I think?

BytesGalore, 2016-07-27:
> let's use the latest Ubuntu LTS as a baseline

+1

josephnoir, 2016-07-27:
So, does that resolve the issue?

kYc0o, 2016-07-28:
I'll modify the wiki to indicate which compiler we recommend for ARM-based boards.

kYc0o, 2016-07-28:
Done and close.

closed 2016-07-28 by kYc0o

OlegHahm, 2016-07-31:
Could you share the link, please?

kYc0o, 2016-07-31:
Yes! [here](https://github.com/RIOT-OS/RIOT/wiki/Family%3A-ARM) you can find the mention.