RIOT-OS/RIOT #5699: test/msg_send_receive: fials on samr21-xpro

BytesGalore, 2016-07-27:
Found during release 2016.07 testing https://github.com/RIOT-OS/Release-Specs/issues/24#issuecomment-235584801
The test fails on samr21-xpro board because of a wrong value present in [`msg_resp.content.ptr`](https://github.com/RIOT-OS/RIOT/blob/master/tests/msg_send_receive/main.c#L55) in the last iteration.
Printing the values of `counter` and `msg_resp.content.ptr (resp val)` for each iteration revealed:
```
# main(): This is RIOT! (Version: 2013.08-10643-gb5cb6-user01-LLED-E6410)
# Incremented counters to 1 and 1
# resp val: 1, counter: 1
```
... all ok until the last one:
```
# Incremented counters to 10 and 10
# resp val: 841, counter: 10
# Test failed.
```

kYc0o, 2016-07-27:
any hint @kaspar030 ?

OlegHahm, 2016-07-27:
Try to revert https://github.com/RIOT-OS/RIOT/pull/5444

BytesGalore, 2016-07-27:
@OlegHahm reverting #5444 didn't help

OlegHahm, 2016-07-27:
Have you tested on other platforms?

kaspar030, 2016-07-27:
Looks to me like the threads are sending around pointers to their stacks, and in the last iteration, one of the threads already exited.

Could you try to add ```thread_sleep()``` before the return in thread2()?

BytesGalore, 2016-07-27:
@kaspar030 that did it

BytesGalore, 2016-07-27:
(I mean the test succeeds now)

kaspar030, 2016-07-27:
I'll prepare a PR.

BytesGalore, 2016-07-27:
nice

kaspar030, 2016-07-27:
#5701

PR #5701 (2016-07-27): tests: msg_send_receive: fix "sent ptr goes out of scope" bug
Fixes #5699.
closed 2016-07-27 by kYc0o