RIOT-OS/RIOT #5707: Support for Nucleo f103, f303 and f334 boards broken

aabadie, 2016-07-29:
Due to the change I made in nucleo-common code for #5706 (added saul support), I wanted to verify that it wasn't introducing any regression on other supported Nucleo boards.

It appears that one cannot access anymore to the UART using the USB debug interface on the Nucleo boards listed in the title of this issue.

kaspar030, 2016-07-29:
Tested on nucleo-f334, works fine here with default example.

aabadie, 2016-07-29:
> Tested on nucleo-f334, works fine here with default example.

That's weird, I tested a second time but still no UART. I'm flashing with Ubuntu 16.04. My gcc is
```
arm-none-eabi-gcc (15:4.9.3+svn231177-1) 4.9.3 20150529 (prerelease)
```
and openocd is 0.9.0.

kaspar030, 2016-07-29:
I'm on arch (gcc version 6.1.1, openocd 0.9.0).

Just to clarify - I'm just trying "make term" (e.g., the default UART). Or do you mean sth else?

aabadie, 2016-07-29:
> Or do you mean sth else?

No, I'm trying the same thing with:
```make BOARD=nucleo-f334 flash term```

The tty is `/dev/ttyACM0` at 115200

lebrush, 2016-07-29:
The UART/USB on the nucleo boards does not play nicely with ubuntu (14.04 here). It never worked for me: neither with the f103 nor with the f401 nor the l1. I don't think is a regression :-)

kaspar030, 2016-07-29:
@aabadie anything weird in ```dmesg```?

aabadie, 2016-07-29:
dmesg says:
```
[22172.550425] usb 2-2: new full-speed USB device number 24 using xhci_hcd
[22172.680056] usb 2-2: New USB device found, idVendor=0483, idProduct=374b
[22172.680059] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[22172.680060] usb 2-2: Product: STM32 STLink
[22172.680061] usb 2-2: Manufacturer: STMicroelectronics
[22172.680062] usb 2-2: SerialNumber: 0672FF525056805087103008
[22172.680238] usb 2-2: ep 0x84 - rounding interval to 1024 microframes, ep desc says 2040 microframes
[22172.735310] usb-storage 2-2:1.1: USB Mass Storage device detected
[22172.735477] scsi host14: usb-storage 2-2:1.1
[22172.735774] cdc_acm 2-2:1.2: ttyACM0: USB ACM device
[22173.735075] scsi 14:0:0:0: Direct-Access     MBED     microcontroller  1.0  PQ: 0 ANSI: 2
[22173.735442] sd 14:0:0:0: Attached scsi generic sg1 type 0
[22173.735698] sd 14:0:0:0: [sdb] 176 512-byte logical blocks: (90.1 kB/88.0 KiB)
[22173.735832] sd 14:0:0:0: [sdb] Write Protect is off
[22173.735834] sd 14:0:0:0: [sdb] Mode Sense: 03 00 00 00
[22173.735972] sd 14:0:0:0: [sdb] No Caching mode page found
[22173.735974] sd 14:0:0:0: [sdb] Assuming drive cache: write through
[22173.742249]  sdb:
[22173.743111] sd 14:0:0:0: [sdb] Attached SCSI removable disk
```

lebrush, 2017-01-24:
Nucleo f103 still works with latest upstream/master as expected.

Btw, regarding getting the uart output I flashed the latest fw to the embedded stlink in the nucleo and from that moment I can use it as serial interface as well :+1: 
http://www.st.com/content/st_com/en/products/embedded-software/development-tool-software/stsw-link007.html

aabadie, 2017-01-24:
Thanks for digging this up. @kaspar030 it was working for nucleo-f334 and I can make it work for nucleo-f103 and f303.

We can safely close this one.

closed 2017-01-24 by aabadie