RIOT-OS/RIOT #5717: netdev2: send/recv length arguments should be unsigned

gebart, 2016-08-02:
The length arguments on some of the functions declared in the netdev2 API are signed, which does not make sense and causes warnings in some driver implementations when `-Wsign-compare` is used.

https://github.com/RIOT-OS/RIOT/blob/master/drivers/include/net/netdev2.h#L140
https://github.com/RIOT-OS/RIOT/blob/master/drivers/include/net/netdev2.h#L161

Warning in at86rf2xx:
https://github.com/RIOT-OS/RIOT/blob/master/drivers/at86rf2xx/at86rf2xx_netdev.c#L156
```
riot/drivers/at86rf2xx/at86rf2xx_netdev.c:156:17: error: comparison between signed and unsigned integer expressions [-Werror=sign-compare]
     if (pkt_len > len) {
                 ^
```

miri64, 2016-08-02:
Agreed. See https://github.com/RIOT-OS/RIOT/pull/5495

miri64, 2016-10-14:
Fixed in #5495.

closed 2016-10-14 by miri64