RIOT-OS/RIOT #5723: There are unsatisfied feature requirements: arduino cpp

romainvause, 2016-08-04:
I always have this message when I launch the task-01 tutoriel:

> `sudo BOARD=arduino-uno make all flash term`

Thx in advance.

kYc0o, 2016-08-04:
The arduino-uno board is not supported yet. Let's address this issue when the board will be officially supported.

romainvause, 2016-08-04:
Actually, there is a fork for arduino-uno from @mali

https://github.com/mali/RIOT/tree/73e02ecc42dff8290a44ecd0709bd5345ca0fa63

aabadie, 2016-08-04:
@romainvause, what is the message you have exactly ?
Notice that you normally don't need to build/flash/open a serial term using `sudo`. What system are you using ?

miri64, 2016-08-04:
Yes, but (a) fork != supported by RIOT (that's only the case when merged) and (b) that fork is still under development (see https://github.com/RIOT-OS/RIOT/pull/5451). ;-)

romainvause, 2016-08-04:
@aabadie 
With the command: `sudo BOARD=arduino-uno make all flash term`, here is the fully output:

> There are unsatisfied feature requirements: arduino cpp
>
> EXPECT ERRORS!
>
> Building application "Task01" for "arduino-uno" with MCU "atmega328p".
> 
> /home/romain/RIOT/Tutorials/task-01/_sketches.cpp: In function ‘int main()’:
> /home/romain/RIOT/Tutorials/task-01/_sketches.cpp:6:11: error: ‘setup’ was not declared in this scope
>      setup();
>
> /home/romain/RIOT/Tutorials/task-01/_sketches.cpp:9:14: error: ‘loop’ was not declared in this scope
>          loop();
>
> /home/romain/RIOT/Tutorials/RIOT/Makefile.base:66: recipe for target '/home/romain/RIOT/Tutorials/task-01/bin/arduino-uno/Task01/_sketches.o' failed
> make[1]: *** [/home/romain/RIOT/Tutorials/task-01/bin/arduino-uno/Task01/_sketches.o] Error 1
> /home/romain/RIOT/Tutorials/task-01/../RIOT/Makefile.include:259: recipe for target 'all' failed
> make: *** [all] Error 2
> 

With the command `BOARD=arduino-uno make all flash term`, here is the fully ouput

> /home/romain/RIOT/Tutorials/RIOT/dist/tools/arduino/pre_build.sh: 17: /home/romain/RIOT/Tutorials/RIOT/dist/tools/arduino/pre_build.sh: cannot create /home/romain/RIOT/Tutorials/task-01//_sketches.cpp: Permission denied
> /home/romain/RIOT/Tutorials/RIOT/dist/tools/arduino/pre_build.sh: 24: /home/romain/RIOT/Tutorials/RIOT/dist/tools/arduino/pre_build.sh: cannot create /home/romain/RIOT/Tutorials/task-01//_sketches.cpp: Permission denied
> There are unsatisfied feature requirements: arduino cpp
> 
> 
> EXPECT ERRORS!
> 
> 
> Building application "Task01" for "arduino-uno" with MCU "atmega328p".
> 
> /home/romain/RIOT/Tutorials/task-01/_sketches.cpp: In function ‘int main()’:
> /home/romain/RIOT/Tutorials/task-01/_sketches.cpp:6:11: error: ‘setup’ was not declared in this scope
>      setup();
>
> /home/romain/RIOT/Tutorials/task-01/_sketches.cpp:9:14: error: ‘loop’ was not declared in this scope
>          loop();
>
> /home/romain/RIOT/Tutorials/RIOT/Makefile.base:66: recipe for target '/home/romain/RIOT/Tutorials/task-01/bin/arduino-uno/Task01/_sketches.o' failed
> make[1]: *** [/home/romain/RIOT/Tutorials/task-01/bin/arduino-uno/Task01/_sketches.o] Error 1
> /home/romain/RIOT/Tutorials/task-01/../RIOT/Makefile.include:259: recipe for target 'all' failed
> make: *** [all] Error 2
> 

Thx for your answer :)

romainvause, 2016-08-04:
@miri64 Thanks for this information, I don't use Github a lot..

miri64, 2016-08-04:
Where's this `_skettches.cpp` coming from? There should only be three files:

```
$ ls task-01
main.c  Makefile  README.md
```

romainvause, 2016-08-04:
@miri64 It's directly generated (by idk what), here is the content of `_skettches.cpp`:
```
#include "arduino.hpp"

int main(void)
{
    /* run the Arduino setup */
    setup();
    /* and the event loop */
    while (1) {
        loop();
    }
    /* never reached */
    return 0;
}
```

miri64, 2016-08-04:
Okay if it's automatically generated I don't know why... That's something people with an Arduino should answer ;-)

romainvause, 2016-08-04:
@miri64 Hope that some people will see it !

Thx for messages.

miri64, 2016-08-04:
@mali https://github.com/RIOT-OS/Tutorials

miri64, 2016-08-04:
Oh btw @romainvause: note, that the version downloaded automatically is fixed to https://github.com/RIOT-OS/RIOT/tree/6b768a114510f968acc845b80f5ebf403ccc1aa7, so unless you reset `RIOTBASE` in the Makefile before `make` you will not have @mali's branch in there.

aabadie, 2016-08-05:
The actual port of arduino-uno doesn't provide the required features to enable arduino sketches with this board.
I commented in the related PR #5451, [here](https://github.com/RIOT-OS/RIOT/pull/5451#discussion-diff-73668707). Can you try to apply the change and tell us if it work ?

miri64, 2016-08-08:
Since arduino-uno isn't merged yet I would suggest to discuss all the kinks of that port in https://github.com/RIOT-OS/RIOT/pull/5451 and this issue to be closed.

closed 2016-08-08 by miri64