RIOT-OS/RIOT #5724: Delay in example arduino-hello-world

romainvause, 2016-08-04:
Hey, in this file
https://github.com/RIOT-OS/RIOT/blob/master/examples/arduino_hello-world/hello-world.sketch

You should remove the delay. After doing some tests, it looks like that the serial line is not working because the delay is too big.

I just remove it, and everything was working !

aabadie, 2016-08-05:
@romainvause, I don't think removing the delay is a solution. The provided sketch is a valid arduino code.

On which board did you try ?

aabadie, 2016-08-05:
In the official RIOT code base (current master at 0f73a40), the only boards supported by this feature are stm32f4discovery and arduino-due.
I tried with arduino-due and everything works like a charm. If someone with a stm32f4discovery could try as well, it would be great.
I suspect that you tried with the unofficial arduino-uno port from #5451 as you recently created other issues related to this topic. It would be better to comment in the related PR, e.g #5451.

romainvause, 2016-08-08:
@aabadie Yep, I tried with arduino-uno. You're right, sorry to comment here! Thx

closed 2016-08-08 by miri64