RIOT-OS/RIOT #5726: All Cortex-M platforms broken on current master

haukepetersen, 2016-08-05:
Just wanted to bulid something for the `arduino-due` on the current master, but something broke (actually it broke for all cortex-m based boards):
```
/home/hauke/dev/riot/RIOT/cpu/cortexm_common/vectors_cortexm.c: In function `hard_fault_handler':
/home/hauke/dev/riot/RIOT/cpu/cortexm_common/vectors_cortexm.c:276:9: error: `orig_sp' may be used uninitialized in this function [-Werror=maybe-uninitialized]
         __asm__ volatile (
         ^
/home/hauke/dev/riot/RIOT/cpu/cortexm_common/vectors_cortexm.c:271:15: error: `pc' may be used uninitialized in this function [-Werror=maybe-uninitialized]
         printf("In GDB:\n  set $pc=0x%" PRIx32 "\n  frame 0\n  bt\n", pc);
               ^
cc1: all warnings being treated as errors
```
What happened here?

haukepetersen, 2016-08-05:
oh, forgot to mention, using `gcc (Ubuntu 4.8.4-2ubuntu1~14.04.3) 4.8.4`

kaspar030, 2016-08-05:
I can confirm the issue with ```arm-none-eabi-gcc (4.8.2-14ubuntu1+6) 4.8.2```, arch's toolchain works fine.

5cd91bc784e70052fcd75efab68c91776adf4da8 introduced this.

haukepetersen, 2016-08-05:
So if I see this right, the majority of the boards is broken for a very common toolchain for the last 21 days?! I wonder how this could slip by unnoticed -> is there something we can do to prevent this in the future?

haukepetersen, 2016-08-05:
hm, on the other hand, the toolchain is from 2014q3, it might be valid for an almost 2 year old build of the toolchain to fail?

kaspar030, 2016-08-05:
> I wonder how this could slip by unnoticed

The toolchain you're using - is it the one from PPA?

haukepetersen, 2016-08-05:
just tried different ones. Updated mine to `arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 5.4.1 20160609 (release) [ARM/embedded-5-branch revision 237715]` (from launchpad) -> works fine

Used `arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.8.4 20140725 (release) [ARM/embedded-4_8-branch revision 213147]` before (also from launchpad) -> showed error as posted above

Also tried the shipped `arm-none-eabi-gcc (Ubuntu 4.8.4-2ubuntu1~14.04.3) 4.8.4` -> errored as above

kaspar030, 2016-08-05:
https://github.com/RIOT-OS/RIOT/pull/5727

kYc0o, 2016-08-05:
Very strange since all the compilation tests passed OK. I was actually doing interoperability tests yesterday without any trouble.

What's the firmware you want to build?

kaspar030, 2016-08-05:
Murdock uses a more recent launchpad toolchain...

kaspar030, 2016-08-05:
> I wonder how this could slip by unnoticed -> is there something we can do to prevent this in the future?

We could extend the CI to compile everything with older toolchains, as well.

kYc0o, 2016-08-05:
Actually in #5697 I was using the embedded toolchain for Ubuntu 14.04 (gcc 4.9.3) which throw only c++ errors, so I'm confused which is the oficial version of gcc for this Ubuntu version.

Anyways, the wiki states now that's recommended to use the toolchain from launchpad or the Ubuntu 16.04 so this could prevent people to use 4.8.3.

PR #5727 (2016-10-14): cpu: cortexm_common: fix compile warning
Some gcc complain (wrongly) about possibly uninitialized variables.

Fixes #5726.
closed 2016-10-14 by gebart