RIOT-OS/RIOT #5737: u8g2 riot interface function (RIOT/pkg/u8g2/patches/0002-u8g2-add-riot-os-interface.patch)

olikraus, 2016-08-08:
@basilfx 

affected file:
RIOT/pkg/u8g2/patches/0002-u8g2-add-riot-os-interface.patch

Although not used at the moment in u8g2, but two of the interface messages to u8g2 are implemented wrong in procedure "uint8_t u8x8_gpio_and_delay_riotos"
Correct code is this:
+uint8_t u8x8_gpio_and_delay_riotos(u8x8_t *u8g2, uint8_t msg, uint8_t arg_int, void *arg_ptr)
+{
+    (void) arg_ptr;
+
+    switch (msg) {
+        case U8X8_MSG_GPIO_AND_DELAY_INIT:
+            u8x8_enable_pins(u8g2->pins, u8g2->pins_enabled);
+            break;
+        case U8X8_MSG_DELAY_MILLI:
+            xtimer_usleep(arg_int * 1000);
+            break;
+        case U8X8_MSG_DELAY_10MICRO:
+            xtimer_usleep(arg_int * 10);   // Multiply with arg_int !
+            break;
+        case U8X8_MSG_DELAY_100NANO:
+            xtimer_nanosleep(arg_int * 100);   // Multiply with arg_int !
+            break;

aabadie, 2016-08-08:
Thanks for the report.
Maybe the required change can be added to #5702.

basilfx, 2016-08-08:
I will fix this as soon as I am back from holidays next week. Thanks @olikraus

basilfx, 2016-08-20:
I have fixed this in #5702.

closed 2016-09-11 by OlegHahm