RIOT-OS/RIOT #5738: Making a .c file and .sketch file

romainvause, 2016-08-09:
Hi,

Is it possible to launch a `.c file` and a `.sketch file` at the same time ? One has to run on my computer, and the other one on my arduino.

Because it looks like that only the .c file is executed.

Thx in advance.

PeterKietzmann, 2016-08-12:
Hi r@romainvauce. Could you rephrase your question please? I didn't really understand what you're trying to do and what the relation to RIOT is.

PeterKietzmann, 2016-08-12:
Ah you mean RIOT arduino library support? And what needs to run on your computer? @haukepetersen is the right contact person. He won't be available before Semeptember though.

romainvause, 2016-08-12:
Hi @PeterKietzmann,
Basically I just want to run the MQTT protocol with RIOT.
So just send a program to the arduino (a simple MQTT client), and run the MQTT server on my Linux machine.

If I understood well, you can create a .c file to be executed with RIOT (by doing a `make all term`) and also execute a .sketch file for the arduino with RIOT (by doing a `make all flash term`)

But after severals retries, I can't launch both. It's only the .c file executed into RIOT.

Thx for your answer.

PeterKietzmann, 2016-08-15:
@romainvause probably I still don't get your issue (maybe it's my lack of knowledge about the arduino port). But I'm gonna try to find out, so let me ask some questions and give first hints.
- which MQTT implementation do you intend to use in RIOT?
- you can *either* write a c-program with a .c ending *or* (and this is the special thing) write a .sketch file in arduino-code. I don't know which peripheral features are currently implemented for the latter
- if you write arduino-code your file must have the ending ".sketch" and you need to include the arduino module in the Makefile, like done e.g. [here](https://github.com/RIOT-OS/RIOT/blob/master/examples/arduino_hello-world/Makefile#L10)
- `make all term` builds the project and starts our own terminal program called "pyterm" for serial I/O
- in addition you need to export the board name and to flash the board. So if you are on an arduino-due you should type: `BOARD=arduino-due make all flash` and then `BOARD=arduino-due make term`
- you said that you "want to launch both". So which program should run on which device?
- over which interface do you intend to communicate with your arduino board?

romainvause, 2016-08-17:
@PeterKietzmann 
Hi,
So if I got it, `BOARD=arduino-due make all flash` only send the code to the arduino ? And then 
`BOARD=arduino-due make term` execute the `.c` file ?
Thx a lot, I'm gonna try it !

PeterKietzmann, 2016-08-18:
@romainvause, `make` builds the the code, resulting in a binary (and cpp file as an interim step, if I see it correclty ). `flash` writes the binary to the boards memory  and `term` connects the serial output to a terminal program to see the STDIO. Anyway, the steps you listed should work.

PeterKietzmann, 2016-09-30:
@romainvause can we close this issue?

romainvause, 2016-09-30:
Yes, you can ! Sorry I didn't close it

closed 2016-09-30 by PeterKietzmann