RIOT-OS/RIOT #5748: gnrc: nodes crashing with too small packet buffer

OlegHahm, 2016-08-15:
I just ran into a problem, when I manually configured the packet buffer too small in the `gnrc_networking` example. In fact, I copied the size from the `gnrc_minimal` example (512 bytes). Testing this on the iotlab testbed with ~350 nodes (Neighbor cache size set to 100, FIB size to 350) I ran into several crashes. Nodes were dying with hard faults after some time.

(On the pro side: with the default packet buffer size, I was able to transmit a UDP datagram from the one end of the testbed to the other without problems. :))

miri64, 2016-08-15:
I think I have the same problem at the moment. Will look into that.

miri64, 2016-08-15:
@OlegHahm are you on native or other platform? (native resets `GNRC_PKTBUF_SIZE` to 2048)

OlegHahm, 2016-08-15:
I tested only on iotlab.

miri64, 2016-08-15:
My problem was unrelated (see #5749), but nevertheless: will look into this ASAP.

OlegHahm, 2016-08-16:
I figured that you will probably get more debug information if you revert #5665.

miri64, 2016-08-30:
@OlegHahm its really just a hunch, but can you check if https://github.com/RIOT-OS/RIOT/pull/5694 fixes the problem?

OlegHahm, 2016-10-30:
Just tested on master with same settings and 220 nodes on Lille site: nodes are still crashing.

kYc0o, 2016-11-09:
Maybe this would be solved by #6083 ?

kYc0o, 2016-11-09:
Ups sorry, didn't refresh the page and this was already referenced.

miri64, 2016-12-16:
#6086 sadly did not fix this

kYc0o, 2017-04-26:
Does anyone know if this is still an issue?

kYc0o, 2018-07-20:
So far, I only found issues with the kw2x driver. With at86rf2xx everything works like a charm. Thus, I conclude there's no problem with gnrc.

Ok to close?

miri64, 2018-07-20:
Might be that #9434 fixed this issue as well.

kYc0o, 2018-08-14:
I'll remake the tests to configure the cases you mention and give a more complete feedback.

miri64, 2019-01-29:
@cgundogan can you maybe retest this for the release?

obgm, 2019-03-19:
Not sure if this is related: Testing via tap on native, after some successful UDP transmissions, `_gnrc_netif_thread` always gets stuck. DEBUG says: 
```
gnrc_netif: waiting for incoming messages
gnrc_netif: GNRC_NETDEV_MSG_TYPE_EVENT received
gnrc_netif: event triggered -> 2
gnrc_netif_ethernet: cannot allocate pktsnip.
gnrc_netif: waiting for incoming messages
gnrc_netif: GNRC_NETDEV_MSG_TYPE_EVENT received
gnrc_netif: event triggered -> 2
gnrc_netif_ethernet: cannot allocate pktsnip.

```

Increasing  `GNRC_PKTBUF_SIZE` increases the number of successful transmissions but sooner or later, the same error will occur.

miri64, 2019-05-23:
> Not sure if this is related: Testing via tap on native, after some successful UDP transmissions, `_gnrc_netif_thread` always gets stuck. DEBUG says:
>
> ```
> gnrc_netif: waiting for incoming messages [...]

This hints at a leak in the packet buffer. So it might be interesting which modules you are including and what your scenario it is so we can reproduce it. I don't see the relation to a smaller packet buffer however.

miri64, 2019-09-24:
Ping @obgm?

obgm, 2019-09-24:
Sorry for the delay: I wanted to re-test with my code and up-to-date fork. I will hopefully be able to provide recent data within the next days.

miri64, 2019-09-24:
I'm swamped in work anyways, so I won't be able to fix this immediately, but it's good to keep the issue rolling ;-).

miri64, 2020-07-08:
I was trying to reproduce this issue but I was not able to. Here is an detailed account of what I did:
- [still uploading]

commit 490b907c (2016-12-16): Merge pull request #6086 from miri64/gnrc_pktbuf_static/fix/i5748

gnrc_pktbuf_static: fix #5748
commit 5e236cdc (2016-12-16): gnrc_pktbuf_static: fix #5748
closed 2020-07-08 by miri64