RIOT-OS/RIOT #5752: WIFI Module like CC3000?

kbumsik, 2016-08-15:
Hello.
I am new to RIOT. I have some wifi modules, CC3000 and ATWINC1500. I was looking for references for porting them. 

It looks like there were a CC3000 driver directly uses TI's codes, not through GRNC: https://lists.riot-os.org/pipermail/notifications/2015-March/030723.html . But I cannot see this driver in the master branch's source code.

Was it removed for some reason? Or is there other ways I could user WIFI other than using manufacterer's code?
If this project is planing to integrate wifi I would like to join it.

miri64, 2016-08-15:
Welcome to the RIOT community! Yes, as you can see if you go to that [PR](https://github.com/RIOT-OS/RIOT/pull/2603) it has been closed, since the developer abandoned it. We are definitely planning to integrate WiFi at some point, but at the moment no one is working at this particular driver to my knowledge, so feel free to do so. To be on the safe side I would recommend however to also ask around on the [devel mailinglist(https://lists.riot-os.org/mailman/listinfo/devel) if somebody else is already working on it.

kbumsik, 2016-08-15:
Thank you for a prompt reply.
Just in case I tried to PR for wifi modules, I am worried that these wifi drivers usually offer their own TCP/IP stack. I would like to port these drivers directly first, not integrating GRNC or RIOT's stack. Also these drives may contain their own BSD/GPL-like license too. Could these be reason for rejecting my PR?
I show Coding Conventions and PR Guidelines but still I am not sure about it.

miri64, 2016-08-15:
I reassign to @haukepetersen because I think he has a stronger opinion on that matter.

kbumsik, 2016-08-16:
Thanks. Please understand that unlike 802.15.4 transceivers, Wifi modules designed for MCUs such as CC3000/CC3100/CC3200 and ATWINC1500 have in-built processors. These in-built processors takes over up to TCP/IP layer instead of host MCUs. Although most of them offers BSD-style sockets, it might be impossible to directly control them from Low-level.

OlegHahm, 2016-08-16:
Maybe you want to take a look at https://github.com/RIOT-OS/RIOT/pull/5733.

kbumsik, 2016-08-16:
Oh I didn't looked at that PR in detail but there are actually a lot of things to see for me. Thank you very much!

miri64, 2016-10-18:
I deem this question/issue resolved ;-)

closed 2016-10-18 by miri64

maribu, 2021-04-07:
For the record: Support for ATWINC has been added by @gschorcht in the meantime - in case someone stumbles upon this issue.