RIOT-OS/RIOT #5755: Machine to Machine Communication using RIOT OS

dondonprad, 2016-08-17:
hello..
I have a project, make Machine to Machine Communication using 6LoWPAN protocol
Specification Device
- 2 Arduino Due
- 2 Xbee S1 -
- DHT 11 (temperature and Humidity Sensor)
- RTC (Real Time Clock)
  
i'm create 2 node ( 1 sensor node, 1 sink node), sensor node work sensing temperature, humidity, and sending data to sink node using Xbee S1 

a question, how create coding using RIOT OS and how step create xbee can communication each other?
any suggestion?

Sorry my english so bad :D

PeterKietzmann, 2016-09-30:
Hi @dondonprad. I've never used the Xbee shields but have you already looked at the neworkig examples `examples/gnrc_minimal` (which contains an Xbee configuration) an `examples/gnrc_netowrking` (which is an "extended" networking example including a shell)? You might need to add this module to your Makefile (compare [here](https://github.com/RIOT-OS/RIOT/blob/master/examples/gnrc_minimal/Makefile#L12)). Regarding DHT11 and RTC, please have a look at `tests/driver_dht` and `tests/periph_rtc`.

OlegHahm, 2017-01-15:
Timeout

closed 2017-01-15 by OlegHahm