RIOT-OS/RIOT #5773: feature request: persisten system config

OlegHahm, 2016-08-27:
RIOT should provide a platform independent interface to store persistent configuration data, e.g. unique identifiers, radio channel, L2 address, crypto keys etc. This feature should be implemented with a minimum memory requirement, i.e. just storing a small struct somewhere in flashrom should be sufficient.

This issue replaces #162.

miri64, 2019-08-10:
There is work happening by the config task force.

miri64, 2019-08-10:
See #10910, #10911 and others for examples.

maribu, 2022-09-19:
I think with MTD there is little issue in storing configuration data in flash / EEPROM etc. If portability is no issue and one is not afraid of suffering great amounts of pita when changing the data format, even writing / reading a `struct` directly into storage would be possible.

I guess that https://github.com/RIOT-OS/RIOT/pull/17612 is a pretty good match. Maybe we can close this when https://github.com/RIOT-OS/RIOT/pull/17612 is in?

miri64, 2022-09-19:
IMHO MTD or at a higher level VFS completely suffices for the requirements set out by @OlegHahm and we just forgot to close this when ever those were merged. Everything else is just additional sugar on top. As such, I believe this issue can be closed. Feel free to reopen if you disagree.

closed 2022-09-19 by miri64