RIOT-OS/RIOT #5774: cpu: cortexm_common: context switching code breaks when compiling with LTO

kaspar030, 2016-08-29:
```
/tmp/ccLE0ZTV.s: Assembler messages:
/tmp/ccLE0ZTV.s:327: Error: invalid offset, value too big (0x00001078)
/tmp/ccLE0ZTV.s:377: Error: invalid offset, value too big (0x0000103C)
lto-wrapper: fatal error: arm-none-eabi-gcc returned 1 exit status
compilation terminated.
/usr/lib/gcc/arm-none-eabi/6.1.1/../../../../arm-none-eabi/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
m
```

To get this error, merge #5742, remove the "SRC_NOLTO" line from ```cpu/cortexm_common/Makefile``` and compile with ```LTO=1```.

miri64, 2016-10-18:
I don't understand... did #5742 introduce this error or did it fix it...

kaspar030, 2016-10-18:
#5742 allows selectively excluding files from being compiled with LTO, and does so with the cortexm context switching code.

miri64, 2016-10-18:
That did not answer my question, just described things I don't understand :)

kaspar030, 2016-10-18:
Well, see, the answer is not in the set you gave. The problem existed before and still does.

kaspar030, 2016-10-18:
Sorry, is that clear enough? Before #5742, LTO compiles were not possible at all for most platforms. That PR includes many fixes originally made by @gebart, plus a way to exclude files if they still pose problems when compiling with LTO, like the hand-written assembler code in cortexm's ctx switching code. I opened this issue so we don't forget about that.

kaspar030, 2016-10-18:
As LTO is quite effective without fixing this issue, and it's still optional and shouts "EXPERIMENTAL!" when enabled, this can be postponed...

miri64, 2016-10-18:
So, since this is optional it's not a known issue? (that's all I want to know ;-))

kaspar030, 2016-10-18:
I'd say, no. I mean, thanks to the workaround in #5742, LTO works fine on cortex-m, just the one file will not be link time optimized...

miri64, 2016-10-18:
"No, it's not a known issue" or "No, it is a known issue"? (Double negatives are always tricky in natural languages ;-))

kaspar030, 2016-10-18:
No need to mention this anywhere else but in this issue. ;)

OlegHahm, 2017-01-15:
This is still an issue, right?

kaspar030, 2017-08-29:
Yep, the problematic file is still excluded from LTO.

kYc0o, 2018-07-17:
@kaspar030 can you confirm that #7776 was the fix for this?

kaspar030, 2019-08-19:
Seems like gcc 9.1 has fixed this.

kaspar030, 2019-08-19:
> Seems like gcc 9.1 has fixed this.

Let's wait until our Docker container uses a recent enough version of gcc, then remove the workaround.

closed 2020-07-15 by miri64