RIOT-OS/RIOT #5775: cpu: cortex-m: compiling vectors.c with LTO triggers compiler bug

kaspar030, 2016-08-29:
```
lto1: internal compiler error: in add_symbol_to_partition_1, at lto/lto-partition.c:135
Please submit a full bug report,
with preprocessed source if appropriate.
See <https://bugs.archlinux.org/> for instructions.
lto-wrapper: fatal error: arm-none-eabi-gcc returned 1 exit status
compilation terminated.
/usr/lib/gcc/arm-none-eabi/6.1.1/../../../../arm-none-eabi/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
make: *** [/home/kaspar/src/riot.tmp/examples/default/../../Makefile.include:261: all] Error 1
```

In order to get this error, remove SRC_NOLTO line in cpu/<your-board>/Makefile and compile with LTO=1.

kaspar030, 2016-08-29:
As this seems to be a bug in gcc, I've marked it as memo. It is intended as reference point for the workarounds within our Makefiles.

gebart, 2016-08-29:
There's an upstream bug report in the GCC Bugzilla regarding the same error when building chibios with LTO, but I lost the url.

basilfx, 2016-08-29:
@gebart https://gcc.gnu.org/bugzilla/show_bug.cgi?id=69866?

gebart, 2016-08-29:
@basilfx that's the one I meant.

kaspar030, 2019-08-19:
Still present with gcc 9.1.

kaspar030, 2020-01-29:
Seems to be fixed in `arm-none-eabi-gcc (Arch Repository) 9.2.0`.

Xeverous, 2020-02-07:
Got the same crash but with GCC 9.2.1 (not on this project). Turned out the crash happened only when the executable code had LTO enabled but a static library did not. Having everything compiled with LTO or nothing solved the problem.

closed 2020-07-15 by miri64