RIOT-OS/RIOT #5776: make: Predefining CFLAGS are parsed weirdly

miri64, 2016-08-29:
When I define some `CFLAGS` in my application's Makefile before I include the Makefile.include, they get scrambled up somehow. At least in [this instance](https://github.com/miri64/RIOT_playground/blob/7b7e0fee11479ffe7efc0dce4a85ff32c6fa1770/stack_comparison/time_tx/lwip/Makefile#L8):

```
$ make clean all
make: Nothing to be done for 'clean'.
Building application "lwip_time_rx" for "iotlab-m3" with MCU "stm32f1".

"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/pkg/lwip
git -C /home/mlenders/Repositories/RIOT-OS/RIOT_playground/stack_comparison/time_rx/lwip/bin/pkg/iotlab-m3/lwip checkout -f fd4a109ffa6513b28a0c780a952cef1110423717
Warning: you are leaving 2 commits behind, not connected to
any of your branches:

  133c6c3 Add RIOT Makefiles
  e0ad5e0 Fix warnings

If you want to keep them by creating a new branch, this may be a good time
to do so with:

 git branch <new-branch-name> 133c6c3

HEAD is now at fd4a109... lowpan6.c: Implement SNMP counters
git -C /home/mlenders/Repositories/RIOT-OS/RIOT_playground/stack_comparison/time_rx/lwip/bin/pkg/iotlab-m3/lwip am --ignore-whitespace "/home/mlenders/Repositories/RIOT-OS/RIOT/pkg/lwip"/patches/*.patch
Applying: Fix warnings
Applying: Add RIOT Makefiles
touch /home/mlenders/Repositories/RIOT-OS/RIOT_playground/stack_comparison/time_rx/lwip/bin/pkg/iotlab-m3/lwip/.git-patched
make -C /home/mlenders/Repositories/RIOT-OS/RIOT_playground/stack_comparison/time_rx/lwip/bin/pkg/iotlab-m3/lwip
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT_playground/stack_comparison/time_rx/lwip/bin/pkg/iotlab-m3/lwip/src/api
/bin/sh: 4: Syntax error: ")" unexpected
/home/mlenders/Repositories/RIOT-OS/RIOT/Makefile.base:60: recipe for target '/home/mlenders/Repositories/RIOT-OS/RIOT_playground/stack_comparison/time_rx/lwip/bin/iotlab-m3/lwip_api/pppapi.o' failed
make[3]: *** [/home/mlenders/Repositories/RIOT-OS/RIOT_playground/stack_comparison/time_rx/lwip/bin/iotlab-m3/lwip_api/pppapi.o] Error 2
/home/mlenders/Repositories/RIOT-OS/RIOT/Makefile.base:20: recipe for target 'ALL--/home/mlenders/Repositories/RIOT-OS/RIOT_playground/stack_comparison/time_rx/lwip/bin/pkg/iotlab-m3/lwip/src/api' failed
make[2]: *** [ALL--/home/mlenders/Repositories/RIOT-OS/RIOT_playground/stack_comparison/time_rx/lwip/bin/pkg/iotlab-m3/lwip/src/api] Error 2
Makefile:9: recipe for target 'all' failed
make[1]: *** [all] Error 2
/home/mlenders/Repositories/RIOT-OS/RIOT_playground/stack_comparison/time_rx/lwip/../../../../RIOT/Makefile.include:301: recipe for target '/home/mlenders/Repositories/RIOT-OS/RIOT_playground/stack_comparison/time_rx/lwip/bin/iotlab-m3/lwip.a' failed
make: *** [/home/mlenders/Repositories/RIOT-OS/RIOT_playground/stack_comparison/time_rx/lwip/bin/iotlab-m3/lwip.a] Error 2
```

Bisecting it I saw that before #5097 this problem did not exist, but with its introduction I get the message above (baring other changes to RIOT that https://github.com/miri64/RIOT_playground/commit/45b988a41f095756b15972ed0836281457118a3e provides a fix for). I think the fact that this particular one is a string containing parenthesis is the problem.

 A workaround is to put the CFLAGS after the `Makefile.include` (or adding more escapes which I find equally counter-intuitive), but I don't know, what this could mean for CFLAGS defined as environment variables (I haven't tested it either).

Kijewski, 2016-08-29:
```text
/bin/sh: 4: Syntax error: ")" unexpected
```
That's [this line](https://github.com/RIOT-OS/RIOT/blob/bece851972726abc34aa1fee2eece69860381845/dist/tools/genconfigheader/genconfigheader.sh#L4). Is your /bin/sh broken?

miri64, 2016-08-29:
No, it's the `CFLAGS` I defined [here](https://github.com/miri64/RIOT_playground/blob/7b7e0fee11479ffe7efc0dce4a85ff32c6fa1770/stack_comparison/time_tx/lwip/Makefile#L8), the line 4 just is refering to the line in the [make rule of `%.o`](https://github.com/miri64/RIOT_playground/blob/7b7e0fee11479ffe7efc0dce4a85ff32c6fa1770/stack_comparison/time_tx/lwip/Makefile#L8)

miri64, 2016-08-29:
(edited the link due to copy-pasta)

gebart, 2016-08-29:
I'm guessing it's caused by the spaces, and the fact that make does not work well with spaces

Does the problem go away if you remove the whitespace inside the quotes?

gebart, 2016-08-29:
Also, the line 4 is not referring to the line 4 of the recipe, since all of the line endings have been escaped. The line 4 must be from newlines contained inside one of the variables, e.g. $(CFLAGS).

miri64, 2016-08-29:
Yes, without the whitespaces it works.

gebart, 2016-08-29:
Okay, I guess there's some `sort`, `addprefix`, `patsubst`, or other make function which operates on words which messes up this variable.

gebart, 2016-08-29:
Probably this line: https://github.com/RIOT-OS/RIOT/pull/5097/files#diff-ebbb3165b3276a049759e5cb4e04d6a4R504

gebart, 2016-09-26:
An article about Makefiles and parsing quoted strings that may be of interest in this matter:
http://www.dctrwatson.com/2012/06/gnu-make-double-quotes-and-lists/

cladmi, 2017-09-23:
From what I understood, it works in the article because they are always using variable with double quotes. So when replacing spaces by `+`, and quotes by spaces, they introduce a new separation between blocks, its as if they configured make to say "separator is double quote".

I tried a solution which would make this work, by removing only the `-D` entry if there is no double quote.
So its kept in the CFLAGS but does not break everything.

```
DQUOTE = "
# Stupid highlighting, let's give it another double-quote "

# double quotes should be replaced by something else, either findstring returns nothing
SUBST_QUOTE = _THERE_IS_A_QUOTE_
has_dquote = $(findstring $(SUBST_QUOTE),$(subst $(DQUOTE),$(SUBST_QUOTE),$(1)))
hide_defines_if_no_quote = $(if $(call has_dquote,$1),$1,$(patsubst -D%,,$1))

CFLAGS := $(foreach i,$(CFLAGS),$(call hide_defines_if_no_quote,$i))
```
And when applied with values with quotes, they are kept

```
CFLAGS=-g -DMEM_SIZE="(THREAD_STACKSIZE_DEFAULT + 3624)" -DPBUF_POOL_BUFSIZE=200 -O2
CFLAGS=-g -DMEM_SIZE="(THREAD_STACKSIZE_DEFAULT + 3624)"   -O2
```

cladmi, 2019-08-12:
This could be fixed by doing a change like https://github.com/RIOT-OS/RIOT/issues/9589#issuecomment-408587132 if there is any interest in reviewing.

HendrikVE, 2019-09-17:
I just had the same issue. I think it's desirable to be able to use spaces in CFLAGS definitions. Not everywhere alternatives like underscores work well. For example I would like to define a device's name which will be visible to users, but without the possibility of using spaces it will just look ugly as hell. If it's fixable, it should be fixed imo.

closed 2019-09-28 by aabadie