RIOT-OS/RIOT #5777: iotlab-term: make tmux optional

miri64, 2016-08-29:
This issue is two-fold:

1) I don't really understand the `iotlab-term` target in `Makefile.iotlab` anymore since the `tmux` logging feature was introduced in #5351 (main reason why I can't provide a fix myself). Considering that `make term` should be the simplest command of all that is a little disquieting.
2) Not everyone uses `tmux` and not everyone knows how to use it. The `iotlab-*` targets should simplify the usage of the IoT-Lab, so adding knowledge about another tool reduces the usability of these targets in my opinion.

cgundogan, 2016-08-29:
>Considering that make term should be the simplest command of all that is a little disquieting.

Why should `term` be the `simplest` command? The idea of this makefile is to abstract and hide complex command executions.

>Not everyone uses tmux and not everyone knows how to use it.

See https://github.com/RIOT-OS/RIOT/blob/master/dist/testbed-support/README.iotlab.md#iotlab-term

One of the incentives of using the aggregator within tmux is that a user can now de-/attach from a running experiment. This was not possible before. I.e. you can start an experiment without the `iotlab-term` first, and just attach whenever desired and still see the full output (with or without logging enabled).

If you have discomfort with scrolling (I think we had an offline discussion about this once), then just put `setw -g mode-mouse on` in your `.tmux.conf` on the remote server. This could be put into the README as an improvement of the doc, but I don't see why we should drop tmux.

OlegHahm, 2016-08-29:
While I mostly agree with @cgundogan here, I understand @miri64's concerns. How about offering a simple `iotlab-term` and an advanced `iotlab-tmux` target (maybe with different names)?

miri64, 2016-08-29:
It's not like I don't know how to handle tmux, but as @OlegHahm states: it hightens the entry bar in my opinion.

closed 2019-09-10 by stale[bot]