RIOT-OS/RIOT #5780: make: Changing between LTO=1, LTO=0 does not trigger rebuild

gebart, 2016-08-30:
`make LTO=1` followed by `make` does not trigger a complete rebuild.
Any CFLAGS changes should trigger a rebuild IMO.

kaspar030, 2016-08-30:
This can be generalized: any changes to any Makefile (-variables) should trigger a rebuild.

I don't think there's an elegant way of doing this in make. Even making everything depend on all Makefiles won't fix commandline supplied variables.

I resorted to using ccache and always rebuild everything.

kaspar030, 2016-08-30:
One way would be to add CFLAGS and LINKFLAGS to riotbuild.sh (as comment), so changing them causes a rebuild.

kaspar030, 2016-08-30:
> CFLAGS and LINKFLAGS to riotbuild.sh (as comment), so changing them causes a rebuild.

Actually the whole CFLAGS are part of riotbuild.h, and every OBJC depends on it.

kaspar030, 2016-08-30:
Ok, I'm getting there. Sorry for the noise. :)

As LTOFLAGS are only added to CFLAGS on a per-file basis, changing them doesn't change global CFLAGS and thus not riotbuild.sh.

kaspar030, 2016-08-30:
Fixed by #5781.

PR #5781 (2016-08-30): make: add LTOFLAGS to CFLAGS used to generate riotbuild.h
Fixes #5780.
closed 2016-08-30 by BytesGalore