RIOT-OS/RIOT #5787: OSX: linker option --gc-sections not supported

smlng, 2016-08-30:
The linker on OSX does not support `--gc-sections` as introduced by commit `5fcb09f6`. Alternative would be `-dead_strip` on OSX.

gebart, 2016-08-31:
Could someone else take over this assignment? I don't have an OSX machine to work on. @thomaseichinger, @emmanuelsearch, @kYc0o perhaps?

kYc0o, 2016-08-31:
I cannot find the commit, could you give the PR number?

thomaseichinger, 2016-08-31:
@kYc0o I think it should've been #5742.

smlng, 2016-08-31:
its 5fcb09f

kYc0o, 2016-08-31:
Confirmed.

```
ld: unknown option: --gc-sections
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [all] Error 1
```

kYc0o, 2016-08-31:
@smlng can you provide a PR using `-dead_strip`?

smlng, 2016-08-31:
@kYc0o: well I could add some OS (or linker) specific `#ifdef` magic. However, I've to admit I'm not very familiar with those linker flags and optimizations; I'll check back with a colleague tomorrow who is running OS X, too.

smlng, 2016-09-01:
Does anyone know (comparing binaries compiled w/ and w/o), how much we actual gain (or reduce size) using this linker option?

gebart, 2016-09-01:
I don't think code size matters that much on native, but keeping the build
as close to the real targets is quite desirable IMHO.

smlng, 2016-09-01:
Nah, unsure about that - I mean, compiling for native target will use system/OS specific libs, and those are likely larger (not optimized) compared to the embedded ones, or not? So comparing native target with any real board binary is kind of arbitrary/ambiguous anyway ...

smlng, 2016-09-01:
Btw. [here](http://blog.senthilthecoder.com/post/lto-linker-commandline/) an interesting post on compiler and linker flags/options when using LTO. Maybe we should apply/integrate this as well?

smlng, 2016-09-01:
made a PR #5794

gebart, 2016-09-26:
I believe this was fixed by #5794, reopen if this is not the case.

closed 2016-09-26 by gebart