RIOT-OS/RIOT #5790: ND: Lost of Global IPV6 on node after sending lot of UDP frame from BR

biboc, 2016-08-31:
I'm using two SAMR21, one with BR (on A) and the other (on B) with
gnrc_networking example.
I'm on April release.
Switch on A (border router) then switch on B, I'm able to send UDP
message from Linux to 2001:db8 address of B.
If I try to send lot of data (every 100ms), it works for a while and
then B does not receive message anymore. After I see it failed, I
stopped the loop sending UDP message, I checked on BR and 2001:db8
address of B was not here anymore. I can ping B with fe80:: address on
iface 6 but can't with 2001:db8.

Why or how can a node be removed from ncache? Why the node does not
automatically reconnect with BR?

Answer from Martine:
Hi,
can you check if this is https://github.com/RIOT-OS/RIOT/issues/5467?

> Why or how can a node be removed from ncache? Why the node does not
automatically reconnect with BR?

As the implementer of the neighbor discovery I can say this: because
the neighbor discovery is sh** ;-). It was written in a haste and has
a lot of design flaws. I'm working on a replacement (see
https://github.com/RIOT-OS/RIOT/issues/5704, happy to hear your
thoughts ;-)), but sadly I did not find some time to put heavy
implementation efforts into this yet (hopefully in autumn though, so
we have something for the October release, fingers crossed).

Thanks for reporting and kind regards,
Martine

biboc, 2016-10-18:
I tried again to send lot of data in a row (20 packets/sec)
And actually NDP sends also messages between these packets and at some point, the node shows: 
`ndp internal: set 2001:db8::fec2:3d00:4:abb1 state to STALE (go into DELAY on next packet)`
Then I can't communicate to the node with global IPV6 but I can with local IPV6

miri64, 2016-10-18:
Directly after?

biboc, 2016-10-18:
node starts and get global IPV6: great
I start sending data (~20 packets/sec), I receive 380 packets and it stops with last message: 
`ndp internal: set 2001:db8::fec2:3d00:4:abb1 state to STALE (go into DELAY on next packet)`

In the middle, I've got NDP receiving SL2A:
```
(h:15 m:20 s:32) received line: 113
(h:15 m:20 s:32) Waiting for incoming packet...
(h:15 m:20 s:32) received line: 114
(h:15 m:20 s:32) Waiting for incoming packet...
(h:15 m:20 s:32) received line: 115
(h:15 m:20 s:32) Waiting for incoming packet...
(h:15 m:20 s:32) received line: 116
(h:15 m:20 s:32) Waiting for incoming packet...
(h:15 m:20 s:32) ndp: received SL2A (link-layer address: fc:c2:3d:00:00:04:ab:b1)
(h:15 m:20 s:32) ndp internal: send neighbor advertisement (iface: 6, tgt: 2001:db8::fec2:3d00:4:c0ba, dst: 2001:db8::fec2:3d00:4:abb1, supply_tl2a: 0)
(h:15 m:20 s:32) ndp internal: set 2001:db8::fec2:3d00:4:abb1 state to DELAY (probe with unicast NS in 5 seconds)
(h:15 m:20 s:33) received line: 121
...
(h:15 m:20 s:45) received line: 380
(h:15 m:20 s:45) Waiting for incoming packet...
(h:15 m:20 s:55) ndp internal: set 2001:db8::fec2:3d00:4:abb1 state to STALE (go into DELAY on next packet)
```

And you can see that node misses some lines when NDP is talking

miri64, 2016-10-18:
what are those "lines"?

biboc, 2016-10-18:
Some UDP testing lines ("hello world')

miri64, 2016-10-18:
Ah okay, thanks. I will consider this for my tests when testing the new ND!

biboc, 2016-10-18:
This mainly show that if there are many messages arriving at the same time on a node, it can loose its global IPV6 address

Do you know why or how packets from 116 to 121 are not received?

biboc, 2016-10-18:
If I do the same test with local IPV6, it will loose some line when ND runs but it will continue to the end because local IPV6 is not lost (comparing to global IPV6)

miri64, 2016-10-18:
> Do you know why or how packets from 116 to 121 are not received?

I have multiple suspicions.

* Some kind of collision at L2
* Next hop determination fails due to the packet being in DELAY state
* some issue in the  packet buffer
* full packet queues of the IPv6 thread

Any thing sounds like something you observed?

biboc, 2016-10-18:
> Some kind of collision at L2
> Next hop determination fails due to the packet being in DELAY state
> some issue in the packet buffer
> full packet queues of the IPv6 thread

I can test but I need to know where to debug?
I didn't observed more than one I said yet

miri64, 2016-10-18:
> I can test but I need to know where to debug?

Try the following:

>> * Some kind of collision at L2

Check with a sniffer

>> * Next hop determination fails due to the packet being in DELAY state

Add `DEBUG()` statements before any return of `gnrc_sixlowpan_nd_next_hop_l2addr` in `gnrc_sixlowpan_nd.c`

>> * some issue in the packet buffer

Call `gnrc_pktbuf_stats()` whenever this is occuring (if it helps: this should be easily implementable as a shell commands).

>> * full packet queues of the IPv6 thread

Call `msg_queue_print` in `_event_loop` in `gnrc_ipv6.c` before and after the `msg_receive()` call.

biboc, 2016-10-18:
With IPV6 debug, I added the number of lost lines since the start of sending packets:

```
(h:16 m:0 s:29) ipv6: Received (src = fe80::fec2:3d00:4:abb1, dst = 2001:db8::fec2:3d00:4:c0ba, next header = 17, length = 31)
(h:16 m:0 s:29) ipv6: forward nh = 17 to other threads
(h:16 m:0 s:29) ipv6: Send receive command for 0x20002cd8 to 5
(h:16 m:0 s:29) ipv6: waiting for incoming message.
(h:16 m:0 s:29) received line: 360
(h:16 m:0 s:29) Nb lost lines: 81
(h:16 m:0 s:29) Waiting for incoming packet...
(h:16 m:0 s:29) ipv6: GNRC_NETAPI_MSG_TYPE_RCV received
(h:16 m:0 s:29) ipv6: Received (src = fe80::fec2:3d00:4:abb1, dst = 2001:db8::fec2:3d00:4:c0ba, next header = 17, length = 31)
(h:16 m:0 s:29) ipv6: forward nh = 17 to other threads
(h:16 m:0 s:29) ipv6: Send receive command for 0x20002cd8 to 5
(h:16 m:0 s:29) ipv6: waiting for incoming message.
(h:16 m:0 s:29) received line: 361
(h:16 m:0 s:29) Nb lost lines: 81
(h:16 m:0 s:29) Waiting for incoming packet...
(h:16 m:0 s:29) ipv6: GNRC_NETAPI_MSG_TYPE_RCV received
(h:16 m:0 s:29) ipv6: Received (src = fe80::fec2:3d00:4:abb1, dst = 2001:db8::fec2:3d00:4:c0ba, next header = 17, length = 31)
(h:16 m:0 s:29) ipv6: forward nh = 17 to other threads
(h:16 m:0 s:29) ipv6: Send receive command for 0x20002cd8 to 5
(h:16 m:0 s:29) ipv6: waiting for incoming message.
(h:16 m:0 s:29) received line: 362
(h:16 m:0 s:29) Nb lost lines: 81
(h:16 m:0 s:29) Waiting for incoming packet...
 (h:16 m:0 s:37) ipv6: GNRC_NETAPI_MSG_TYPE_RCV received
(h:16 m:0 s:37) ipv6: Received (src = fe80::fec2:3d00:4:abb1, dst = ff15::abcd, next header = 17, length = 14)
(h:16 m:0 s:37) ipv6: packet destination not this host
(h:16 m:0 s:37) ipv6: decrement hop limit to 63
(h:16 m:0 s:37) ipv6: do not forward packets with link-local source or destination address
(h:16 m:0 s:37) ipv6: waiting for incoming message.
(h:16 m:0 s:39) ipv6: Neigbor cache state timeout received
(h:16 m:0 s:39) ndp internal: set 2001:db8::fec2:3d00:4:abb1 state to STALE (go into DELAY on next packet)
(h:16 m:0 s:39) ipv6: waiting for incoming message.
(h:16 m:1 s:37) ipv6: GNRC_NETAPI_MSG_TYPE_RCV received
(h:16 m:1 s:37) ipv6: Received (src = fe80::fec2:3d00:4:abb1, dst = ff15::abcd, next header = 17, length = 14)
(h:16 m:1 s:37) ipv6: packet destination not this host
(h:16 m:1 s:37) ipv6: decrement hop limit to 63
(h:16 m:1 s:37) ipv6: do not forward packets with link-local source or destination address
(h:16 m:1 s:37) ipv6: waiting for incoming message.
```

biboc, 2016-10-18:
I added the followings:
> Add DEBUG() statements before any return of gnrc_sixlowpan_nd_next_hop_l2addr in gnrc_sixlowpan_nd.c

> Call msg_queue_print in _event_loop in gnrc_ipv6.c before and after the msg_receive() call.

> Call gnrc_pktbuf_stats() whenever this is occuring (if it helps: this should be easily implementable as a shell commands).

gnrc_pktbuf_stats() is called every received packet

```
(h:16 m:19 s:21) received line: 363
(h:16 m:19 s:21) Nb lost lines: 63
(h:16 m:19 s:21) Waiting for incoming packet...
(h:16 m:19 s:21) Message queue of thread 4
(h:16 m:19 s:21)     size: 8 (avail: 0)
(h:16 m:19 s:21) received line: 364
(h:16 m:19 s:21) Nb lost lines: 63
(h:16 m:19 s:21) Waiting for incoming packet...
(h:16 m:19 s:21) ndp: received SL2A (link-layer address: fc:c2:3d:00:00:04:ab:b1)
(h:16 m:19 s:21) ndp internal: send neighbor advertisement (iface: 6, tgt: 2001:db8::fec2:3d00:4:c0ba, dst: 2001:db8::fec2:3d00:4:abb1, supply_tl2a: 0)
(h:16 m:19 s:21) Message queue of thread 4
(h:16 m:19 s:21)     size: 8 (avail: 1)
(h:16 m:19 s:21)     * 5: sender: 4, type: 0x0514, content: 536882104 (0x20002bb8)
(h:16 m:19 s:21) gnrc_sixlowpan_nd_next_hop_l2addr: gnrc_ipv6_nc_get_l2_addr(l2addr, l2addr_len, nc_entry)
(h:16 m:19 s:21) Message queue of thread 4
(h:16 m:19 s:21)     size: 8 (avail: 1)
(h:16 m:19 s:21)     * 6: sender: 3, type: 0x0513, content: 536881840 (0x20002ab0)
(h:16 m:19 s:21) Message queue of thread 4
(h:16 m:19 s:21)     size: 8 (avail: 0)
(h:16 m:19 s:21) received line: 365
(h:16 m:19 s:21) Nb lost lines: 63
(h:16 m:19 s:21) Waiting for incoming packet...
(h:16 m:19 s:21) Message queue of thread 4
(h:16 m:19 s:21)     size: 8 (avail: 0)
(h:16 m:19 s:21) received line: 367
(h:16 m:19 s:21) Nb lost lines: 64
(h:16 m:19 s:21) Waiting for incoming packet...
(h:16 m:19 s:22) Message queue of thread 4
(h:16 m:19 s:22)     size: 8 (avail: 0)
(h:16 m:19 s:22) received line: 368
(h:16 m:19 s:22) Nb lost lines: 64
(h:16 m:19 s:22) Waiting for incoming packet...
(h:16 m:19 s:22) Message queue of thread 4
(h:16 m:19 s:22)     size: 8 (avail: 0)
(h:16 m:19 s:22) received line: 369
(h:16 m:19 s:22) Nb lost lines: 64
(h:16 m:19 s:22) Waiting for incoming packet...
(h:16 m:19 s:22) Message queue of thread 4
(h:16 m:19 s:22)     size: 8 (avail: 0)
(h:16 m:19 s:22) received line: 370
(h:16 m:19 s:22) Nb lost lines: 64
(h:16 m:19 s:22) Waiting for incoming packet...
(h:16 m:19 s:22) Message queue of thread 4
(h:16 m:19 s:22)     size: 8 (avail: 0)
(h:16 m:19 s:22) received line: 371
(h:16 m:19 s:22) Nb lost lines: 64
(h:16 m:19 s:22) Waiting for incoming packet...
(h:16 m:19 s:22) Message queue of thread 4
(h:16 m:19 s:22)     size: 8 (avail: 0)
(h:16 m:19 s:22) received line: 372
(h:16 m:19 s:22) Nb lost lines: 64
(h:16 m:19 s:22) Waiting for incoming packet...
(h:16 m:19 s:22) Message queue of thread 4
(h:16 m:19 s:22)     size: 8 (avail: 0)
(h:16 m:19 s:22) received line: 373
(h:16 m:19 s:22) Nb lost lines: 64
(h:16 m:19 s:22) Waiting for incoming packet...
(h:16 m:19 s:22) Message queue of thread 4
(h:16 m:19 s:22)     size: 8 (avail: 0)
(h:16 m:19 s:22) received line: 374
(h:16 m:19 s:22) Nb lost lines: 64
(h:16 m:19 s:22) Waiting for incoming packet...
(h:16 m:19 s:22) Message queue of thread 4
(h:16 m:19 s:22)     size: 8 (avail: 0)
(h:16 m:19 s:22) received line: 375
(h:16 m:19 s:22) Nb lost lines: 64
(h:16 m:19 s:22) Waiting for incoming packet...
(h:16 m:19 s:22) Message queue of thread 4
(h:16 m:19 s:22)     size: 8 (avail: 0)
(h:16 m:19 s:22) received line: 376
(h:16 m:19 s:22) Nb lost lines: 64
(h:16 m:19 s:22) Waiting for incoming packet...
(h:16 m:19 s:22) Message queue of thread 4
(h:16 m:19 s:22)     size: 8 (avail: 0)
(h:16 m:19 s:22) received line: 377
(h:16 m:19 s:22) Nb lost lines: 64
(h:16 m:19 s:22) Waiting for incoming packet...
(h:16 m:19 s:22) Message queue of thread 4
(h:16 m:19 s:22)     size: 8 (avail: 0)
(h:16 m:19 s:22) received line: 378
(h:16 m:19 s:22) Nb lost lines: 64
(h:16 m:19 s:22) Waiting for incoming packet...
(h:16 m:19 s:22) Message queue of thread 4
(h:16 m:19 s:22)     size: 8 (avail: 0)
(h:16 m:19 s:22) received line: 379
(h:16 m:19 s:22) Nb lost lines: 64
(h:16 m:19 s:22) Waiting for incoming packet...
(h:16 m:19 s:22) Message queue of thread 4
(h:16 m:19 s:22)     size: 8 (avail: 0)
(h:16 m:19 s:22) received line: 380
(h:16 m:19 s:22) Nb lost lines: 64
(h:16 m:19 s:22) Waiting for incoming packet...
(h:16 m:19 s:22) Message queue of thread 4
(h:16 m:19 s:22)     size: 8 (avail: 0)
(h:16 m:19 s:22) received line: 381
(h:16 m:19 s:22) Nb lost lines: 64
(h:16 m:19 s:22) Waiting for incoming packet...
(h:16 m:19 s:32) ndp internal: set 2001:db8::fec2:3d00:4:abb1 state to STALE (go into DELAY on next packet)
(h:16 m:19 s:32) Message queue of thread 4
(h:16 m:19 s:32)     size: 8 (avail: 0)
(h:16 m:19 s:42) Message queue of thread 4
(h:16 m:19 s:42)     size: 8 (avail: 0)

```

miri64, 2016-10-18:
I'm a little bit confused if this is the sender or the receiver now? Also, come to think of it `received line: xxx` suggests to me that the lines don't get lost but that the message is just not printed...

```
(h:16 m:19 s:21) received line: 365
(h:16 m:19 s:21) Nb lost lines: 63
(h:16 m:19 s:21) Waiting for incoming packet...
(h:16 m:19 s:21) Message queue of thread 4
(h:16 m:19 s:21)     size: 8 (avail: 0)
(h:16 m:19 s:21) received line: 367
(h:16 m:19 s:21) Nb lost lines: 64
```

Seems like none of my suspicions are true (if this is the sender)

biboc, 2016-10-18:
This is the receiver, I only gave receiver output.
Line are not received because you can see that lost lines var is incremented

miri64, 2016-10-18:
> This is the receiver, I only gave receiver output.
> Line are not received because you can see that lost lines var is incremented

I have 2 questions:

* How do you no the line got lost?
* How is the loss of the global IPv6 address at the *receiver*'s neighbor cache related to packet loss?

I think I still don't understand your setup :(, but maybe the issue is a duplicate of #5122?

> Add DEBUG() statements before any return of gnrc_sixlowpan_nd_next_hop_l2addr in gnrc_sixlowpan_nd.c

This only helps on the sender side.

biboc, 2016-10-18:
Ok I've got a BR + one node.
BR sends UDP packets (20/sec) with payload:"[n° packet]:hello world"
I showed here the output of the node

> How do you no the line got lost?

Because each sent line has a number in the payload :-)

> How is the loss of the global IPv6 address at the receiver's neighbor cache related to packet loss

Good question, only thing I see is that when ND runs (SL2A solicitation), the receiver misses some lines. And if the packets are sent to global IPV6 of the node, it stops receiving at some point as shown earlier.

Is it clearer?

miri64, 2016-10-19:
Yes. Can you do the same debugging on the BR? Or isn't that a RIOT node? Another theory I have might be, that the missing MAC might let you loose packets (both UDP and NDP) and that the issue stems from that.

biboc, 2016-10-20:
Ok I got debug for BR and node. BR tries to send 1000 lines (with payload= "[Num Line] hello". I kept only interesting part of debug.

node:
```
Waiting for incoming packet...
received line: 0
Nb lost lines: 0
Waiting for incoming packet...
received line: 1
Nb lost lines: 0
...
Waiting for incoming packet...
received line: 116
Nb lost lines: 0
Waiting for incoming packet...
ndp: received SL2A (link-layer address: fc:c2:3d:00:00:04:ab:b1)
ndp internal: send neighbor advertisement (iface: 6, tgt: 2001:db8::fec2:3d00:4:c0ba, dst: 2001:db8::fec2:3d00:4:abb1, supply_tl2a: 0)
ndp internal: set 2001:db8::fec2:3d00:4:abb1 state to DELAY (probe with unicast NS in 5 seconds)
received line: 119
Nb lost lines: 2
...
Waiting for incoming packet...
received line: 130
Nb lost lines: 2
Waiting for incoming packet...
ndp: received SL2A (link-layer address: fc:c2:3d:00:00:04:ab:b1)
ndp internal: send neighbor advertisement (iface: 6, tgt: 2001:db8::fec2:3d00:4:c0ba, dst: 2001:db8::fec2:3d00:4:abb1, supply_tl2a: 0)
received line: 135
Nb lost lines: 6
...
Waiting for incoming packet...
received line: 148
Nb lost lines: 6
Waiting for incoming packet...
ndp: received SL2A (link-layer address: fc:c2:3d:00:00:04:ab:b1)
ndp internal: send neighbor advertisement (iface: 6, tgt: 2001:db8::fec2:3d00:4:c0ba, dst: 2001:db8::fec2:3d00:4:abb1, supply_tl2a: 0)
received line: 151
Nb lost lines: 8
...
Waiting for incoming packet...
received line: 162
Nb lost lines: 8
Waiting for incoming packet...
received line: 163
Nb lost lines: 8
Waiting for incoming packet...
ndp internal: set 2001:db8::fec2:3d00:4:abb1 state to PROBE (probe with 3 unicast NS every 1000000 us)
ndp internal: send neighbor solicitation (iface: 6, src: @!,  tgt: 2001:db8::fec2:3d00:4:abb1, dst: 2001:db8::fec2:3d00:4:abb1)
6lo nd: address registration successful
ndp internal: set 2001:db8::fec2:3d00:4:abb1 state to REACHABLE (reachable time = 18000000 us)
received line: 214
Nb lost lines: 58
Waiting for incoming packet...
received line: 215
Nb lost lines: 58
...
Waiting for incoming packet...
received line: 329
Nb lost lines: 58
Waiting for incoming packet...
ndp: received SL2A (link-layer address: fc:c2:3d:00:00:04:ab:b1)
ndp internal: send neighbor advertisement (iface: 6, tgt: 2001:db8::fec2:3d00:4:c0ba, dst: 2001:db8::fec2:3d00:4:abb1, supply_tl2a: 0)
received line: 332
Nb lost lines: 60
...
Waiting for incoming packet...
received line: 347
Nb lost lines: 60
Waiting for incoming packet...
ndp: received SL2A (link-layer address: fc:c2:3d:00:00:04:ab:b1)
ndp internal: send neighbor advertisement (iface: 6, tgt: 2001:db8::fec2:3d00:4:c0ba, dst: 2001:db8::fec2:3d00:4:abb1, supply_tl2a: 0)
received line: 350
Nb lost lines: 62
...
Waiting for incoming packet...
received line: 362
Nb lost lines: 62
Waiting for incoming packet...
ndp: received SL2A (link-layer address: fc:c2:3d:00:00:04:ab:b1)
ndp internal: send neighbor advertisement (iface: 6, tgt: 2001:db8::fec2:3d00:4:c0ba, dst: 2001:db8::fec2:3d00:4:abb1, supply_tl2a: 0)
received line: 366
Nb lost lines: 65
...
Waiting for incoming packet...
received line: 376
Nb lost lines: 65
Waiting for incoming packet...
 ndp internal: set 2001:db8::fec2:3d00:4:abb1 state to STALE (go into DELAY on next packet)
```

BR:
```
line: 0
line: 1
line: 2
line: 3
...
line: 115
line: 116
ndp internal: set 2001:db8::fec2:3d00:4:c0ba state to PROBE (probe with 3 unicast NS every 1000000 us)
ndp internal: send neighbor solicitation (iface: 6, src: @!,  tgt: 2001:db8::fec2:3d00:4:c0ba, dst: 2001:db8::fec2:3d00:4:c0ba)
line: 117
...
line: 129
line: 130
ndp internal: send neighbor solicitation (iface: 6, src: @!,  tgt: 2001:db8::fec2:3d00:4:c0ba, dst: 2001:db8::fec2:3d00:4:c0ba)
line: 131
line: 132
...
line: 147
line: 148
ndp internal: send neighbor solicitation (iface: 6, src: @!,  tgt: 2001:db8::fec2:3d00:4:c0ba, dst: 2001:db8::fec2:3d00:4:c0ba)
line: 149
line: 150
...
line: 163
line: 164
ndp internal: set fe80::1 state to DELAY (probe with unicast NS in 5 seconds)
line: 165
...
line: 212
line: 213
ndp: received SL2A (link-layer address: fc:c2:3d:00:00:04:c0:ba)
ndp internal: send neighbor advertisement (iface: 6, tgt: 2001:db8::fec2:3d00:4:abb1, dst: 2001:db8::fec2:3d00:4:c0ba, supply_tl2a: 0)
ndp internal: set 2001:db8::fec2:3d00:4:c0ba state to DELAY (probe with unicast NS in 5 seconds)
line: 214
line: 215
...
line: 279
line: 280
ndp internal: set fe80::1 state to PROBE (probe with 3 unicast NS every 1000000 us)
ndp internal: send neighbor solicitation (iface: 7, src: @!,  tgt: fe80::1, dst: fe80::1)
ndp internal: set fe80::1 state to REACHABLE (reachable time = 24000000 us)
line: 281
line: 282
...
line: 328
line: 329
ndp internal: set 2001:db8::fec2:3d00:4:c0ba state to PROBE (probe with 3 unicast NS every 1000000 us)
ndp internal: send neighbor solicitation (iface: 6, src: @!,  tgt: 2001:db8::fec2:3d00:4:c0ba, dst: 2001:db8::fec2:3d00:4:c0ba)
line: 330
line: 331
...
line: 346
line: 347
ndp internal: send neighbor solicitation (iface: 6, src: @!,  tgt: 2001:db8::fec2:3d00:4:c0ba, dst: 2001:db8::fec2:3d00:4:c0ba)
line: 348
line: 349
...
line: 361
line: 362
ndp internal: send neighbor solicitation (iface: 6, src: @!,  tgt: 2001:db8::fec2:3d00:4:c0ba, dst: 2001:db8::fec2:3d00:4:c0ba)
line: 363
line: 364
...
line: 372
line: 373
ndp: received SL2A (link-layer address: 3a:c6:cd:6c:0c:90)
ndp internal: send neighbor advertisement (iface: 7, tgt: fe80::2, dst: fe80::1, supply_tl2a: 0)
line: 374
line: 375
line: 376
...
```

biboc, 2016-10-20:
>  missing MAC might let you loose packets

What do you mean? When does the node miss the MAC?

kaspar030, 2016-10-20:
@biboc Do you have the same issue with debugging disabled?

On the samr21, output over serial is *very* expensive.

miri64, 2016-10-20:
>> missing MAC might let you loose packets
>
> What do you mean? When does the node miss the MAC?

There is no MAC protocol implemented in RIOT yet. That's what I meant with missing MAC, but maybe the answer lies in @kaspar030's remark ;-).

biboc, 2016-10-20:
>@biboc Do you have the same issue with debugging disabled?
Yes same problem if ND debug is disabled
I'll try to disable more debug like waiting for incoming packet

biboc, 2016-10-20:
@miri64 Does anything seems weird on the debug log I gave?

biboc, 2016-10-20:
@kaspar030, see, I removed most of the debug (I show only if line%100 == 0)

```
100
ndp: received SL2A (link-layer address: fc:c2:3d:00:00:04:ab:b1)
ndp internal: send neighbor advertisement (iface: 6, tgt: 2001:db8::fec2:3d00:4:c0ba, dst: 2001:db8::fec2:3d00:4:abb1, supply_tl2a: 0)
ndp internal: set 2001:db8::fec2:3d00:4:abb1 state to DELAY (probe with unicast NS in 5 seconds)
ndp: received SL2A (link-layer address: fc:c2:3d:00:00:04:ab:b1)
ndp internal: send neighbor advertisement (iface: 6, tgt: 2001:db8::fec2:3d00:4:c0ba, dst: 2001:db8::fec2:3d00:4:abb1, supply_tl2a: 0)
ndp: received SL2A (link-layer address: fc:c2:3d:00:00:04:ab:b1)
ndp internal: send neighbor advertisement (iface: 6, tgt: 2001:db8::fec2:3d00:4:c0ba, dst: 2001:db8::fec2:3d00:4:abb1, supply_tl2a: 0)
ndp internal: set 2001:db8::fec2:3d00:4:abb1 state to PROBE (probe with 3 unicast NS every 1000000 us)
ndp internal: send neighbor solicitation (iface: 6, src: @!,  tgt: 2001:db8::fec2:3d00:4:abb1, dst: 2001:db8::fec2:3d00:4:abb1)
6lo nd: address registration successful
ndp internal: set 2001:db8::fec2:3d00:4:abb1 state to REACHABLE (reachable time = 18000000 us)
300
ndp: received SL2A (link-layer address: fc:c2:3d:00:00:04:ab:b1)
ndp internal: send neighbor advertisement (iface: 6, tgt: 2001:db8::fec2:3d00:4:c0ba, dst: 2001:db8::fec2:3d00:4:abb1, supply_tl2a: 0)
ndp: received SL2A (link-layer address: fc:c2:3d:00:00:04:ab:b1)
ndp internal: send neighbor advertisement (iface: 6, tgt: 2001:db8::fec2:3d00:4:c0ba, dst: 2001:db8::fec2:3d00:4:abb1, supply_tl2a: 0)
ndp: received SL2A (link-layer address: fc:c2:3d:00:00:04:ab:b1)
ndp internal: send neighbor advertisement (iface: 6, tgt: 2001:db8::fec2:3d00:4:c0ba, dst: 2001:db8::fec2:3d00:4:abb1, supply_tl2a: 0)
ndp internal: set 2001:db8::fec2:3d00:4:abb1 state to STALE (go into DELAY on next packet)
```

I don't receive anything afterwards

kaspar030, 2016-10-20:
@biboc hm, try Line%10, but disabling debug in ndp, and see if that works better. @miri64 That shouldn't happen anyways, right?

miri64, 2016-10-20:
@biboc the only thing suspicious that I see is that there are 3 NS, but no NA in reply seems to be received, though the node is sending it. So packet loss seems to be what is happening here. Now we only need to find out where in the line they get lost. So please first try completely without debug output.

@kaspar030 of course not!

biboc, 2016-10-20:
@kaspar030: try Line%10, but disabling debug in ndp
Nothing changed if debug is removed partially or totally

biboc, 2016-10-20:
>I see is that there are 3 NS, but no NA in reply seems to be received, though the node is sending it

Yes I see that, node NA packet gets lost because there are too many packet sent by the BR?

biboc, 2016-10-20:
Just to be sure, in the [RFC 6775](https://tools.ietf.org/html/rfc6775) there are three types of node: Host, Router and Border Router.
In RIOT, I don't see Host node?
gnrc_networking makes every node as a router? `USEMODULE += gnrc_ipv6_router_default`
gnrc_border_router makes the BR as a BR : `USEMODULE += gnrc_sixlowpan_border_router_default`

What's a Host for RIOT ND?

miri64, 2016-10-20:
@biboc `gnrc_minimal` is a host, but very reduced. Other alternatives are to remove the `_router` from the `USEMODULE` include of `ipv6` and the RPL related modules or to use [the host application I provided in my application repository](https://github.com/miri64/RIOT_playground/tree/master/gnrc_host). But I did not test this in month.

miri64, 2016-10-20:
(the application in my repo is basically what I described for modifications of `gnrc_networking` ;-))

biboc, 2016-10-20:
If I start two nodes (gnrc_networking) and one BR, sometimes the second nodes does not have its global IPV6 in BR's cache and I have the feeling that first node act as a router and the second node global IPV6 is in first node cache
Is that possible?

biboc, 2016-10-21:
Considering that I don't need my nodes to be router but only host, I removed the `_router` part

Here is now what I get as output:

Node:
```
6lo nd: retransmit multicast rtr sol in 10 sec
ndp internal: send router solicitation (iface: 6, dst: ff02::2)
ndp: building source link-layer address option (l2addr: fc:c2:3d:00:00:04:7b:6a)
ndp: building router solicitation message

ndp: received SL2A (link-layer address: fc:c2:3d:00:00:04:ab:b1)
ndp internal: send neighbor solicitation (iface: 6, src: 2001:db8::fec2:3d00:4:7b6a,  tgt: fe80::fec2:3d00:4:abb1, dst: fe80::fec2:3d00:4:abb1)
ndp: building source link-layer address option (l2addr: fc:c2:3d:00:00:04:7b:6a)
ndp: building neighbor solicitation message
ndp internal: set fe80::fec2:3d00:4:abb1 state to STALE (go into DELAY on next packet)
ndp internal: send neighbor solicitation (iface: 6, src: @!,  tgt: fe80::fec2:3d00:4:abb1, dst: fe80::fec2:3d00:4:abb1)
ndp: building source link-layer address option (l2addr: fc:c2:3d:00:00:04:7b:6a)
ndp: building neighbor solicitation message
ndp internal: set fe80::fec2:3d00:4:abb1 state to DELAY (probe with unicast NS in 5 seconds)

ndp: received neighbor advertisement (src: 2001:db8::fec2:3d00:4:abb1, dst: 2001:db8::fec2:3d00:4:7b6a, tgt: fe80::fec2:3d00:4:abb1)
6lo nd: address registration successful
ndp internal: set fe80::fec2:3d00:4:abb1 state to REACHABLE (reachable time = 18000000 us)
ndp: received neighbor advertisement (src: fe80::fec2:3d00:4:abb1, dst: fe80::fec2:3d00:4:7b6a, tgt: fe80::fec2:3d00:4:abb1)
6lo nd: address registration successful
ndp internal: set fe80::fec2:3d00:4:abb1 state to REACHABLE (reachable time = 18000000 us)

ndp: received neighbor solicitation (src: 2001:db8::fec2:3d00:4:abb1, dst: 2001:db8::fec2:3d00:4:7b6a, tgt: 2001:db8::fec2:3d00:4:7b6a)
ndp: received SL2A (link-layer address: fc:c2:3d:00:00:04:ab:b1)
ndp internal: send neighbor advertisement (iface: 6, tgt: 2001:db8::fec2:3d00:4:7b6a, dst: 2001:db8::fec2:3d00:4:abb1, supply_tl2a: 0)
ndp: building neighbor advertisement message
ndp: received neighbor solicitation (src: fe80::fec2:3d00:4:abb1, dst: fe80::fec2:3d00:4:7b6a, tgt: fe80::fec2:3d00:4:7b6a)
ndp: received SL2A (link-layer address: fc:c2:3d:00:00:04:ab:b1)
ndp internal: send neighbor advertisement (iface: 6, tgt: fe80::fec2:3d00:4:7b6a, dst: fe80::fec2:3d00:4:abb1, supply_tl2a: 0)
ndp: building neighbor advertisement message
ndp: received SL2A (link-layer address: fc:c2:3d:00:00:04:ab:b1)
ndp internal: send neighbor solicitation (iface: 6, src: @!,  tgt: fe80::fec2:3d00:4:abb1, dst: fe80::fec2:3d00:4:abb1)
ndp: building source link-layer address option (l2addr: fc:c2:3d:00:00:04:7b:6a)
ndp: building neighbor solicitation message
ndp: received neighbor advertisement (src: fe80::fec2:3d00:4:abb1, dst: fe80::fec2:3d00:4:7b6a, tgt: fe80::fec2:3d00:4:abb1)
6lo nd: address registration successful

0
10
20
30
40
50
60
70
80
ndp: received neighbor solicitation (src: 2001:db8::fec2:3d00:4:abb1, dst: 2001:db8::fec2:3d00:4:7b6a, tgt: 2001:db8::fec2:3d00:4:7b6a)
ndp: received SL2A (link-layer address: fc:c2:3d:00:00:04:ab:b1)
ndp internal: send neighbor advertisement (iface: 6, tgt: 2001:db8::fec2:3d00:4:7b6a, dst: 2001:db8::fec2:3d00:4:abb1, supply_tl2a: 0)
ndp: building neighbor advertisement message
ndp internal: set fe80::fec2:3d00:4:abb1 state to DELAY (probe with unicast NS in 5 seconds)
Total lost lines: 1
100
ndp: received neighbor solicitation (src: 2001:db8::fec2:3d00:4:abb1, dst: 2001:db8::fec2:3d00:4:7b6a, tgt: 2001:db8::fec2:3d00:4:7b6a)
ndp: received SL2A (link-layer address: fc:c2:3d:00:00:04:ab:b1)
ndp internal: send neighbor advertisement (iface: 6, tgt: 2001:db8::fec2:3d00:4:7b6a, dst: 2001:db8::fec2:3d00:4:abb1, supply_tl2a: 0)
ndp: building neighbor advertisement message
110
ndp: received neighbor solicitation (src: 2001:db8::fec2:3d00:4:abb1, dst: 2001:db8::fec2:3d00:4:7b6a, tgt: 2001:db8::fec2:3d00:4:7b6a)
ndp: received SL2A (link-layer address: fc:c2:3d:00:00:04:ab:b1)
ndp internal: send neighbor advertisement (iface: 6, tgt: 2001:db8::fec2:3d00:4:7b6a, dst: 2001:db8::fec2:3d00:4:abb1, supply_tl2a: 0)
ndp: building neighbor advertisement message
120
130
[... 136 lines cut ...]
ndp: building source link-layer address option (l2addr: 00:21:d3:60:5a:8d)
ndp: building neighbor solicitation message
ndp: received neighbor advertisement (src: fe80::1, dst: fe80::2, tgt: fe80::1)
ndp internal: set fe80::1 state to REACHABLE (reachable time = 24000000 us)

```

As you see BR sends  neighbor solicitation:
 `ndp internal: send neighbor solicitation (iface: 6, src: @!,  tgt: 2001:db8::fec2:3d00:4:7b6a, dst: 2001:db8::fec2:3d00:4:7b6a) `
and the host receives it (global address) and answers:
 ```
ndp: received neighbor solicitation (src: 2001:db8::fec2:3d00:4:abb1, dst: 2001:db8::fec2:3d00:4:7b6a, tgt: 2001:db8::fec2:3d00:4:7b6a)
ndp: received SL2A (link-layer address: fc:c2:3d:00:00:04:ab:b1)
ndp internal: send neighbor advertisement (iface: 6, tgt: 2001:db8::fec2:3d00:4:7b6a, dst: 2001:db8::fec2:3d00:4:abb1, supply_tl2a: 0)
ndp: building neighbor advertisement message
```

but BR does not receive the answer.

What's interesting also is that the host sends neighbor solicitation (with local address):
`ndp internal: send neighbor solicitation (iface: 6, src: @!,  tgt: fe80::fec2:3d00:4:abb1, dst: fe80::fec2:3d00:4:abb1)`
and the BR receives it:
`ndp: received neighbor solicitation (src: fe80::fec2:3d00:4:7b6a, dst: fe80::fec2:3d00:4:abb1, tgt: fe80::fec2:3d00:4:abb1)` 

What's your assumptions?

biboc, 2016-10-21:
By the way, between each packet there is a sleep of 30ms (for above debug)

I tried with a sleep of 1s and all packets are received `neighbor solicitation` and `neighbor advertisement`

biboc, 2016-10-21:
My conclusion is that when BR sends lot of packet quickly (30ms between each packet), advertisement answer from host get lost somewhere and BR and host consider themselves as not reachable so communication stops.

How can I stop NDP to run during the time my packets are sent? Or at least set the reachable time to high value so no neigbor solicitation/advertisement will be sent during the process?

miri64, 2017-01-06:
@biboc can you check if https://github.com/RIOT-OS/RIOT/issues/5122#issuecomment-270926714 fixes this? This would at least be a quick fix for the release.

kYc0o, 2017-04-26:
@biboc ping.

biboc, 2017-04-26:
I'm sorry @kYc0o , I won't be able to test this for now.

miri64, 2017-11-06:
Fixed by #7925

biboc, 2017-11-07:
Great, I believe you!

miri64, 2017-11-27:
After pondering this a little bit, I I'm not sure if this is actually a conjestion issue (which can't be fixed with normal IPv6). Can you retry?

miri64, 2017-11-28:
Testing now with

```bash
while true; do
    sleep 0.1
    echo "Hello World!" | nc -u 2001:db8::7b62:3323:ec77:c86 1337 -w0
done
```

From Linux host.

miri64, 2017-11-28:
(I'll run it a bit longer, but sniffer already tells me, that the ND messages come through clear as day, both for link-local addresses and GUAs)

biboc, 2017-11-28:
Great if you solve it! 
Congrats for your job on network @miri64

miri64, 2017-11-28:
Ran for over an our now and multiple updates to the neighbor cache and default router list happened on both border and downstream router. I think we can deem this issue fixed.

closed 2017-11-28 by miri64