RIOT-OS/RIOT #5795: gnrc_netapi: dropped messages (receiver queue is full)

aeneby, 2016-09-01:
Sending 6lowpan packets which must be split into more than 9 fragments fails, since the message queue fills up and fragments >9 don't get sent.

I'm not 100% sure where the problem actually lies, or if it's even reproducible on other platforms, but here's how to reproduce it on a cc2538:

 - Enable DEBUG in `sys/net/gnrc/netapi/gnrc_netapi.c`
 - Build and run `gnrc_networking` example on target platform
 - At the shell, run `ping6 1 <any_link_local_address> 1000`

You should see at least one message of the form:

> gnrc_netapi: dropped message to 7 (receiver queue is full)

Sniffing the traffic shows that the last (n-9) fragments of the ping aren't being sent out.

miri64, 2016-09-01:
I can confirm that we don't have this problem on other platforms (at least on 2016.07 release), since I ran multiple experiments both with `at86rf2xx` devices and `netdev2_test` the last few days. I can't confirm for master, but since `gnrc_sixlowpan` wasn't touched since the release, I guess we can assume it is the case there, too. Have you tried it with a bigger queue size for `gnrc_netdev2`?

miri64, 2016-09-01:
Just saw, that all devices have the same priority as 6LoWPAN. Can you look, if a lower priority value (i.e. higher priority) has an effect on this issue?

aeneby, 2016-09-01:
> Have you tried it with a bigger queue size for gnrc_netdev2?

Setting `NETDEV2_NETAPI_MSG_QUEUE_SIZE` to 16 solves the problem.

> Can you look, if a lower priority value (i.e. higher priority) has an effect on this issue?

Changing cc2538_rf's priority from 3 to 2 (i.e. compared with 6lo's priority of 3) also seems to solve the problem.

So... what does this mean? Where does the issue actually lie, and why doesn't it affect other platforms (with the same message queue size and priority) in the same way? And what's the most appropriate fix likely to be?

miri64, 2016-09-01:
> So... what does this mean? Where does the issue actually lie, and why doesn't it affect other platforms (with the same message queue size and priority) in the same way? And what's the most appropriate fix likely to be?

I'm not sure exactly, but if I must guess I guess that `cc2538_rf` blocks sometime, when `at86rf2xx` usually doesn't resulting in the queue getting filled since its thread doesn't get to `msg_receive()`. While a quick fix (and also an appropriate fix because actually that was how we originally had the priority in mind: MAC having the lowest, then 6Lo, then IPv6, etc...) I'm wondering what cc2538_rf is doing, that its thread takes "so long" to get to `msg_receive()` (where it usually should wait for the majority of the time). On same priority it and 6Lo should at least switch with every block. Does the driver use any other blocking functions somewhere?

Resizing the queue shouldn't be the fix because it (a) uses more RAM (b) doesn't guarantee a fix under the guessed situation.

OlegHahm, 2016-09-02:
Couldn't it be that the `cc2538_rf` actually never get a chance to run until `6lo` is done with its fragmentation if they have the same priority? Having threads with the same priority is often problematic and is discouraged unless you know exactly what you're doing.

miri64, 2016-09-02:
@OlegHahm then we should fix our `netif_auto_init` ;-)

OlegHahm, 2016-09-02:
What do you mean?

miri64, 2016-09-02:
All of our MAC threads for all of our supported devices currently run on the same priority as 6LoWPAN, so if running on the same priority is a problem we should fix that.

OlegHahm, 2016-09-02:
Oh, I wasn't aware of that. Was there any reason for this design?

miri64, 2016-09-02:
As far as I know, no. I guess what happened was that we started without 6LoWPAN, we just set our thread priorities based on that and at some point I implemented 6LoWPAN and just decremented the IPv6 priority by one without checking the MAC priorities.

aeneby, 2016-09-02:
> All of our MAC threads for all of our supported devices currently run on the same priority as 6LoWPAN

But then why doesn't e.g. at86rf2xx have the same issue, since I assume its MAC thread is running at the same priority as 6lo too?

>  Does the driver use any other blocking functions somewhere?

There is some mutex locking happening, but I don't see how this should affect things. There is also a blocking wait for any in-progress transmissions before sending. Disabling the mutex makes no difference, but the TX wait is necessary.

(OT: Every time I look at this code I wonder if these mutex locks are even necessary. Is there any scenario where the upper layers could trigger a new netdev send halfway through an in-progress one? Given the message queue seems to be processed serially?)

miri64, 2016-09-02:
> But then why doesn't e.g. at86rf2xx have the same issue, since I assume its MAC thread is running at the same priority as 6lo too?

Probably because it uses no mutexes ;-)

```
$ git grep "mutex" -- drivers/at86rf2xx/
```

And though kw2xrf includes the header it doesn't use it either:

```
$ git grep "mutex" -- drivers/kw2xrf/
drivers/kw2xrf/kw2xrf.c:#include "mutex.h
```

> There is some mutex locking happening, but I don't see how this should affect things. There is also a blocking wait for any in-progress transmissions before sending. Disabling the mutex makes no [...]

Basically any additional blocking takes time away from the message receive. I agree, that the mutexes aren't really necessary: sending is always sequential and receiving is handled via the RX_EVENT bootstrap mechanism, ideally by the same thread that is handling it (otherwise the user should guarantee mutual exclusion IMHO, not the device driver). Is it easy to remove them? If yes, I would propose to do that as fix. The priority issue I see more separate and as we see with the current setup: It SHOULD work regardless ;-).

aeneby, 2016-09-02:
Ah, I think I see what is happening. I'm using a macro (`RFCORE_WAIT_UNTIL()`) to wait for ongoing transmissions to complete before sending, but this is not a blocking wait - it yields, which I guess is allowing `6lo` to fill up the message queue before the wait is complete. I should be using a regular blocking `while();` here, probably.

I'll submit a PR for this, and also one to get rid of the mutex lock since it doesn't seem necessary to have it. @miri64, will you open an issue for the priority levels?

miri64, 2016-09-02:
Will do.

aeneby, 2016-09-02:
For the record, though, I'm still a little confused as to why giving cc2538_rf a higher priority than 6lo solves the problem, given that it's still yielding and (presumably) still giving 6lo an opportunity to fill the message queue up...

aeneby, 2016-09-03:
Nevermind. I see that thread_yield() passes control to processes with the same or higher prio. Makes sense.

miri64, 2016-09-03:
> @miri64, will you open an issue for the priority levels?

See https://github.com/RIOT-OS/RIOT/pull/5803

aeneby, 2016-09-03:
> See #5803

Thank you. IMHO the issue is resolved with this PR.

OlegHahm, 2016-10-30:
So, this issue can be closed?

miri64, 2016-10-31:
Guess so

closed 2016-10-31 by miri64