RIOT-OS/RIOT #5796: native: tlsf: early malloc will lead to a crash

OlegHahm, 2016-09-01:
In case an application uses the TLSF package built for native, the process will crash if a TLSF function (like `malloc()`) is called before TLSF initialized its buffer - which is typically done in the application.

Possible solutions to this problem would be either to patch TLSF to check if the buffer was already reserved or to implement #4490.

kaspar030, 2016-09-01:
Is an early auto_init an option?

OlegHahm, 2016-09-01:
That's basically what #4490 proposes, yes.

miri64, 2016-10-18:
Maybe I miss something, but I don't get how this issue is distinct from #4490 accept the focus on native.

OlegHahm, 2017-01-15:
#4490 is a feature request. This issue indicates an existing problem with native that *could* be used implementing #4490. However, there are other means to fix it.

kYc0o, 2018-07-18:
@jcarrano another issue with tlsf. Do you confirm this one?

jcarrano, 2018-07-18:
Steps to reproduce:

- Go to `examples/hello-world`.
- Add `USEMODULE += tlsf-malloc` to the Makefile.
- Recompile.
- Run. I get `Segmentation fault (core dumped)`

Why is this marked as `native`? It should be a problem in **all** platforms.

jcarrano, 2018-07-18:
@OlegHahm Please see https://github.com/jcarrano/RIOT/blob/tlsf-init-experiments/examples/tlsf-init-test/main.c.

The thing with TLSF is that it does not have any means of requesting more memory from the OS (no `sbrk`) and no way to grow the heap (other than adding pools, but one ends up with a fragmented heap.

What I did in that test was to statically allocate 8MB, which should be plenty.

Another issue is that I'm not sure that `__attribute__((constructor))` runs the function early enough. It does not crash in my machine, but that does not mean everything is OK. I did not try to debug it with other tools.

jcarrano, 2019-09-10:
This is still a bug, it cannot be closed.

miri64, 2020-07-01:
Fixed by https://github.com/RIOT-OS/RIOT/pull/12032 which is currently stale.