RIOT-OS/RIOT #5799: kw2x: 15.4 duplicate transmits

smlng, 2016-09-02:
I found that the kw2x is transmitting each 15.4 message twice with an interval of 2ms. 

I first thought my sniffer has some problems, but then I started samr21 node which does not exhibit this behavior. But I saw this behavior with different kw2x nodes. I put a small wireshark capture in my dropbox [here](﻿﻿https://dl.dropboxusercontent.com/u/19629108/kw2x_sniff.pcapng), which shows it.

Note: source addresses ending on `3016` and `600a` are kw2x boards, the one with `47ee` is the samr21. Though this capture only shows RA this also applies to other messages (and types), too.

smlng, 2016-09-20:
FYI: @cgundogan and @PeterKietzmann

jfischer-phytec-iot, 2016-09-22:
short: in the current driver some things were not done properly :smile:. #5469 should fix it, testers are welcome.

kYc0o, 2017-04-25:
I think this should be fixed by #6453. Can you @smlng try it again?

smlng, 2017-06-15:
seems to be fixed with the adaption of netdev2 in #6453. I made some test and wireshark sniffs and cannot observe any duplicates anymore. So closing this!

closed 2017-06-15 by smlng