RIOT-OS/RIOT #5805: [RFC] Global constant header

miri64, 2016-09-03:
At the moment many (user-editable) constants are scattered in form of macros over the code base (e.g. address timeouts, thread priorities and stack sizes, packet buffer size, maximum link-layer address length, etc.). How about we collect them together in one single header file, that then gets included by those who need those constants?

OlegHahm, 2016-09-03:
This goes in the direction of something like a `config.h` as discussed several times ago, right?

miri64, 2016-09-03:
Did we? I didn't find an issue for that. But in general: yes.

OlegHahm, 2016-09-03:
We had something like this back in old Firekernel days.

miri64, 2016-10-18:
Why was it removed then?

miri64, 2016-10-18:
To be clear I would like to have this header be optional for applications or - alternatively - in a central spot like `sys/include/` on `core/include`

kaspar030, 2016-10-18:
I don't like it. If the constants are magic numbers belonging into one module, that's where they should be defined. If there's anything user configurable, it should still be there, just possibly enclosed by "#ifndef X, #define X <reasonable default>, #endif".

OlegHahm, 2017-01-15:
Maybe this is rather a topic for the mailing list or a meeting?

miri64, 2019-08-10:
Somewhat superceded by the efforts of the config TF.

closed 2019-08-10 by miri64