RIOT-OS/RIOT #5825: Bootstrap RIOT toolchain(s) on any OS

smlng, 2016-09-06:
While Vagrant is a very useful and convenient way to get started with RIOT development, I guess in the long run most users want to install the toolchain(s) right into their base OS. 

IMHO it would be useful to have some kind of generic OS bootstrap script that installs all required tools (`gcc`, `openocd` etc) like the Vagrant bootstrap / provisioning [script](https://github.com/RIOT-OS/RIOT/blob/master/dist/tools/vagrant/bootstrap.sh) we already have. 

For instance, _letsencrypt_ / _certbot_ provides such a bootstrap [script](https://github.com/certbot/certbot/blob/master/letsencrypt-auto#L487) too,  that detects the (Unix/Linux-based) OS and installs required package dependencies using the package manager of the specific OS, e.g. `apt` for Debian, `yum` for Fedora/RedHat and so on. 

Thoughts?

OlegHahm, 2016-09-06:
One immediate thought: in the early Firekernel/µkleos days we had a script to build the toolchain for ARM7. (Actually, I just realized that it is still there: https://github.com/RIOT-OS/RIOT/tree/master/dist/tools/toolchains .) The main reason why we decided to recommend other pre-compiled toolchains instead was the maintenance effort for this script. I have the impression that providing such a bootstrapping script as you propose would need even more maintenance, if we want to support all the usual suspects (Ubuntu, Debian, Arch, Fedora, Gentoo, FreeBSD, OSX, Windows...) I'm not sure we can handle this.

thomaseichinger, 2016-09-07:
I think this is a good idea in general but instead of actually compiling we should as the mentioned bootstrapping script use the OS's package manager. This might reduce the maintenance effort to a manageable amount.

LudwigKnuepfer, 2016-09-07:
I think it's a waste of time.

OlegHahm, 2016-09-07:
@thomaseichinger, that was my understanding anyway - still I think it is a lot of effort.

@LudwigKnuepfer, can you elaborate just a little bit more?

smlng, 2016-09-07:
Just to shed some light on this:

With the current support of vagrant by @cgundogan we already have such a bootstrap script for Ubuntu (only), which can naturally be used without vagrant to setup RIOT toolchains in _native_ Ubuntu. 

Further, we experienced some incompatibilities with certain tools in the past, i.e. OpenOCD or gcc-none-eabi. For example, specific versions of these tools where not working with RIOT (or hardware) on distinct Linux distros or releases. Solving such issues really is very annoying (pain in the ...) to new developers/users who want to implement or use applications with RIOT. 

Helping out with an easy to use bootstrap script for distinct (not all, nor outdated) Linux distros/releases might lower the entry bar - and might also be useful for test setups or even Murdock CI, if we ever want to test on different OS.

Nevertheless, it would be a lot of effort. But we have many developers all over the place, and they likely use different OS and have a working RIOT toolchain setup there in. Hence, they know which packets are required, which are not working, and how to work around these problems; and hopefully will contribute to this.

Maybe the wiki would be a good place to start this in the meantime and collect working toolchain setups?

OlegHahm, 2016-09-07:
To be clear: I completely agree that this would be a helpful amendment, I just want to point out that this is creates additional maintenance efforts. I'm not so much worried that we don't have the personal power and/or experience, but more about the fact that someone has to keep an eye on this to make sure that it doesn't get outdated. In my experience, it shows that people tend to forget to update these things after a while. I think we would need a mechanism to automatically notify the responsible maintainer if support for a certain OS is broken/outdated - and we need at least one responsible maintainer per OS, of course.

cgundogan, 2016-09-08:
>I think we would need a mechanism to automatically notify the responsible maintainer if support for a certain OS is broken/outdated

that would be a great use case for #5828 with a proper integration into murdock, so that tests run on different OSs automagically

cgundogan, 2016-09-08:
BTW: with #5828 we would already need such provisioning/bootstrapping scripts, if we go down that path to support multiple platforms/diferent versions of OSs. So both, this issue and #5828, go hand in hand.

OlegHahm, 2016-09-08:
@cgundogan, when would you run those tests on Murdock and how would you make sure that no unrelated PR is affected from changes in some OS' packaging system?

cgundogan, 2016-09-08:
@OlegHahm such tests would need to be executed periodically (cron et al.), independently of active PRs.

kaspar030, 2016-09-08:
> #5828 with a proper integration into murdock, so that tests run on different OSs automagically

Let's not forget that we're already using docker for the CI. Adding different flavours of Linux is easy there.

LudwigKnuepfer, 2016-09-08:
> @LudwigKnuepfer, can you elaborate just a little bit more?

I am not aware of any developer who can spend the time needed to make this work and keep it working who is not better off working on something else in RIOT. I consider vagrant a good enough option for all developers not working on one of the distributions that are already supported (via mostly very clear instructions in the wiki).

x3ro, 2016-10-29:
This may be a stupid question, but if we're already using Docker for the CI, couldn't we use that image as an official way for building stuff for RIOT on non-supported systems?

miri64, 2019-09-10:
I think we could still benefit from this. E.g. with OpenWRT the toolchain is also always bootstrapped e.g. when cross-compiling packets.

miri64, 2020-07-01:
I guess that goes into the general direction what @kaspar030 with https://github.com/RIOT-OS/toolchains?

closed 2021-02-04 by stale[bot]

kaspar030, 2021-02-04:
> I guess that goes into the general direction what @kaspar030 with https://github.com/RIOT-OS/toolchains?

yup...