RIOT-OS/RIOT #5845: How to run RIOT on Particle Photon?

LaszloLango, 2016-09-16:
I'd like to run RIOT on a [Particle Photon](https://docs.particle.io/datasheets/photon-datasheet) board. The porting documentation on wiki didn't helped much. I tried to make a new board based on Nucleo-f207 and Spark Core, but it is not working. What is the minimal set of defines (in `board.h` and `periph_conf.h`) to run simple "hello world" or led blinking example on this board?

closed 2016-09-21 by LaszloLango

OlegHahm, 2016-09-22:
I heard some rumours that @tilmannOSG was working on a port for that...

Anyway, can you maybe elaborate on what is missing in the porting guide?

LaszloLango, 2016-09-23:
@OlegHahm, it was hard to figure out what should be in `periph_conf.h`, but we could make it. The `F_CPU` is also misleading, because non of the existing ports use this define. The wiki page does not mention the `CLOCK_*` and  `*_NUMOF` defines. I know they are CPU specific macros, but it would be good to know where to look for them. :)

tilmannOSG, 2016-09-23:
@OlegHahm I can confirm that rumor :) In fact @LaszloLango is working in my team and together with fellow team members he has successfully ported RIOT to the Photon board in the last couple of days.  
Our intention is to contribute the port upstream eventually but that might still take a while :)

thomaseichinger, 2016-09-23:
@LaszloLango @tilmannOSG That's great to hear! I started working on it a couple of days ago too, mind to join forces on it?

tilmannOSG, 2016-09-29:
@thomaseichinger Yes, we would be happy to work together with you! I've sent you an email with some information about the current state of the port.

kosme, 2017-01-03:
@tilmannOSG Sounds terrific. Any possibility this port would also work on a Particle Electron?

tilmannOSG, 2017-03-01:
@kosme I'm not really familiar with the hardware of the Electron but I guess if it's close enough to the Photon then it shouldn't be too hard to get that working too :) (nothing we plan to work on though)

tilmannOSG, 2017-03-10:
WIP patch is now available at https://github.com/RIOT-OS/RIOT/pull/6730 :)