RIOT-OS/RIOT #5848: arduino: Race condition in sys/arduino/Makefile.include

gebart, 2016-09-16:
https://ci.riot-labs.de/RIOT-OS/RIOT/5794/291c1a54ebfde8e5d7ff3db97d5b01befae3b7f0/output.html#cortex_m3_1
observed in #5794 

https://github.com/RIOT-OS/RIOT/blob/master/sys/arduino/Makefile.include#L6 is executed in parallel for all targets of a make buildtest

https://github.com/RIOT-OS/RIOT/blob/master/dist/tools/arduino/pre_build.sh#L17 does not write the file atomically, which means that the file may be truncated at a time when the compiler from one of the boards tries to read it.

Suggested solution: Create the _sketches.cpp file inside `${BINDIR}` and create it as a part of the build process, not during evaluation of the Makefile. Also, write _sketches.cpp atomically by building a temporary file first, then renaming to the real file.

kYc0o, 2017-04-25:
@gebart Did you have time to work on the solution? @kaspar030 any other suggestion?

gebart, 2017-05-02:
I don't think I will have any time to spend on this issue, it is of very low priority for my use cases.

cladmi, 2017-09-27:
To fix that correctly by generating something in `BINDIR`, I think the best would be to create an `$(APPLICATION)_sketches` module in BINDIR.

I have use cases where the RIOT repository should be read-only and build are be produced in directories outside, so it definitely makes sense to keep APPDIR as is.

My reasons for thinking about a module is that, from what I understood, `Makefile.base` is meant to build from sources files that are in the current directory.
https://github.com/RIOT-OS/RIOT/blob/2017.07-branch/Makefile.base#L86
So to move them out, would require adding there location to `DIR` to build them from there.

So inside `Makefile.include` create a module in $(BINDIR) with the sketches renames as .cpp and add this as USEMODULE.

I will try something to see if it works.

cladmi, 2019-08-12:
I updated the pull request in https://github.com/RIOT-OS/RIOT/pull/7654

tcschmidt, 2019-08-17:
@cladmi what about this one: move or close?

cladmi, 2019-08-19:
As said in the previous comment, I updated the pull request.

PR #7654 (2019-08-21): arduino/sketches: build sketches as a module
Generate a module for arduino sketches in a subfolder of BINDIR.
This prevents issues when doing concurrent builds or out of tree build with readonly sources.

This fixes #5848

The concept is instead of generating a cpp file in the source directory, it is generated into the build directory. However, this requires defining a new module for this directory.
It is now generated during build and not parsing.
I did not addressed the "write atomically" as I think it is not necessary anymore.

To do this, I replaced the previous bash script by a Makefile s [... cut]
closed 2019-08-21 by smlng