RIOT-OS/RIOT #5849: pkg: ccn-lite: rework convenient functions

OlegHahm, 2016-09-18:
The convenient functions in CCN-lite's RIOT adapter for interest sending and waiting for content chunks need serious rework:
1.) They should be moved into the RIOT repository for easier modifications.
2.) The interface needs to be reworked. The "wait_for_chunk" function, for instance, need to provide access to the chunks meta data, like the name.
3.) The "wait_for_chunk" function currently sets a wrong timeout if interests retransmits are disabled.

miri64, 2016-09-19:
How about integrating this into `sock` right away?

kaspar030, 2016-09-19:
I think it is "convenience function".

https://github.com/RIOT-OS/RIOT/issues/5849

OlegHahm, 2016-09-19:
@miri64, yes, I will take a look if this makes sense.

@kaspar030, oops, thanks!

PeterKietzmann, 2019-02-21:
To 1) We actively decided to move the adapter code to the ccn-lite Repo. There has been a mayor restructuring some time ago.
To 2) and 3) The function `ccnl_wait_for_chunk` is not used anymore. 

I'm closing this issue as it is outdated. @OlegHahm, @cgundogan if you disagree with that, I'd ask you to open an updated version in the ccn-lite repository.

closed 2019-02-21 by PeterKietzmann