RIOT-OS/RIOT #5858: gnrc: 6lo: potential problem with reassembly of fragments

OlegHahm, 2016-09-21:
**Abstract** Packet loss seems to confuse the 6lo reassembly state machine.

### How to reproduce
**Hardware:** iotlab-m3
**Application:** gnrc_networking example
**Setup:**
 * two nodes in radio range
 * ping with small payload (e.g. default) -> works
 * ping with bigger payload (e.g. 256 bytes) -> works
 * ping with often with bigger payload and no delay, e.g. `ping6 100 <addr> 1024 0` -> works for some time, after the first timeout, all subsequent pings fails, too - until reboot
**Hypothesis:** as soon as one fragments gets lost on the "wire", the reassembly state machine gets somehow confused.

OlegHahm, 2016-09-21:
Interestingly:
```
5534738df8cfc6bdc308f82761801366912a8301 is the first bad commit
commit 5534738df8cfc6bdc308f82761801366912a8301
Author: Cenk Gündoğan <cnkgndgn@gmail.com>
Date:   Sat May 14 12:47:41 2016 +0200

    examples: auto-initialize RPL for gnrc_networking
```

cgundogan, 2016-09-21:
>Author: Cenk Gündoğan <cnkgndgn@gmail.com>

ow c'mon (;

miri64, 2016-09-21:
Mh, that commit doesn't do much besides including the auto-initialization of RPL (which does the same as `rpl init <if>` does)... Could be interesting if this happens too if RPL is manually initialized.

smlng, 2016-09-22:
@OlegHahm did you get through with _bisecting_? And have you found anything apart from RPL that's suspicious?

OlegHahm, 2016-09-22:
Yes, I fully bisected between 32be958eebb602562b35df9342981512edffad84 and b70204913427304b7baa4c370ee3c32e2875c82b. However, to identify the problem one has sometimes to ping more than 100 times - unless at least one ping timeout occurs. And in one case, the ping resumed after the first timeout, but hung after the next - if my theory holds that could mean that in the first case the very last fragment got lost. But as I said: we need a debugger here.

smlng, 2016-09-22:
@OlegHahm not in my case, I don't need 100 pings. With current master I *never*  got any successful ping (100% packet loss) between `pba-d-01-kw2x` and `samr21-xpro` for fragmented pings. But with `2016.04-branch` I had virtually not packet loss for large payloads, even `512B` with no delay was working.

OlegHahm, 2016-09-22:
That seems to be a separate problem, since pinging on current master works on iotlab-m3 and samr21-xpro - in *most* cases.

kYc0o, 2017-04-26:
Is this still an issue?

smlng, 2017-04-26:
I think (de)fragmentation is still a problem, specifically between different platforms and boards. But I haven't tested lately - did you run any tests?

kYc0o, 2017-04-26:
Just tested a iotlab-m3 vs samr21-xpro: works like a charm with the experimental setup provided by @OlegHahm.

Tested iotlab-m3 vs pba-d-01-kw2x: lots of timeouts even with small payload.

So I'd say that we can close this issue and maybe open a new one stating that pba-d-01-kw2x has some problems, which I think are more related to the radio driver than something on gnrc or 6lo.

@smlng would you mind to test iotlab-m3 against remote-revb for instance?

kYc0o, 2018-07-17:
Just tested with at86rf231, at86rf233 and cc2538 with very successful results (100% packets received). kw2x has problems so I'd say there's no problem with 6lo but with the radio driver.

Thus, I close this issue.

closed 2018-07-17 by kYc0o