RIOT-OS/RIOT #5863: gnrc_networking: cannot send udp packets > 23B

smlng, 2016-09-23:
Hi,

just ran into a curious problem, I used `gnrc_networking` example on two RIOT nodes, and run `udp server start 1234`. On the other node I run: `udp send <addr> 1234 foobar`. 

Which works up to a string of length 23, afterwards the `udp send ...` command just blocks, and while I saw all previous transmissions in the sniffer and where dumped on the server side, 24B and above seem to be stuck at the sender.

Here some output:
```
2016-09-23 13:21:16,628 - INFO # RIOT network stack example application
2016-09-23 13:21:16,631 - INFO # All up, running the shell now
udp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobar
2016-09-23 13:21:27,889 - INFO # > udp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobar
2016-09-23 13:21:27,895 - INFO # Success: send 18 byte to [fe80::1430:2488:724e:e2f0]:1234
udp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobara
2016-09-23 13:21:30,441 - INFO # > udp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobara
2016-09-23 13:21:30,447 - INFO # Success: send 19 byte to [fe80::1430:2488:724e:e2f0]:1234
udp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobara
2016-09-23 13:21:31,965 - INFO # > udp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobara
2016-09-23 13:21:31,971 - INFO # Success: send 19 byte to [fe80::1430:2488:724e:e2f0]:1234
udp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobarab
2016-09-23 13:21:34,879 - INFO # > udp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobarab
2016-09-23 13:21:34,886 - INFO # Success: send 20 byte to [fe80::1430:2488:724e:e2f0]:1234
udp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobarabc
2016-09-23 13:21:36,220 - INFO # > udp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobarabc
2016-09-23 13:21:36,226 - INFO # Success: send 21 byte to [fe80::1430:2488:724e:e2f0]:1234
udp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobarabcd
2016-09-23 13:21:37,662 - INFO # > udp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobarabcd
2016-09-23 13:21:37,668 - INFO # Success: send 22 byte to [fe80::1430:2488:724e:e2f0]:1234
udp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobarabcde
2016-09-23 13:21:38,996 - INFO # > udp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobarabcde
2016-09-23 13:21:39,002 - INFO # Success: send 23 byte to [fe80::1430:2488:724e:e2f0]:1234
udp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobarabcdef
```

smlng, 2016-09-23:
btw. latest master of RIOT it is!

OlegHahm, 2016-09-23:
My first guess would be that this limit is induced by the shell input buffer. Can you check this?

OlegHahm, 2016-09-23:
Could not reproduce your problem on master, using native or iotlab-m3 everything seems to run smoothly.

smlng, 2016-09-23:
mhm,

```
#define SHELL_DEFAULT_BUFSIZE   (128)
```

my command has 65 chars and is as follows, which is not working (one less is):
```
udp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobarabcdef
```

smlng, 2016-09-23:
I'm testing on `samr21-xpro`.

OlegHahm, 2016-09-23:
Is also working for me:
```
2016-09-23 13:39:40,854 - INFO # > udp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobarfoobarfoobarfoobarfoobarfoobarfoobar
2016-09-23 13:39:40,860 - INFO # Success: send 54 byte to [fe80::1430:2488:724e:e2f0]:1234
```

OlegHahm, 2016-09-23:
Would be helpful if you could identify which thread is blocking your setup. Using `make debug` and `call ps()` should help for that.

smlng, 2016-09-23:
I narrowed it down to OS X with `gcc-arm-none-eabi-4_9-2015q2` and `samr21-xpro`, there is problem with more than 64 chars in the shell. 

On Linux with `gcc-arm-none-eabi-5_2` everything works as expected.

smlng, 2016-09-26:
I updated my toolchain to latest `gcc-arm-none-eabi-5_4-2016q2`, still the same issue:

* on OS X and board `samr21-xpro` is the shell command exceeds 64 chars the command handling fails
* on OS X and board `pba-d-01-kw2x`, with same toolchain, everything works as expected 64+ no problem
* on Linux and board `samr21-xpro` with same toolchain (v5.4) everything works, too

Might be related to the UART/serial connection of the `samr21-xpro` under OS X. However, the issue persists.

kYc0o, 2016-10-18:
I experienced the same issue while testing `txtsnd` on a default RIOT example on OS X too. I-d say then that this is a OS X related problem. I don't have much time now so I don't promise to work on this soon.

For info my test was between a iotlab-m3 and a z1.

kYc0o, 2016-10-19:
@smlng Can you also check your python version? Since `make term` actually launches `pyterm` maybe this is more related to the Python version rather than Mac OS X.

smlng, 2016-10-19:
@kYc0o, I have python 2.7.12 install via `brew`. Just tested again against current master, and the issue persist. However, its not about sending packets or any other networking stuff; its the shell command input and parsing. It somehow has a problem with lines longer than 64 chars - I'll adapt the issue title accordingly.

Btw. the issue is only with `BOARD=samr21-xpro` and OSX, another board such as `pba-d-01-kw2x` works fine under OSX. And the `samr21-xpro` works fine under Linux. And further, I use `gcc-arm-none-eabi-5_4` on OSX and Linux.

Yonezawa-T2, 2016-10-20:
Maybe we should check returned value from `_native_write` in `uart_write`:
https://github.com/RIOT-OS/RIOT/blob/7020b7c09c3d511cb3470eae9d5aeecade8c0edf/cpu/native/periph/uart.c#L175

[Man write(2)](https://developer.apple.com/legacy/library/documentation/Darwin/Reference/ManPages/man2/write.2.html) says: “When using non-blocking I/O on objects, such as sockets, that are subject to flow control, write() and writev() may write fewer bytes than requested; the return value must be noted, and the remainder of the operation should be retried when possible.”

miri64, 2016-10-20:
I thought this issue was about 

a) a limited *reading* of input chars
b) compiling am image for samr21 on OSX

How would checking the return value of `write()` on native help in that case? ;)

Yonezawa-T2, 2016-10-20:
I misunderstood that macOS native RIOT is communicating to a board via UART.

kYc0o, 2016-10-20:
Ok let me make a test on those boards and I'll give feedback soon.

kYc0o, 2016-11-10:
Strange... but I can confirm that samr21-xpro shell (default example) only allows 64 bytes, while pba-d-01-kw2x allows 128 and when exceeded sends an error (shell: command not found).

smlng, 2016-11-10:
The 128 bytes is expected as this is the shell buffer size, see [1]. The 64 bytes/chars on the samr21-xpro is strange, because it only persists on macOS/OSX with any gcc version I checked (4.9, 5.2, and 5.4). Using the same gcc version on a Linux machine with samr21-xpro does not show this behavior.

[1]: https://github.com/RIOT-OS/RIOT/blob/master/sys/include/shell.h#L34

smlng, 2018-01-29:
still persist on macOS 10.13.3 with latest `arm-none-eabi-gcc` version 7.2 on samr21-xpro

smlng, 2018-01-29:
its work for pba-d-01-kw2x, same compiler

miri64, 2020-07-01:
Is this still the case, now that the shell saw a larger overhaul with #13197? @fjmolinas, can you please test?

fjmolinas, 2020-07-02:
I don't have a samr21-xpro with me today, but I'll test tomorrow, this might have been realated (and therefore fixed) by https://github.com/RIOT-OS/RIOT/pull/13195

aabadie, 2021-05-20:
> I don't have a samr21-xpro with me today, but I'll test tomorrow, this might have been realated (and therefore fixed) by #13195

Were you able to test this @fjmolinas ? I'm under the same impression.

fjmolinas, 2021-05-21:
Tested on OSX, it does not get stuck anymore, so fixed by #13195

```
> help
2021-05-21 11:17:41,756 #  help
2021-05-21 11:17:41,759 # Command              Description
2021-05-21 11:17:41,763 # ---------------------------------------
2021-05-21 11:17:41,768 # udp                  send data over UDP and listen on UDP ports
2021-05-21 11:17:41,772 # reboot               Reboot the node
2021-05-21 11:17:41,776 # version              Prints current RIOT_VERSION
2021-05-21 11:17:41,781 # pm                   interact with layered PM subsystem
2021-05-21 11:17:41,787 # ps                   Prints information about running threads.
2021-05-21 11:17:41,790 # ping6                Ping via ICMPv6
2021-05-21 11:17:41,793 # ping                 Alias for ping6
2021-05-21 11:17:41,797 # random_init          initializes the PRNG
2021-05-21 11:17:41,802 # random_get           returns 32 bit of pseudo randomness
2021-05-21 11:17:41,807 # nib                  Configure neighbor information base
2021-05-21 11:17:41,812 # ifconfig             Configure network interfaces
2021-05-21 11:17:41,819 # rpl                  rpl configuration tool ('rpl help' for more information)
2021-05-21 11:17:41,824 # 6ctx                 6LoWPAN context configuration tool
> reboot
2021-05-21 11:17:43,292 #  reboot
2021-05-21 11:17:43,357 # main(): This is RIOT! (Version: 2021.04-devel-1301-g958db-pr_edhoc_c)
2021-05-21 11:17:43,360 # RIOT network stack example application
2021-05-21 11:17:43,363 # All up, running the shell now
udp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobar
2021-05-21 11:18:01,093 #  udp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobar
2021-05-21 11:18:01,100 # Success: sent 18 byte(s) to [fe80::1430:2488:724e:e2f0]:1234
> udp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobarabcdef
help
2021-05-21 11:18:28,597 #  udp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobarabcdef�dp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobarabcdef�dp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobarabcdef�dp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobarabcdef�dp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobarabcdef�dp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobarabcdef�dp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobarabcdef�elp
2021-05-21 11:18:28,600 # shell: maximum line length exceeded
> 2021-05-21 11:18:28,748 #  help
2021-05-21 11:18:28,752 # Command              Description
2021-05-21 11:18:28,754 # ---------------------------------------
2021-05-21 11:18:28,760 # udp                  send data over UDP and listen on UDP ports
2021-05-21 11:18:28,763 # reboot               Reboot the node
2021-05-21 11:18:28,768 # version              Prints current RIOT_VERSION
2021-05-21 11:18:28,773 # pm                   interact with layered PM subsystem
2021-05-21 11:18:28,778 # ps                   Prints information about running threads.
2021-05-21 11:18:28,782 # ping6                Ping via ICMPv6
2021-05-21 11:18:28,785 # ping                 Alias for ping6
2021-05-21 11:18:28,789 # random_init          initializes the PRNG
2021-05-21 11:18:28,794 # random_get           returns 32 bit of pseudo randomness
2021-05-21 11:18:28,799 # nib                  Configure neighbor information base
2021-05-21 11:18:28,804 # ifconfig             Configure network interfaces
2021-05-21 11:18:28,811 # rpl                  rpl configuration tool ('rpl help' for more information)
2021-05-21 11:18:28,816 # 6ctx                 6LoWPAN context configuration tool
> 2021-05-21 11:18:28,884 #  help
2021-05-21 11:18:28,887 # Command              Description
2021-05-21 11:18:28,890 # ---------------------------------------
2021-05-21 11:18:28,897 # udp                  send data over UDP and listen on UDP ports
2021-05-21 11:18:28,899 # reboot               Reboot the node
2021-05-21 11:18:28,904 # version              Prints current RIOT_VERSION
2021-05-21 11:18:28,909 # pm                   interact with layered PM subsystem
2021-05-21 11:18:28,914 # ps                   Prints information about running threads.
2021-05-21 11:18:28,918 # ping6                Ping via ICMPv6
2021-05-21 11:18:28,921 # ping                 Alias for ping6
2021-05-21 11:18:28,925 # random_init          initializes the PRNG
2021-05-21 11:18:28,930 # random_get           returns 32 bit of pseudo randomness
2021-05-21 11:18:28,935 # nib                  Configure neighbor information base
2021-05-21 11:18:28,940 # ifconfig             Configure network interfaces
2021-05-21 11:18:28,947 # rpl                  rpl configuration tool ('rpl help' for more information)
2021-05-21 11:18:28,952 # 6ctx                 6LoWPAN context configuration tool
> hlep
2021-05-21 11:18:30,084 #  hlep
2021-05-21 11:18:30,087 # shell: command not found: hlep
> udp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobarabcdef
help
2021-05-21 11:18:40,332 #  udp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobarabcdef�dp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobarabcdef�elp
2021-05-21 11:18:40,335 # shell: maximum line length exceeded
> udp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobarabcdef\n
help
2021-05-21 11:18:48,677 #  udp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobarabcdef�udp send fe80::1430:2488:724e:e2f0 1234 foobarfoobarfoobarabcdef�help
2021-05-21 11:18:48,680 # shell: maximum line length exceeded
> 
```

closed 2021-05-21 by fjmolinas